MarlinFirmware/Marlin #18527: [FR] (feature request title) //#define MULTI_NOZZLE_DUPLICATION

SlickNickeL, 2020-07-03:
This topic was closed. Duplication mode does not work. As of July
 1 2020.

thisiskeithb, 2020-07-03:
Is this a feature request or a bug report? Either way... 

Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

SlickNickeL, 2020-07-03:
This was an open request that was prematurely shut down. Not going to rehash this thread. Just bringing  it to someone’s attention that it was closed, when there was no solution to it.

boelle, 2020-07-03:
you got a link to what was closed? this one has never been closed

sjasonsmith, 2020-07-03:
@SlickNickeL, I assume you are referring to this bug? #16524 (MULTI_NOZZLE_DUPLICATION limited to 2 copies.)

There was also a feature request about this. Since the code _looks_ like it should support multi-nozzle duplication dealing with it as bugs seems to make more sense.

I have no idea whether that bug which closed due to inactivity still reflects your current issues.

sjasonsmith, 2020-07-03:
@SlickNickeL we also now have a better mechanism for dealing with abandoned issues now, which will post with a warning that an issue is about to be closed due to inactivity. All you will have to do is reply to the thread and it will prevent it from happening.

For now, I think we just need to identify:
1. Which issues or feature requests were closed without reaching resolution?
2. Are they still accurate, or would it be better to file new bug reports with very specific failure details.
    It is easier for people to contribute fixes when the problem is very well scoped and described.

sjasonsmith, 2020-07-03:
@SlickNickeL, based on your comment on #16524 I have re-opened that issue, and will close this one.

closed 2020-07-03 by sjasonsmith