MarlinFirmware/Marlin #18529: [BUG] LCD Temperature Menu Shows Two Fans When Configured For Just One

borland1, 2020-07-03:
### Bug Description

Marlin v2.0.5.3 Configuration is for only One Fan (extruder E0), however LCD 'Temperature' menu shows: 
```
Fan Speed 1:     0%
Fan Speed 2:     0%
```

**Expected behavior:** Since only one fan being used on printer and configuration is for only one fan, expect to see only one fan in LCD Temperature menu:

`Fan Speed:    0% `

### My Configurations

[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4870593/ConfigFiles.zip)

thisiskeithb, 2020-07-03:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

GMagician, 2020-07-03:
@thisiskeithb I have the same issue, my boards has configured pins for 3 fans but my printer has only 1 extruder and in LCD I see 3 fans controll. Never payed attention to that, I thought it was "normal"

sjasonsmith, 2020-07-03:
I can also confirm that, because that is how I tested the broken fan menu a few days ago. I don’t know whether it is correct behavior or not. I had one extruder but three fans in the menu.

I can imagine a scenario where people have extra fans they control through their slicer. I don’t know whether that is important, or if this number of fans should match the number of extruders.

thisiskeithb, 2020-07-03:
If multiple fans are defined in the pins file, I assume they'll be enabled in Marlin whether you have them hooked up or not.

borland1, 2020-07-03:
On bugfix-2.0.x, I get compile errors with Arduino 1.8.10. After cutting and pasting from 2.0.5.3 into bugfix-2.0.x config.h and config_adv.h, unable to test.  2.0.5.3 compiles fine.  Also, I didn't see any recent changes that would have fixed this problem.

Keith,  The file `Configuration_adv.h`, provides an ability to override the pins file for auto fans, but it's unclear how to disable additional fans.  I assumed I needed to comment out the un-used fans feature.  Here's what's in the `Configuration_adv.h` file:
```
* Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 */

// #define E0_AUTO_FAN_PIN -1     // using this fan, pin file has correct pin assigned, so comment out, or leave at -1 ????
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define E6_AUTO_FAN_PIN -1
#define E7_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1
```
Would expect to see` #ifndef` ... `#define` code blocks for the fans, so it is clear what pin assignments can be overridden in either of the config files.

In file `menu_temperature.cpp`  I see code:  `#if FAN_COUNT > 1`  `#if HAS_FAN0`  `#if HAS_FAN1`
Then in file `Conditional_post.h` I see code that determines the `FAN_COUNT` value.

boelle, 2020-07-03:
@borland1 are the compile errors also present when using vscode+platform.io?

sjasonsmith, 2020-07-03:
@borland1 & @boelle, compile errors are to be expected when using a 2.0.5.3 configuration on the bugfix branch. There are significant enough changes that they will require some rework.

borland1, 2020-07-03:
Compile errors were minor: `'multiple libraries found for: SPI.h, SoftwareSerial.h, U8glib.h, TMCStepper.h,   Fork/exec: tools\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc.exe: The filename or extension, and file names too long.`

Trying to keep it simple. Did not try this with Platform.io.  I still use Arduino IDE with Mega 2560 processor.  With Platform.io IDE, I have only had success with LPC1768 processor.

sjasonsmith, 2020-07-03:
At this point I suggest we not focus on the compiler issues, and instead on whether showing all the fans is expected behavior or not.

I see several different possibilities, I'm not sure which is correct:

1. This problem is a configuration error and easily avoided
   - Response: If true, someone please tell us and this can be closed. I don't immediately know how to do it.
1. This is expected behavior, and should remain the default
   - Response: This turns into a **feature request** to make disabling those fans possible.
1. This is not expected behavior, fan count should default to extruder count
   - Response: This is a **bug**, and the menu should be limited to the extruder count. A new option may be needed to override the default fan count, for users who use fans for purposes other than extruder cooling.

I **personally** think it should be option 3, with the default set to the number of fans actually used by extruders. A related question would be whether that disables those fans entirely (including through M106), or only removes them from the menu.

ellensp, 2020-07-04:
"Fork/exec: tools\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc.exe: The filename or extension, and file names too long" 
This is the windows exec path limit being exceeded.. 
The command path + parameters in full cannot be > 32k characters or windows spits the dummy.

borland1, 2020-07-04:
With latest `bugfix-2.0.x` , was able upload and verify this LCD Temperature Menu bug still exists.   Thanks for the help.

mapfde, 2020-07-05:
If I configure 3 fan ports for my printer, I expect to be able to control them in the menu.

sjasonsmith, 2020-07-05:
> If I configure 3 fan ports for my printer, I expect to be able to control them in the menu.

The problem is that for many users their boards have more fan ports than are in use, and it would be nice to only list the fans which are actually present.

In general, there is a desire to not require users to edit their pins files for common scenarios, so this is currently a gap. We just need some input from "decision makers" on what the default experience should be, to know what should be implemented.

@thinkyhead if you look at my comment a little ways up, I propose three different approaches for this. If you can voice your opinion of which of those is correct, I'm sure it can be implemented by somebody else.

hamster65, 2020-07-06:
my printer also shows 2 Fans in the menu. The board (mks sgen l) supports 2 extruders,so there are 2 connectors for fans.
However, I use the second fan output for the heated chamber. Things get weird if I (accidentally )set the second fan to anything other than 0%.

GMagician, 2020-07-06:
One solution to this may be to add to configuration "FANS" something like "EXTRUDERS" to define how many devices are installed. To be back compatible it defaults to nr of fan pins

borland1, 2020-08-01:
> One solution to this may be to add to configuration "FANS" something like "EXTRUDERS" to define how many devices are installed. To be back compatible it defaults to nr of fan pins

Yes, apparently that was done early in RAMPS compatible Marlin FW.  I found these code comments in file `pins_RAMPS.h` which gives a clue as to how the Fan count was derived from the motherboard name in file `Configuration.h`  ...
```
/**
 * Arduino Mega with RAMPS v1.4 (or v1.3) pin assignments
 *
 * Applies to the following boards:
 *
 *  RAMPS_14_EFB (Hotend, Fan, Bed)
 *  RAMPS_14_EEB (Hotend0, Hotend1, Bed)
 *  RAMPS_14_EFF (Hotend, Fan0, Fan1)
 *  RAMPS_14_EEF (Hotend0, Hotend1, Fan)
 *  RAMPS_14_SF  (Spindle, Controller Fan)
 *
 *  RAMPS_13_EFB (Hotend, Fan, Bed)
 *  RAMPS_13_EEB (Hotend0, Hotend1, Bed)
 *  RAMPS_13_EFF (Hotend, Fan0, Fan1)
 *  RAMPS_13_EEF (Hotend0, Hotend1, Fan)
 *  RAMPS_13_SF  (Spindle, Controller Fan)
 *
 *  Other pins_MYBOARD.h files may override these defaults
```
So, my workaround for this bug is to add `IS_RAMPS_EFB` to the `MOTHERBOARD` #define like this..
```
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_GEN_L
  #define IS_RAMPS_EFB  // ( Hotend, Fan, Bed )
#endif 
```
Now only one Fan appears in the LCD's Temperature menu.  If I enable the extruders Auto Fan feature ( E0_AUTO_FAN_PIN ), then no Fan appears in the LCD's Temperature menu.  This is just a workaround, not a bug fix.

GMagician, 2020-08-01:
> Yes, apparently that was done early in RAMPS compatible Marlin FW. I found these code comments in file pins_RAMPS.h which gives a clue as to how the Fan count was derived from the motherboard name in file Configuration.

That's not exactly what I said. These ramps definitions change the pins definitions. They move fans pin and or remove them so as a result you see what you described. What I proposed was a new definition in cnf called FANS that may remove some of the fan related menu items and code actions, but I'm not sure this is a viable solution

GMagician, 2020-08-01:
@borland1 

> Keith, The file Configuration_adv.h, provides an ability to override the pins file for auto fans, but it's unclear how to disable additional fans. Here's what's in the Configuration_adv.h file:

This is simply done by defining Ex_AUTO_FAN_PIN. The only limit you have you can't do that on FAN_PIN

PR #18883 (2020-08-20): Let user limit number of fans
This should fix #18529

Once approved I'll update all other configurations files
closed 2020-09-01 by thisiskeithb