MarlinFirmware/Marlin #18530: Firmware not completely uploading to BTT SKR 1.4

Thijsvanzwam, 2020-07-03:
When uploading the firmware to the BTT SKR 1.4, using the sdcard, it does not completely upload.

The bin file gets changed to a cur file and some parts are uploaded but not all.

For some reason the preheat option does not change in the firmware, I wanted to lower the bed temperature from 60 to 30 at PLA but it stays on 60 on the printer.

uploaded a few times and run commands M500 and M501 but it stays at 60.
other parts like the stepper current does get changed.
I want to know for sure everything uploads correctly because I changed the thermal runaway settings as well and cannot check if they are correct now.

Tried several sdcards and compiled the firmware several Times in VScode. no luck...

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4871178/config.zip)

GMagician, 2020-07-03:
m502 & m500 is correct sequence!

sjasonsmith, 2020-07-03:
It seems like this is likely a minor configuration issue or user error, and not an actual bug. One recommendation prior to the standard text blob I'm about to post, is to change `CUSTOM_MACHINE_NAME` to something unique. You can then go to `About Printer` then `Printer Info` to see if the name of the printer changed. That would tell you for sure that you are running the new firmware. (You almost certainly are if firmware.bin changed to FIRMWARE.CUR)

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-03 by sjasonsmith

Thijsvanzwam, 2020-07-04:
Found it!
the problem was in the config.ini file in the BTT TFT firmware. Settings here overwrite Marlin settings.
No bug but perhaps good to know if anyone else has this issue.

sjasonsmith, 2020-07-04:
@Thijsvanzwam thanks for sharing that. One more thing to worry about messing up settings, along with slicers! 😁