MarlinFirmware/Marlin #18531: [FR] Add some git instructions to the readme in new branches

bill-orange, 2020-07-04:
I hear from @thinkyhead that version 2.0.6 is coming soon. 

Perhaps some instructions could be added to the readme to describe how the new upstream branch can be brought into the user’s own Marlin fork and then into the user’s own local folder. 

This text could also be reused as new branches are created in the future. 

I can not be the only one having trouble figuring this out. GitHub desktop simplifies routine operations but adding a new local branch from a new  upstream branch is confusing. A brief Google search revealed that I am not alone in this confusion.

While this is more of a Github issue than a Marlin issue, the sheer number of Marlin forks and branches makes this a worthwhile issue to address.

thinkyhead, 2020-07-17:
Those who have GitHub accounts should read the GitHub documentation. At most, we might add a link to the GitHub help pages on the topic, or one of the various YouTube videos that cover this topic. But, I certainly don't want to duplicate that effort, myself, and as we all know, people don't read instructions, and the level of user who would need this kind of help will most likely still need a human to hold their hand.

bill-orange, 2020-07-17:
A link might be helpful. Boy, I wish I could find a human to hold my hand on this topic. Unfortunately, in my circle of tech savvy friend no one would even know what I am talking about :). I will close this issue.

closed 2020-07-17 by bill-orange