MarlinFirmware/Marlin #18534: [FR] Z-probe control

DrumClock, 2020-07-04:
I use Touch-MI Probe by hotends.fr
Is it possible to set a check that the probe is deploy ?

If the probe is not unlocked at G28  -  **the HotBed hits the nozzle.**

Configuration Z-probe:
```

#define TOUCH_MI_PROBE
#if ENABLED(TOUCH_MI_PROBE)
  #define TOUCH_MI_RETRACT_Z 1.0                  // Height at which the probe retracts
  #define TOUCH_MI_DEPLOY_XPOS (X_MAX_BED + 2)    // For a magnet on the right side of the bed
  //#define TOUCH_MI_MANUAL_DEPLOY                // For manual deploy (LCD menu)
#endif
```

sjasonsmith, 2020-07-05:
@DrumClock is the new title correct? When I read your description, it sounds like you are asking for the opposite. You want a way to make sure the probe is deployed, not stowed?

DrumClock, 2020-07-05:
I want a way to ensure that the probe is deployed front of the G28.
If the probe remains inserted, the hotbed will hit the nozzle.

DrumClock, 2020-07-05:
**Simulation when the probe is not deployed at G28:**
https://www.youtube.com/watch?v=x6D2-QvT55o

sjasonsmith, 2020-07-05:
@DrumClock, you should be able to adjust the title, it seems thinkyhead got the condition wrong when he clarified it.

I understand the issue you are describing, but I don't know enough about the Touch Mi to know if what you want is possible or not. I had only noticed that your description did not match the title.

DrumClock, 2020-07-05:
@sjasonsmith sorry i don't know much english. 
What should I write to make it understandable?

sjasonsmith, 2020-07-05:
"Make sure Touch-MI is deployed before homing"

DrumClock, 2020-07-05:
@sjasonsmith thank you 👍 
 I think this will be a useful feature for other types of Z-probes

DrumClock, 2020-07-06:
@sjasonsmith, @ thinkyhead
Very sorry. Everything is alright. **Everything is alright**

I had **faulty Touch MI electronics**.
The above problem occurs when the HIGH level is permanently at the Z-probe output.

closed 2020-07-06 by DrumClock