MarlinFirmware/Marlin #18542: [FR] New Chopper Timing Preset Option For 0.9 Degree Steppers.

Portzal, 2020-07-05:
On the back of https://github.com/MarlinFirmware/Marlin/issues/18286#issue-638088876 
would it be reasonable to include another Chopper timing Preset tuned towards noisy/vibrating 0.9 degree steppers?
Possible Name "CHOPPER_09STEP_24V" (3, -1, 5)
After a full day of testing different chopper timing combinations I have come up with the following numbers.
CHOPPER_TIMING { 3, -1, 5 } 
These numbers give a good balance between StealthChop, above 110mm/s noise levels starring to increase and 
SpreadCycle, above 120mm/s steppers start to calm down.

@h3po did some testing with 3 different settings and posted the following video, https://youtu.be/A7GWkTTyzW0

thisiskeithb, 2020-07-05:
Which stepper manufacturer/model numbers has this been tested with? It may not be universal for all 0.9 degree steppers.

Portzal, 2020-07-06:
My machine has E3D Steppers, I believe it has worked with some other brands as well. It’s not aimed at all 0.9 steppers, only as a option if you are experiencing excessive noise or vibration. It would probably need the inclusion of short explanation for the use of this preset.

Portzal, 2020-07-07:
@boelle @thisiskeithb  I have made the changes in my firmware and it builds successfully. 
could I please get some feedback as to whether I actually have it set up correctly.
I will then work on following the guides and learn how to do a Pull request.
tmc_util.h
#define CHOPPER_DEFAULT_12V  { 3, -1, 1 }
#define CHOPPER_DEFAULT_19V  { 4,  1, 1 }
#define CHOPPER_DEFAULT_24V  { 4,  2, 1 }
#define CHOPPER_DEFAULT_36V  { 5,  2, 4 }
#define CHOPPER_PRUSAMK3_24V { 3, -2, 6 }
#define CHOPPER_MARLIN_119   { 5,  2, 3 }
#define CHOPPER_09STEP_24V   { 3, -1, 5 }  **new preset**

configuration_adv.h
* Optimize spreadCycle chopper parameters by using predefined parameter sets
   * or with the help of an example included in the library.
   * Provided parameter sets are
   * CHOPPER_DEFAULT_12V
   * CHOPPER_DEFAULT_19V
   * CHOPPER_DEFAULT_24V  
   * CHOPPER_DEFAULT_36V
   * CHOPPER_PRUSAMK3_24V // Imported parameters from the official Prusa firmware for MK3 (24V)  
   * CHOPPER_MARLIN_119   // Old defaults from Marlin v1.1.9
   * CHOPPER_09STEP_24V  //option for noisy or vibrating 0.9 degree steppers **new preset**
   *
   * Define you own with
   * { <off_time[1..15]>, <hysteresis_end[-3..12]>, hysteresis_start[1..8] }
   */

thisiskeithb, 2020-07-07:
I haven’t tuned chopper settings beyond the defaults, but I assume it’ll need to be adjusted for everyone’s specific drivers & motor characteristics.

@teemuatlut would likely know more.

Evg33, 2020-07-08:
Very cool !!!
Now there is no that nasty sound!
SKR1.4T + 2209 + 0.9/2A StepperOnline

00xED, 2020-07-09:
Works nice. Tried it with SKR1.3 + TMC2208 + 0.9A (StepperOnline 17HM15-0904S 0.9deg 0.9A 12.0mH).

freighttt, 2020-07-09:
To chime in, I have LDO 0.9’s but these values did improve things for me. I ended up adjusting mine further via the calculator but I do support adding the values posted. They definitely are a big improvement from stock.

closed 2020-07-17 by thinkyhead

wolfgangmauer, 2020-07-27:
Nice work, i have SKR 1.3/TMC2209 on **12V**, hope this will work also with 12V ?!

looxonline, 2020-08-15:
> To chime in, I have LDO 0.9’s but these values did improve things for me. I ended up adjusting mine further via the calculator but I do support adding the values posted. They definitely are a big improvement from stock.

I'm about to install some LDOs. Any chance of explaining what further adjustments you did?

freighttt, 2020-08-27:
> I'm about to install some LDOs. Any chance of explaining what further adjustments you did?

i plugged the specs into the tmc2209 calculator. the chopper settings ended up being 3, -1, 1

teemuatlut, 2020-08-27:
`3, -1, 1` would be the same as [CHOPPER_DEFAULT_12V](https://github.com/MarlinFirmware/Marlin/blob/96e29cd7e5aa91fad72788251caf6ffd38df4624/Marlin/src/feature/tmc_util.h#L32-L38).

SFRSKZ, 2020-09-04:
> i plugged the specs into the tmc2209 calculator. the chopper settings ended up being 3, -1, 1

i searched that calculator but couldnt find. 3,-1, 5 was good for LDO 0.9 but my new Moons 2a 0.9s didnt like it. now motors sound like driven by a4988

Hyrules, 2020-09-21:
It would be nice if someone could post the link to the calculator. I also have moon's stepper moter on 2209 and skr1.4 and they are noisy and vibrating as well. I'm about to try the new preset to see if this fixes it or would like to try to fix it with the calculator if this does not do it. Also is it possible to set the chopping by axis ? My X and Y are 0.9 deg steppers but the Z is still at 1.8.

looxonline, 2020-09-21:
> It would be nice if someone could post the link to the calculator. I also have 0.9 moon's stepper motor on 2209 and skr1.4 and they are noisy and vibrating as well. I'm about to try the new preset to see if this fixes it or would like to try to fix it with the calculator if this does not do it.

Just because you asked so nicely :)
[TMC2209_Calculations-1.xlsx](https://github.com/MarlinFirmware/Marlin/files/5256986/TMC2209_Calculations-1.xlsx)

Hyrules, 2020-09-21:
Thanks alot !

ShohninDmitriy, 2020-09-21:
> TMC2209_Calculations-1.xlsx
You have an error in the TMC Configurator the driver can't work at 168 MHz it works at 12 MHz

Hyrules, 2020-09-21:
> i plugged the specs into the tmc2209 calculator. the chopper settings ended up being 3, -1, 1

I also get this value when I enter my details in the calculator for my moon's MS17HA2P4150 motors.

ShohninDmitriy, 2020-09-21:
> I also get this value when I enter my details in the calculator for my moon's MS17HA2P4150 motors.

3, - 3, 1

Hyrules, 2020-09-22:
Wow thanks ! this does makes thing better indeed. I have some noise with acceleration and now it's almost completly gone.