MarlinFirmware/Marlin #18543: [BUG] PR #18490 breaks PRINTER_EVENT_LEDS and M150 without new "I" option.

ManuelMcLure, 2020-07-05:
### Bug Description

After PR #18490, ``M150`` without the new ``I`` option does nothing, and ``PRINTER_EVENT_LEDS`` no longer works.

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4874349/Configuration_adv.zip)

### Steps to Reproduce

* Use the above configuration (adjust Neopixel settings for your specific printer).
* Execute ``M150 U255`` - all Neopixels should light up bright green, but they do not.
* Execute ``M150 I99 U255`` - now all Neopixels should light up.
* Heat up bed with ``M190`` - ``PRINTER_EVENT_LEDS`` should operate but does not.

It looks like the PR broke Neopixels except for ``M150`` with the ``I`` option specified.

#### Additional Information

Should
```
if (get_neo_index() < 0) { 
```
be
```
if(get_neo_index() >= 0) {
```
?

ellensp, 2020-07-05:
Confirmed its broken,  looks like thinkyhead broke it when he changed index to support -1

ellensp, 2020-07-05:
Your fix is correct.

ManuelMcLure, 2020-07-05:
Created PR #18544.

PR #18544 (2020-07-05): Fix Neopixels

### Description

PR #18490 broke Neopixel ``PRINTER_EVENT_LEDS`` and ``M150`` functionality. This PR fixes the breakage.

### Benefits

Fix Neopixels.

### Related Issues

Fixes #18543
closed 2020-07-06 by ellensp