MarlinFirmware/Marlin #18545: [BUG] Fix from #17770 breaks compiling with AUTO_BED_LEVELING_BILINEAR

Masanetz, 2020-07-05:
The fix from #17770 seems to break compiling with AUTO_BED_LEVELING_BILINEAR, as abl_points nether declared nor has it a value...
Maybe the comment in line 248 of G29.cpp is somewhat misleading:
`#elif ENABLED(PROBE_MANUALLY) // Bilinear`
changed to
`#elif EITHER(PROBE_MANUALLY, AUTO_BED_LEVELING_BILINEAR) // Bilinear`
it compiled again...
  

>     In file included from Marlin/src/gcode/bedlevel/abl/../../../inc/MarlinConfig.h:47:0,
>     from Marlin/src/gcode/bedlevel/abl/G29.cpp:27:
>     Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
>     Marlin/src/gcode/bedlevel/abl/G29.cpp:646:91: error: 'abl_points' was not declared in this scope
>     if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
>     ^
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:111:57: note: in definition of macro '_SEP_2'
>     #define _SEP_2(PRE,V) serial_echopair_PGM(PSTR(PRE),V)
>     ^
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:174:45: note: in expansion of macro '_SELP_3'
>     #define _SELP_5(a,b,V...) do{ _SEP_2(a,b); _SELP_3(V); }while(0)
>     ^~~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:168:28: note: in expansion of macro '_SELP_5'
>     #define __SELP_N(N,V...) SELP##N(V)
>     ^~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:169:28: note: in expansion of macro '__SELP_N'
>     #define _SELP_N(N,V...) __SELP_N(N,V)
>     ^~~~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:201:33: note: in expansion of macro '_SELP_N'
>     #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
>     ^~~~~~~
>     Marlin/src/gcode/bedlevel/abl/G29.cpp:646:30: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
>     if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
>     ^~~~~~~~~~~~~~~~~
>     Marlin/src/gcode/bedlevel/abl/G29.cpp:646:91: note: suggested alternative: 'abc_pos_t'
>     if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
>     ^
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:111:57: note: in definition of macro '_SEP_2'
>     #define _SEP_2(PRE,V) serial_echopair_PGM(PSTR(PRE),V)
>     ^
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:174:45: note: in expansion of macro '_SELP_3'
>     #define _SELP_5(a,b,V...) do{ _SEP_2(a,b); _SELP_3(V); }while(0)
>     ^~~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:168:28: note: in expansion of macro '_SELP_5'
>     #define __SELP_N(N,V...) SELP##N(V)
>     ^~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:169:28: note: in expansion of macro '__SELP_N'
>     #define _SELP_N(N,V...) __SELP_N(N,V)
>     ^~~~~~~~
>     Marlin/src/gcode/bedlevel/abl/../../../inc/../core/serial.h:201:33: note: in expansion of macro '_SELP_N'
>     #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
>     ^~~~~~~
>     Marlin/src/gcode/bedlevel/abl/G29.cpp:646:30: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
>     if (verbose_level) SERIAL_ECHOLNPAIR("Probing mesh point ", int(pt_index), "/", abl_points, ".");
>     ^~~~~~~~~~~~~~~~~
>     *** [.pio/build/chitu_v5_gpio_init/src/src/gcode/bedlevel/abl/G29.cpp.o] Error 1

thisiskeithb, 2020-07-05:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also: Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

RemcoPerlee, 2020-07-05:
I did a different fix to get it to compile. I replaced "abl_points" to "int(GRID_MAX_POINTS)" on line 646

RemcoPerlee, 2020-07-05:
I was using the file that is currently on GitHub (checked just a sec ago)

ellensp, 2020-07-05:
That is latest bugfix.  It got broken 4 hours ago on bugfix only. https://github.com/MarlinFirmware/Marlin/commit/ea520e670a1c5fba9d6c587c944a787dbc09165e

thisiskeithb, 2020-07-05:
> That is latest bugfix. It got broken 4 hours ago on bugfix only. [ea520e6](https://github.com/MarlinFirmware/Marlin/commit/ea520e670a1c5fba9d6c587c944a787dbc09165e)

Bug confirmed.

TheGadge, 2020-07-05:
Thanks for this post, was going insane. I can compile fine without AUTO_BED_LEVELING_BILINEAR,
SKR mini e3 2. Is there any 'fix' we can do ourselves for this please?

qwewer0, 2020-07-05:
> (234) int constexpr abl_points = GRID_MAX_POINTS; // ADDED FOR DEBUG TAW

With this fix it compiles for me.
([Without the fix](https://github.com/MarlinFirmware/Marlin/issues/17770#issuecomment-653923566))

qwewer0, 2020-07-05:
Fixed with https://github.com/MarlinFirmware/Marlin/pull/18552

sjasonsmith, 2020-07-05:
I can confirm that it is now fixed for me as well.

closed 2020-07-05 by sjasonsmith