MarlinFirmware/Marlin #18546: [FR] (physical units independent of microstepping)

curvi, 2020-07-05:
### Description

I'd like to ask if there is a special reason, why microstepping is not independent from the units used.
To me 1 step is a full step of a stepper motor. This general concept is broken by marlins interpretation to have 16 microsteps activated by default and depending on this number in the DEFAULT_AXIS_STEPS_PER_UNIT.

### Feature Workflow

If the microstepping number is taken into account for the steps/unit, one could independently specify these numbers, see crude example below.

#### Additional Information

I've added something like this: (where I am a bit curious why x,y,z microstep constants are defined in the config. scope, but E0_MICROSTEPS is not,.. so as a little "hack" I provided this number again. Of course this has to match the one in config_adv.

#ifndef E0_MICROSTEPS
  #define E0_MICROSTEPS 256
#endif
#define STEPS_PER_MM_X (5*X_MICROSTEPS)
#define STEPS_PER_MM_Y (5*Y_MICROSTEPS)
#define STEPS_PER_MM_Z (25*Z_MICROSTEPS)
#define STEPS_PER_MM_E0 (415/16*E0_MICROSTEPS)
#define DEFAULT_AXIS_STEPS_PER_UNIT { STEPS_PER_MM_X, STEPS_PER_MM_Y, STEPS_PER_MM_Z, STEPS_PER_MM_E0 }
// #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 415 } // this is @ 1/16 microstepping for all axes

ellensp, 2020-07-05:
1 step is 1 pulse on the stepper driver control pin. It has never been 1 full step of the stepper motor.

sjasonsmith, 2020-07-05:
I’m not sure what real benefit this provides. You are of course free to do it in your own config.

curvi, 2020-07-05:
Oh, I see. It makes sense to have the steps that way. Anyway, the benefit of having to set 2 properties independently is, that it is much more accessible. If you look online for tutorials, all I found had something like "this messes up something else, so we gonna keep that at 16 for now"...
For me it is a bummer having to think of it, even though I know what it does. Also something different is going on with the microsteps on E, but it hasn't done me any good anyway, so here I stay with 16.

I am ok with the idea being rejected, I just wanted to draw a little bit attention to this unfortunate coupling. As I have to work with hardware as well, I've learned quickly to abstract all physical units and leave them in SI, while scaling is done as necessary in the program.

zillarob, 2020-07-09:
Could be a neat thing.
With so many boards now coming setup for spi/uart, it might be nice to be able to swap around different microstep without having to change the step/mm.

italocjs, 2020-07-15:
i think It would be more complicated to set up the firmware (you would need to tell which driver is being used, which microstepping is selected by the jumpers, which pulley and which belt are used).  at least for me its easier to use prusa calculator and just select these things from the dropdown list.     Maybe a indication to this tool (if it isnt already there) could be nice.

thinkyhead, 2020-07-15:
> microstepping is not independent from the units used.

It is! In fact, on a SCARA the XY axes specify steps-per-degree.

sjasonsmith, 2020-07-18:
I personally feel that doing this well depends highly on the machine's configuration. For basic machines you could have variables for pulley size, belt pitch, screw pitch, etc, and make it very easy to define a machine. The problem is that as soon as a machine diverges from "normal" there are a huge number of different possible variables to consider.

Overall, I think this is better suited to a separate configuration tool / calculator than building all the math into the configuration files.

vlsi, 2023-09-04:
> For basic machines you could have variables for pulley size, belt pitch, screw pitch, etc, and make it very easy to define a machine

That is fair. However mechanical properties and and programmatic ones differ.
Of course, if you swap a pulley with a bigger one, you would need to adjust the configuration. One can't just replace a pulley, as they must update something else.

Imagine users sharing config files for their printers. It is likely they will have similar hardware builds, and they might want to experiment with different currents, microsteps, and interpolation modes. It would be way easier for them 

At the same time, changing microsteps, current, or interpolation modes does not change the system's physical properties. It would be easier if they could share something like "Reborn2 moves head by 40mm for a single X motor revolution". Then the users could alter non-physical values like "G29 grid size", "stepper currents", or "stepper microsteps".