MarlinFirmware/Marlin #18547: [BUG]Compile error when enable #define POWER_MONITOR_CURRENT

ccccmagicboy, 2020-07-05:
### Bug Description

 I got this compile error below when I enabled **#define POWER_MONITOR_CURRENT** in Configuration_adv.h file.

It is seem that function define of **ftostr21ns**() and **ui16tostr4**() are not founded. 

I want to test with this new feature, is there any clue to fix it?

```
Compiling .pio\build\STM32F103RC_meeb\src\src\feature\touch\xpt2046.cpp.o
Marlin\src\feature\power_monitor.cpp: In static member function 'static void PowerMonitor::draw_current()':
Marlin\src\feature\power_monitor.cpp:51:34: error: 'ftostr21ns' was not declared in this scope
       lcd_put_u8str(amps < 100 ? ftostr21ns(amps) : ui16tostr4((uint16_t)amps));
                                  ^~~~~~~~~~
Marlin\src\feature\power_monitor.cpp:51:34: note: suggested alternative: 'dtostrf'
       lcd_put_u8str(amps < 100 ? ftostr21ns(amps) : ui16tostr4((uint16_t)amps));
                                  ^~~~~~~~~~
                                  dtostrf
Marlin\src\feature\power_monitor.cpp:51:53: error: 'ui16tostr4' was not declared in this scope
       lcd_put_u8str(amps < 100 ? ftostr21ns(amps) : ui16tostr4((uint16_t)amps));
                                                     ^~~~~~~~~~
Marlin\src\feature\power_monitor.cpp:51:53: note: suggested alternative: 'dtostrf'
       lcd_put_u8str(amps < 100 ? ftostr21ns(amps) : ui16tostr4((uint16_t)amps));
                                                     ^~~~~~~~~~
                                                     dtostrf
Marlin\src\feature\power_monitor.cpp: In static member function 'static void PowerMonitor::draw_voltage()':
Marlin\src\feature\power_monitor.cpp:59:35: error: 'ftostr21ns' was not declared in this scope
       lcd_put_u8str(volts < 100 ? ftostr21ns(volts) : ui16tostr4((uint16_t)volts));
                                   ^~~~~~~~~~
Marlin\src\feature\power_monitor.cpp:59:35: note: suggested alternative: 'dtostrf'
       lcd_put_u8str(volts < 100 ? ftostr21ns(volts) : ui16tostr4((uint16_t)volts));
                                   ^~~~~~~~~~
                                   dtostrf
Marlin\src\feature\power_monitor.cpp:59:55: error: 'ui16tostr4' was not declared in this scope
       lcd_put_u8str(volts < 100 ? ftostr21ns(volts) : ui16tostr4((uint16_t)volts));
                                                       ^~~~~~~~~~
Marlin\src\feature\power_monitor.cpp:59:55: note: suggested alternative: 'dtostrf'
       lcd_put_u8str(volts < 100 ? ftostr21ns(volts) : ui16tostr4((uint16_t)volts));
                                                       ^~~~~~~~~~
                                                       dtostrf
Marlin\src\feature\power_monitor.cpp: In static member function 'static void PowerMonitor::draw_power()':
Marlin\src\feature\power_monitor.cpp:67:35: error: 'ftostr21ns' was not declared in this scope
       lcd_put_u8str(power < 100 ? ftostr21ns(power) : ui16tostr4((uint16_t)power));
                                   ^~~~~~~~~~
Marlin\src\feature\power_monitor.cpp:67:35: note: suggested alternative: 'dtostrf'
       lcd_put_u8str(power < 100 ? ftostr21ns(power) : ui16tostr4((uint16_t)power));
                                   ^~~~~~~~~~
                                   dtostrf
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_meeb\src\src\feature\power_monitor.cpp.o] Error 1
============================================================================================== [FAILED] Took 31.27 seconds ==============================================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
MightyBoard1280           IGNORED
MightyBoard2560           IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
melzi_optimized           IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
DUE_archim                IGNORED
DUE_archim_debug          IGNORED
SAMD51_grandcentral_m4    IGNORED
LPC1768                   IGNORED
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_meeb          FAILED    00:00:31.273
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin_nano35          IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_e3              IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
STM32F070RB_malyan        IGNORED
malyan_M300               IGNORED
chitu_f103                IGNORED
chitu_v5_gpio_init        IGNORED
chitu_f103_lvgl           IGNORED
STM32F103RET6_creality    IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
rumba32                   IGNORED
esp32                     IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
linux_native              IGNORED
include_tree              IGNORED
```

### My Configurations

[pins_CCROBOT_MEEB_3DP.zip](https://github.com/MarlinFirmware/Marlin/files/4874741/pins_CCROBOT_MEEB_3DP.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4874742/Marlin.zip)

### Steps to Reproduce

1. checkout newest bugfix-2.0.x branch
2. copy the config files and pin file include
3. compile the project in platformio

ellensp, 2020-07-05:
There are a bunch of similar functions in Marlin/src/libs/numtostr .h and .cpp just not ftostr21ns and ui16tostr4
ftostr31ns exists as does ui16tostr4rj  try those...  it just creates the what is displayed on the lcd.. so worse case lcd will be badly formatted.

ccccmagicboy, 2020-07-05:
@ellensp Thanks for your tips, I will try it on.

sjasonsmith, 2020-07-05:
@ccccmagicboy are you planning to fix this and submit a Pull Request yourself?

ccccmagicboy, 2020-07-06:
@sjasonsmith Yes, let me do it. Any help appreciated!

ccccmagicboy, 2020-07-06:
Let me guess ftostr21ns() mean format float to xx.x, and ui16tostr4() mean format uint16 to xxxx string, it that correct?

ccccmagicboy, 2020-07-06:
@sjasonsmith 
I Got confused, there seems to be a lot of work to be done, a little beyond my ability.

ellensp, 2020-07-06:
"ftostr21ns() mean format float to xx.x"

That would be my guess since it only use this when value is less than 100
but if you read the comment for marlins ftostr31ns it also says  // Convert unsigned float to string with 12.3 format
So seems to be the same thing, you you can just change ftostr21ns to ftostr21ns in power_monitor.cpp.

and "ui16tostr4() mean format uint16 to xxxx string"

Yes.

ellensp, 2020-07-06:
having a play, there are more things broken than just those two string conversions.

ellensp, 2020-07-06:
right that was a mess. most of the lcd code was pre v2 code...
I have it working. will create a PR shortly

ellensp, 2020-07-06:
The PR has been added..  It was quite broken........
But now, if you enable it, the last 5 characters of the display will display 4 digits and a A for amp, or a V for volts or a W for watts
NB you don't get Watts unless you enable both Volts and Amps.  

This also enables gcode M430 eg:

SENDING:M430
Current: 96.01A  Voltage: 1.70V  Power: 163.52W    

NB I don't have probes attached. these are just simulated values from a voltage divider.

ccccmagicboy, 2020-07-06:
Great!, @ellensp I have the current sensor, let me have test it.

ellensp, 2020-07-06:
issue with eeprom fixed, I didn't know Power monitor saved any setting so missed it.

closed 2020-07-06 by thinkyhead