MarlinFirmware/Marlin #18548: BTT SKR 1.4 + TMC2209 Extruder moves with Z axis

K-morpse, 2020-07-05:
hey lads.

I was strugling with the sensorless homeing, etc and finally i managed to solve it, but now my lovely extruder started a syncronous movement with the Z axis. I can only move the extruder when i move the Z axis. Any idea what am i did wrong? My actual firmware is attached to this post. Delete the . zip and add .cur to check it. Thanks for the help.

K
[FIRMWARE.zip](https://github.com/MarlinFirmware/Marlin/files/4874760/FIRMWARE.zip)

thisiskeithb, 2020-07-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-05 by thisiskeithb

K-morpse, 2020-07-05:
Ah, ok. Thanks for the help.