MarlinFirmware/Marlin #18551: [BUG] G76 P probe temperature calibration stalls

tanaes, 2020-07-05:
### Bug Description

I recently added a thermistor to my inductive Z probe to see if I could take advantage of temperature compensation using G76. I was able to perform the bed calibration, but upon moving the probe calibration the printer got the bed up to temperature and then stalled. It appears to not be reaching the following step in the description: 

```
Move the probe to the probing point and lower to just 1mm above the bed.
```

The nozzle moves to the cooldown point, the bed reaches temperature (in my case, 100°), and then it just waits. Here is a sample of my temperature pings from the terminal:

```
Recv: Waiting for bed and probe temperature.
Recv:  T:26.59 /0.00 B:27.07 /100.00 P:27.59 /0.00 @:0 B@:0 T:26.48 /0.00 B:27.19 /100.00 P:27.61 /0.00 @:0 B@:127
Recv: echo:busy: processing
Recv:  T:26.48 /0.00 B:27.24 /100.00 P:27.61 /0.00 @:0 B@:127 T:26.56 /0.00 B:27.19 /100.00 P:27.70 /0.00 @:0 B@:127 T:26.65 /0.00 B:27.36 /100.00 P:27.70 /0.00 @:0 B@:127
...
Recv: echo:busy: processing
Recv:  T:27.61 /0.00 B:100.00 /100.00 P:29.69 /0.00 @:0 B@:17 T:27.64 /0.00 B:99.97 /100.00 P:29.69 /0.00 @:0 B@:18 T:27.50 /0.00 B:100.04 /100.00 P:29.43 /0.00 @:0 B@:16
Recv: echo:busy: processing
Recv:  T:27.56 /0.00 B:100.04 /100.00 P:29.63 /0.00 @:0 B@:16 T:27.64 /0.00 B:100.02 /100.00 P:29.63 /0.00 @:0 B@:17 T:27.70 /0.00 B:100.01 /100.00 P:29.60 /0.00 @:0 B@:17
```

I also tried warming the probe over 30° by hand to see if that would induce the following step, but nothing happened.

```
Recv: echo:busy: processing
Recv:  T:28.01 /0.00 B:100.00 /100.00 P:30.70 /0.00 @:0 B@:15 T:27.90 /0.00 B:100.04 /100.00 P:30.44 /0.00 @:0 B@:14 T:27.95 /0.00 B:100.02 /100.00 P:30.44 /0.00 @:0 B@:14
Recv: echo:busy: processing
Recv:  T:27.87 /0.00 B:100.02 /100.00 P:30.55 /0.00 @:0 B@:15 T:27.84 /0.00 B:100.02 /100.00 P:30.57 /0.00 @:0 B@:14 T:27.90 /0.00 B:100.03 /100.00 P:30.52 /0.00 @:0 B@:14
```

I initially encountered this using a modification of 3DPrintBeginner's 2.0.5.x firmware, but to be sure, I just forked the repository, updated my configs, and tried again, with the same results. 

### My Configurations

Printer: Artillery Genius, stock board (MKS Gen L) with BTT TFT2.4 and BMG extruder

Configurations available in the `Genius` branch of my forked repo, [here](https://github.com/tanaes/Marlin/tree/Genius). 

### Steps to Reproduce

1. Install inductive Z probe with thermistor
2. Send G76 P

**Expected behavior:** [What you expect to happen]

- Move the probe to the cooldown point.
- Heat up the bed to maximum temperature (e.g., 110°C).
- Move the probe to the probing point and lower to just 1mm above the bed.
- Wait until the probe heats up to the target (30°C).
- Probe the bed to get a base value.

**Actual behavior:** [What actually happens]

- Move the probe to the cooldown point.
- Heat up the bed to maximum temperature (e.g., 110°C).
- Continues waiting

thisiskeithb, 2020-07-05:
There have been some `G76` fixes in `bugfix-2.0.x` since `2.0.5.3` was released, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, I ran `G76` on my MK3S/(genuine) PINDA2 build last week and it was working fine on the latest `bugfix-2.0.x`.

tanaes, 2020-07-05:
Ah my bad, I thought I'd based off the bugfix branch but I hadn't! Pulled in the most recent bugfix into my branch and it is now working as expected. 

Thanks!!!

closed 2020-07-05 by tanaes