MarlinFirmware/Marlin #18553: Endstop should not longer ignored, maybe a parameter, so physical tiggerd endstops allways stop a move

Oldie-driver, 2020-07-05:
Endstop is ignored on probe. But why this stupid way?

Every endstop is the very last saftey line on any CNC maschine. If it „triggered, if mean STOP NOW! And not probe is not tiggerd, so let us move with full power on the wall becouse i am stoned and the sun is shining. Like just happens.

I have a Z min endstop, and at a other pin a BL Touch. BL Touch just faild - shit happens, Z-Min was tiggerd but ignored?

M119 says 

20:45:06.189 : Reporting endstop status
20:45:06.189 : x_min: open
20:45:06.189 : y_min: open
20:45:06.189 : z_min: TRIGGERED
20:45:06.189 : z_probe: open

Why Marlin think all is fine an move Z ahead down ? Any part can be failed, if one part say stop, special a endstop, the only safety way is stop and let check the user, and not a from perferct would dreaming firmware.

This problem is old and not only in Marlin. But is more a bug then a feature. If the endstop really tigger to early, is a problem of the user, not a decision of Firmware self, that's is easyer to ignore. In this last case can any safety parts be removed, even useless breaks from a selfdriving car.

thisiskeithb, 2020-07-05:
Enable `ENDSTOPS_ALWAYS_ON_DEFAULT` if you want your endstops to always be enabled and not just when homing.

Oldie-driver, 2020-07-05:
Thanks My fault, i should post my config too.

i have 
#define ENDSTOPS_ALWAYS_ON_DEFAULT
in configuration_adv.h.

This  shit was happen on homing with BL Touch. BL Touch is on Z-Max Pin, on Z-Min is a optical sensor, few mm below the bed level.
I expected, it is a second line, in case of failed of the Touch. But...now i need replace some parts.

I was hope that endstops allways respected and stop the move on it's or all achse. Yes i am a little old style, so i place even with a probe a endstop. In my eyes, this are not the same, even this way is done mostly at home.

Maybe the only work around is a extra hardware, witch after tigger of a endstop power off the maschine. 
But normaly, a firmware should not never ignore a endstop. A little bigger miller can make same damage.

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/4875491/Marlin-config.zip)

thisiskeithb, 2020-07-05:
Your config is from `2.0.5.x`, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Oldie-driver, 2020-07-05:
I will do now, maybe need some min. I thing, i have use bugfix version, but i will check with a new fresh build

Oldie-driver, 2020-07-05:
i see, now version is 020006. I need some time for the config files....

Oldie-driver, 2020-07-05:
Ok, build is done. Near same option like the build before, i remove only ABL. Some compiler error, i look for part this later.

As test, i home the printer, move Z40 mm higher and put fast a bolt in the light-sensitive barrier, which is my z-min switch.

On PC, M119 say z_min TRIGGERED and printer continues move to Z40 without any stop. Well, some people may say, it move to max, so ignoring z_min is ok in this case.

After it reached Z40, I remove bolt, check M119, z_min is open. Put bolt again,  z_min is TRIGGERED.

Then i say Z-HOME and it try this, ignoring the TRIGGERED z_min, just move Z down. And this is not ok.

If you want the current config, i will post it.

ellensp, 2020-07-06:
endstops only ever look at the 1 pins its meant to and ignores all others.
you have USE_PROBE_FOR_Z_HOMING
so it is only looking at the probing pin.

thisiskeithb, 2020-07-06:
@Oldie-driver: Please confirm that commenting out `USE_PROBE_FOR_Z_HOMING` solves your issue.

Oldie-driver, 2020-07-06:
Sorry, i need to work too ;D

I have check this, is more a very dirty work a round, not the real solution or fix.
In this case, it use just the end stop switch for homing.

But end stop should be stopped always, not matter is probe or on homing or normal move. Yes in home self build 
usually try to save money and use one switch for homing and end stop, but this is not a real use and meaning of a end stop.

If Probe not failed with broken wire like yesterday, Probe work for home too.

There was a different view between hobby and more like job. It have not high priority, course you can work so with Marlin. But keep as low priority bug open, until the end stop work as real end stop. This mean always stop on trigger moving in this direction, even on homing with BL Touch or make a bed level or what ever. I think this is the real bug, the sight it work, why should make the work that this work right.

I will better implement soon all end stops in hardware so that they are no longer ignored. At the moment, this may be ok with a  3D printer, but I don't have a good feeling about the laser or the milling machine, if i was sleeping and know this are only fake end stops.

thisiskeithb, 2020-07-06:
As noted above, you have `USE_PROBE_FOR_Z_HOMING` in your config, so Marlin is expecting you to home with a probe, not an endstop for Z.

GMagician, 2020-07-06:
@thisiskeithb if think his note is about what endstops means in real world machine. On a real world machine endstops means...you can't pass it due to mechanical limits reached. Usually you may have two sets ot them. Last is connected to security device that will put system in emergency, first is just to stop axis and let you recover, these limit switch may also be used as home sensors.

Now, just to be complete, Marlin doesn't work this way

thisiskeithb, 2020-07-06:
I understand the differences, but you're explicitly overriding default behavior in Marlin.

Please comment out`USE_PROBE_FOR_Z_HOMING` and try to home Z again.

GMagician, 2020-07-06:
@thisiskeithb 

my 

> Now, just to be complete, Marlin doesn't work this way

was just to tell that I'm aware Marlin doesn't work this way, not that it has to. Don't want to be misunderstood for that, translation to a different language often lead to write something different from what one think to write

ellensp, 2020-07-06:
Only way I know to get both to work is setup the probe as normally closed and the endstop as normally closed and put them in series. Then when either of them get opened it triggers the io line you have it connected.

Oldie-driver, 2020-07-06:
Thanks, sound this is a very good idee, i will test this after work. And maybe a update in the documentation, that Marlin only supports home sensor and no real endstop can avoid some confusion. In real world is a big difference between them. Mit freundlichem Gruße,Andreas Fincke-------- Ursprüngliche Nachricht --------Von: ellensp <notifications@github.com>Datum: Mo., 6. Juli 2020, 14:52An: MarlinFirmware/Marlin <Marlin@noreply.github.com>Cc: Oldie-driver <andreas@fincke-privat.de>, Mention <mention@noreply.github.com>Betreff: Re: [MarlinFirmware/Marlin] Endstop should not longer ignored, maybe a parameter, so physical tiggerd endstops allways stop a move (#18553)
Only way I know to get both to work is setup the probe as normally closed and the endstop as normally closed and put them in series. Then when either of them get opened it triggers the io line you have it connected.

AnHardt, 2020-07-06:
```
// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
```
also helps to have the harware endstopt on a bit more often.

But that has its own problems. Since the software endstops positions are set with triggered home-switches often the hardware endstops trigger before the software endstops can prevent it.

Real security endstops act on hardware - cutting the step- or enable- lines. Recovery is then done with a button rebridging the opened line. Breaking the power-lines makes recovery a bit more difficult at least you have to think about, what power you cut.

Oldie-driver, 2020-07-06:
Yes power off are more complicated on resume. 

But o think is better then be blind course a laser in your eyes or a broken miller. 
Endstops shall never be trigger in normal world. If this happens, stop is the only safety option.

sjasonsmith, 2020-07-06:
CNC machines I use configure software endstops to ensure they trigger before the mechanical end stop. That prevents ever hitting physical endstops after homing. Of course that can still fail due to lost steps with open-loop steppers.

Oldie-driver, 2020-07-06:
I think about just cutting enable line and maybe power of spindle or laser.

It will not really help more on continuing then just power off. Marlin need to know this stop, else it thinks work is still running. So resume is not really possible, course Marlin know at this time nothing about the real position and with command are done. One more reason, why marlin should respect safety switch in realtime. Ok, is not. 

Course the Maschine are 50 meters away, i use maybe for all maschine repetier server. It have a option to try a resume. Cutting the main power will see as failed from server, an it can try to resend the unacknowledged comands. Not always, but mostly this works fine.

thisiskeithb, 2020-07-07:
@Oldie-driver: Marlin "endstop" semantics aside, does disabling `USE_PROBE_FOR_Z_HOMING` work?

Oldie-driver, 2020-07-07:
Not really, in this case Marlin just use the z-min probe for homing. More troube with the bed levling. 

But yes it can home with z-min, if you mean this for just closing.

At this days and in near future, in Marlin alone is a real solution not possible. Maybe if i hack the probe routines. I work on this, but need more time and work for understand every step.

Marlin and it's auto bed level feature can be very usefull on milling not plane objects. But not with ignoring endstops, even only at probe.

If you like later the code, is ok, if not, i need them self.

closed 2020-07-07 by Oldie-driver