MarlinFirmware/Marlin #18555: [BUG] ABL z-offset sporradic compensation (exact same issue as #17626)

txt4nk, 2020-07-05:
Ender 5  Pro
E3D Hemera
SKR E3 Mini V1.2
BLTouch V3.1
Spool3D.ca cast aluminum bed
Raspberry Pi 4

### Bug Description

I am suffering from the exact same issues as the OP in #17626. I think it was around March when my SKR Mini E3 V1.1 board fried one of the TMC drivers and ended up with a TMC connection error. Before this error, my printer was printing the most perfect prints I've ever gotten from any printer. I replaced the board with an SKR Mini DIP V1.1 w/ 2208 Drivers in UART mode and this is where my issue started. I thought it may be a bad DIP board so I bought another, and also got my E3 Mini replaced with a V1.2. Both of these boards both exhibit the problem, so I purchased a TH3D Ezboard and it too has the same issue. I have tried so many different firmware changes and nothing seems to make it any better. I have tried my OEM heat bed (which was just fine before replacing it with the Spool3D piece), I tried a brand new OEM heat bed, and even another Spool3D bed. All 4 of them show exactly same "pattern" and exhibit the exact same printing issues. I have rebuilt my printer frame from ground up, making sure everything is perfectly square. The issue still exists. I have tried removing the BLtouch, reconnecting my end stops and running the printer with no ABL and it prints perfectly fine. I then re-installed the BL touch, and went UBL instead of Bilinear. I get the exact same results, the only way to get it to print is to manually edit\create the grid with the good ole paper between the nozzle and bed gig.

As far as firmware changes.. here's all of the things I've tried (on their own and also in conjunction with eachother):
Bed Leveling After G28: Always kept on
Grid Max Points: I've tried anywhere from 3 to 15.
Multiple Probing: Always set at 2, tried 3 and got no change.
Extrapolate Beyond Grid: Tried it on and off.
Bilinear Subdivisions: Tried it on and off.

M48 results: They range between 0.0018 to 0.0020.

Bed visualizer looks like this on all 4 beds I've tried, and all firmware configurations. 
[image]

### My Configurations

[CE5P_Hemera_BLTouch_E3 Mini.zip](https://github.com/MarlinFirmware/Marlin/files/4875692/CE5P_Hemera_BLTouch_E3.Mini.zip)

### Steps to Reproduce

1. G28
2. G29
3. Start Print

**Expected behavior:** [What you expect to happen]

Epic first layers and the ability to start a print without having to cancel it every single time.

**Actual behavior:** [What actually happens]

Pretty sure I outlined it pretty well in the description above.. but if I run bilinear or UBL without modifying the mesh, I get this.. which happens to look pretty identical to the crazy pattern on the bed visualizer.
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-07-05:
Duplicate of #17262

closed 2020-07-05 by sjasonsmith

txt4nk, 2020-07-05:
The whole reason I made this is because no one is seeming to respond to me in 17262. How am I supposed to help you guys track down a bug if that thread is getting threatened of closure because the OP hasn't responded, yet I am trying to do anything I can to help and its getting overlooked and ignored.

sjasonsmith, 2020-07-05:
There was some confusion as new processes have been adopted to deal with abandoned issues. This issue clearly should not have been flagged. In the future if something is marked stale, _anybody_ should be able to reply to keep it open.

I understand that it seems that issue is being ignored, but adding more issues won't help. When there are multiple issues for the same problem is is hard to focus attention on something to figure it out.

The bed leveling issues in particular are hard to deal with, because there are many variables impacting it and what fails for one person is totally fine for somebody else. It is entirely likely that actually nobody is working on fixing it, because the right person with the right hardware with the right motivation hasn't happened to look at it yet.

Part of the motivation for closing out some stale issues (which again, this should not have been included) is that it helps highlight the remaining issues (such as this one) so they get more attention.

txt4nk, 2020-07-05:
With the amount of ABL issues that have come up here, you'd think people would realize its more than an isolated issue by now. I am not adding "more issues", I am simply trying to add "more data" so if anyone feels like trying to help, they've got as much data as they could possibly need to attempt a fix, if that ever happens.

sjasonsmith, 2020-07-05:
I appreciate you adding data, but you already added it in the other report. By your own admission this is exactly the same issue.
If there is information in this one that is not in the other, please include it over there.

I'm not trying to be rude in any way. Simply trying to avoid noise in the issue queue so that it is easier for people to see how many issues we are facing. When things are truly the same problem, they will be consolidated into one issue.

txt4nk, 2020-07-05:
I too am not trying to be rude, but 17262 is probably the 5th or 6th thread that I've tried to add my data to and have gotten zero response. At this point I am just highly frustrated, I apologize if I have come off as rude but I have done so much, spent so much money on different parts and spent SO many hours trying to self diagnose this and at this point we are at the mercy of someone who is more well versed in the coding of Marlin.

sjasonsmith, 2020-07-05:
Keep in mind that almost everybody contributing does so on their own time without pay. It isn't reasonable to expect that somebody is just going to solve everything immediately. People tend to focus on the issues most impacting themselves, and it takes time to find the right person to help with an issue. In many cases people end up deciding to dig in and figure out how to fix things themselves, then go on to help others since they have now become "someone who is more well versed in the coding of Marlin".

Even with people who do know their way around the code, I don't know if _anybody_ knows their way around _all_ of the code. People tend to have their areas they specialize in, and bed leveling happens to be one of the more complex areas that fewer people are equipped to contribute heavily toward.

txt4nk, 2020-07-05:
Trust me, I never expected immediate results. This has been happening for months, I have posted about this since the very beginning on other people's reports because I was trying not to clutter things up and have information all in one place. Tis why I've gotten to the point of frustration, and ultimately opening this bug report.

Can you tell me one thing, is there a way to get past the sanity check that forces you to use Z safe homing? I've tried to get past it and have failed. This is the only thing I've not been able to do in my own testing. The reason I want to try it without Z safe homing is because I don't remember enabling it late last year when my printer was working great.. then when my TMC driver failed and I compiled new firmware for the DIP board, the sanity check prevented me from keeping it disabled while I had an ABL probe uncommented.

sjasonsmith, 2020-07-05:
> Can you tell me one thing, is there a way to get past the sanity check that forces you to use Z safe homing?

You can probably just comment it out.

thinkyhead, 2020-07-15:
When one side of the bed is always higher/lower than the other, it is a classic symptom of a twisted X gantry or other mechanical problem pulling the probe out of alignment and causing unusable probe values. There is nothing Marlin can do about this. You must ensure that the X gantry is straight and the probe remains perfectly perpendicular to the bed at all points. Wires or filament may also pull the X-carriage out of alignment, so ensure the wires are loose and there is no strain at the far end of the travel.

txt4nk, 2020-07-15:
I've re-built this machine using a ruler to make sure that the frame and gantry are 100% true. I've made sure that the wires have plenty of slack and at no point put any additional stress on the hot end\x gantry. Like I said, this thing was running like a champ and then I got the TMC connection error on the SKR E3 Mini (common problem). Replaced the board with a DIP and this issue started. I've since tried a 2nd DIP board with 2208's (first had 2209's), a TH3D ezboard, and now on an E3 Mini V1.2. Without using ABL, and going back to the X axis endstop mechanical switch, I get amazing prints. If the X gantry was not true, this wouldn't be the case. I go back to ABL and prints just won't work, the bed visualizer looks like a mountain range when I've measured the bed with an indicator and the bed deviates at most 0.002% (I've sanded this bed down on a granite surface to get an absolutely flat bed..)

txt4nk, 2020-07-18:
testing it out now. as far as it being a probe problem, this is also my third authentic bltouch. Had the original, tried one that I had sitting around for another build, and ordered a third just in case the two I have are from a bad batch (ordered at the same time). All 3 have this issue.

txt4nk, 2020-07-18:
Same issue.. Tried printing a 80mm square and the right side of the bed sticks beautifully but the left side is too far from the bed. I've leveled the crap out of it manually, even switched back over to glass and its the same outcome.

txt4nk, 2020-07-18:
as far as it being a motion issue, (this goes for both printers..)
1. Belts a perfectly adjusted. Not too loose, not too tight.
2. Eccentric nuts are all spot on. There's no binding, no unnecessary movement, moves freely with the steppers off, no flat spots on the wheels.
3. Frames are perfectly square, I've rebuilt them using a perfectly flat desk and a ruler to verify there are no delta's from corner to corner\side to side.

txt4nk, 2020-07-18:
Alright I'm making some progress.. I set the grid to 3x3 (i normally use 5x5) and it still had inconsistent readings, so then just as a shot in the dark, I did 5 multiple probings and its coming out MUCH cleaner on both printers.. I haven't tried a print yet, about to start one on the ender 5 (the one that's given me the most grief)

txt4nk, 2020-07-18:
well thats a bust.. with both the 3 and 5 set to 3x3 it showed a decent output on bed visualizer but printing just was not happening.. i've set both of them back to 5x5 and the mountains are back. I'm stumped and frustrated at this point, I have no clue what else to try.

thinkyhead, 2020-07-18:
I dunno what to tell you. AFAIK `spreadCycle` is turned on automatically during probing to provide enough torque for Z. When I was having Z axis slippage issues many moons ago I made a G-code that moved Z up and down rapidly, but gradually moving towards the top. I would measure the distance traveled to make sure it matched what I wanted.

txt4nk, 2020-07-18:
Interesting.. ill see if I can make a gcode like that and see if it yields any results