MarlinFirmware/Marlin #18556: [BUG]  E?_AUTO_FAN_PIN defines cannot share FAN1_PIN, compile hangs in temperature.cpp

allartburns, 2020-07-05:
### Bug Description

I have a dual-extruder RUMBA that only has one fan pin for extruder fans.   That is, E0 and E1 each have their own fan but there's only one fan control pin on the RUMBA.  In 1.0.6 we just turned both fans on at the same time no matter which extruder was being used.

In Configuration_adv.h I set:

#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN FAN1_PIN

The result is that the compiler hangs in Temperature::checkExtruderAutoFans()
[fan-bug.zip](https://github.com/MarlinFirmware/Marlin/files/4875698/fan-bug.zip)

### My Configurations

attached
[fan-bug.zip](https://github.com/MarlinFirmware/Marlin/files/4875704/fan-bug.zip)

### Steps to Reproduce

Use these config files and compile.  I've included a script that compiles only temperature.cpp, this is how I found where it was hanging.

**Expected behavior:** 

Code compiles

**Actual behavior:** [What actually happens]

Build hangs.

#### Additional Information

I'm setting up the E3D BigBox, a dual-extruder capable printer that uses (what I think is) a EU variant of the RUMBA+  https://www.wiki.aus3d.com.au/images/0/06/RUMBAPinout-v14.png

ellensp, 2020-07-06:
You can't do this.
Eg if E0 was hot and E1 was cold 
Every loop marlin would look at the thermistor on E0 and and turn on the fan, then it would check E1 and turn the fan off.

ellensp, 2020-07-06:
This compiles without issues using bugfix on Linux.

allartburns, 2020-07-06:
> 
> 
> You can't do this.
> Eg if E0 was hot and E1 was cold
> Every loop marlin would look at the thermistor on E0 and and turn on the fan, then it would check E1 and turn the fan off.

Well, it worked just fine in 1.0.6.  If either thermistor is hot then turn on the cooling fans on that pin.

allartburns, 2020-07-06:
> 
> 
> This compiles without issues using bugfix on Linux.

Please cite the branch you're using.  I've tried on Mac and Linux and both fail to compile.

ellensp, 2020-07-06:
Current latest bugfix 
last commit 62c00aac2c7c28c395c7eb9aae15cc56fc9a9eea (HEAD -> bugfix-2.0.x, origin/bugfix-2.0.x)

ellensp, 2020-07-06:
himm the comments do say "Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold. "  ill do some testing.

ellensp, 2020-07-06:
Apologizes, this does work as you expect..  If either E0 or E1 > than defined temperature the fan turns on.

allartburns, 2020-07-08:
Ok, got it to compile against bugfix.  I was building against 2.0.x earlier.   It still takes waaay too long in temperature.cpp but the binary installs and I'm testing.

closed 2020-07-08 by allartburns

allartburns, 2020-07-08:
Note: the fans still don't work, this just fixes the "won't compile" bug.

thinkyhead, 2020-07-15:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Don't forget to reset and save to EEPROM with `M502`, `M500` after flashing the new firmware.

allartburns, 2020-07-16:
This bug was against the compilation failure, I think we can close it as I can compile against bugfix-2.0.x.

cbpdk, 2020-07-31:
I have experienced a similar problem:

Trying to compile Marlin 2.0.5.4 for Ramps 1.4 using Arduino IDE 1.8.13 , I am having the same problem using the same pin for both E0_AUTO_FAN_PIN and E1_AUTO_FAN_PIN.

However, if the thread processes compiling temperature.cpp are killed, the compiling and linking continues, ending with no error messages. I have not tested if the result works.

Trying Marlin 2.0.6, the same problem shows up.
Marlin-bugfix-2.0.x at first ended in an error due to include of cstddef in tft_image,cpp.
After removing the involved files, it also hangs up trying to compile temperature.cpp.

cbpdk, 2020-08-02:
As @allartburns mentioned that the compiler hangs in Temperature::checkExtruderAutoFans(), I had a look and decided to replace the code lines in 2.0.5.4:

` #define _EFAN(B,A) _EFANOVERLAP(A,B) ? B :
    static const uint8_t fanBit[] PROGMEM = {
      0
      #if HOTENDS > 1
        #define _NEXT_FAN(N) , REPEAT2(N,_EFAN,N) N
        RREPEAT_S(1, HOTENDS, _NEXT_FAN)
      #endif
      #if HAS_AUTO_CHAMBER_FAN
        #define _CFAN(B) _FANOVERLAP(CHAMBER,B) ? B :
        , REPEAT(HOTENDS,_CFAN) (HOTENDS)
      #endif
    };
`

with code from 1.1.9.1:

`
    #define AUTO_1_IS_0 (E1_AUTO_FAN_PIN == E0_AUTO_FAN_PIN)
    #define AUTO_2_IS_0 (E2_AUTO_FAN_PIN == E0_AUTO_FAN_PIN)
    #define AUTO_2_IS_1 (E2_AUTO_FAN_PIN == E1_AUTO_FAN_PIN)
    #define AUTO_3_IS_0 (E3_AUTO_FAN_PIN == E0_AUTO_FAN_PIN)
    #define AUTO_3_IS_1 (E3_AUTO_FAN_PIN == E1_AUTO_FAN_PIN)
    #define AUTO_3_IS_2 (E3_AUTO_FAN_PIN == E2_AUTO_FAN_PIN)
    #define AUTO_4_IS_0 (E4_AUTO_FAN_PIN == E0_AUTO_FAN_PIN)
    #define AUTO_4_IS_1 (E4_AUTO_FAN_PIN == E1_AUTO_FAN_PIN)
    #define AUTO_4_IS_2 (E4_AUTO_FAN_PIN == E2_AUTO_FAN_PIN)
    #define AUTO_4_IS_3 (E4_AUTO_FAN_PIN == E3_AUTO_FAN_PIN)
    #define AUTO_CHAMBER_IS_0 (CHAMBER_AUTO_FAN_PIN == E0_AUTO_FAN_PIN)
    #define AUTO_CHAMBER_IS_1 (CHAMBER_AUTO_FAN_PIN == E1_AUTO_FAN_PIN)
    #define AUTO_CHAMBER_IS_2 (CHAMBER_AUTO_FAN_PIN == E2_AUTO_FAN_PIN)
    #define AUTO_CHAMBER_IS_3 (CHAMBER_AUTO_FAN_PIN == E3_AUTO_FAN_PIN)
    #define AUTO_CHAMBER_IS_4 (CHAMBER_AUTO_FAN_PIN == E4_AUTO_FAN_PIN)

    static const uint8_t fanBit[] PROGMEM = {
      0,
      AUTO_1_IS_0 ? 0 :               1,
      AUTO_2_IS_0 ? 0 : AUTO_2_IS_1 ? 1 :               2,
      AUTO_3_IS_0 ? 0 : AUTO_3_IS_1 ? 1 : AUTO_3_IS_2 ? 2 :               3,
      AUTO_4_IS_0 ? 0 : AUTO_4_IS_1 ? 1 : AUTO_4_IS_2 ? 2 : AUTO_4_IS_3 ? 3 : 4,
      AUTO_CHAMBER_IS_0 ? 0 : AUTO_CHAMBER_IS_1 ? 1 : AUTO_CHAMBER_IS_2 ? 2 : AUTO_CHAMBER_IS_3 ? 3 : AUTO_CHAMBER_IS_4 ? 4 : 5
    };
`

Compiled and tested only that the fans turned on/off when passing 50 degrees C.
The compiler used by the Arduino IDE is 7.3.0-atmel3.6.1-arduino7, OS is Kubuntu 18.04.
I am not able to figure out why the macros fails.