MarlinFirmware/Marlin #18559: [BUG] Thermal runaway detected while using #1047 & #1010 & #1000 temp sensors

setar, 2020-07-06:
Bug Description

Thermal runaway is detected at startup and stops Marlin.

My Configurations

[setar_conf_bf.20x.zip](https://github.com/MarlinFirmware/Marlin/files/4877415/setar_conf_bf.20x.zip)

Steps to Reproduce

Download Marlin bugfix 2.x after c43bbcce152b4d3b4697e1e188d3bcada76a24a7
define TEMP_SENSOR_0 1047 , 1010 or 1000
Compile and install
Restart
Expected behavior: Normal Marlin startup

Actual behavior: Thermal runaway detection "MAX TEMP on E1" or something like that.

Additional Information

in termistors.h type must be int16_t, but it not all , need additional search and debug

#define PtAdVal(T,R0,Rup) (short)(1024 / (Rup / PtRt(T, R0) + 1))

sjasonsmith, 2020-07-06:
@setar did this work for you in older Marlin versions? Assuming so, do you know what the last version was that it worked for you in?

setar, 2020-07-06:
today i work on 70fa4c9323543dfb65d10969838899575406cae6 with rebase my hardware conf.

ellensp, 2020-07-07:
Definitely effects 
*  1047 : Pt1000 with 4k7 pullup (E3D)
*  1010 : Pt1000 with 1k pullup (non standard)
*   147 : Pt100 with 4k7 pullup
*   110 : Pt100 with 1k pullup (non standard)
*  1000 : Custom - Specify parameters in Configuration_adv.h

ellensp, 2020-07-07:
@setar I don't have these thermistor types.   but to me it look like they are PTC not NTC 
But marlin hasn't been told this.

Can you try something for me?

To tell marlin you have a PTC you add the line #define REVERSE_TEMP_SENSOR_RANGE_NNNN 1 to the thermistor table.

Eg for table 1047 
add the line `#define REVERSE_TEMP_SENSOR_RANGE_1047 1`  to the file Marlin/src/module/thermistor/thermistor_1047.h
after the "#pragma once" line.

Eg for table 147 
add the line `#define REVERSE_TEMP_SENSOR_RANGE_147 1`  to the file Marlin/src/module/thermistor/thermistor_147.h
after the "#pragma once" line.

Does this now read correctly and behave as it should?

setar, 2020-07-07:
change need to be applied to current version bugfix-2.0.x ? 
e.g  588d40aedf4d0f148186280cf6f387f5436302c2
I will check it after 6-7 hours

ellensp, 2020-07-07:
Yes a current bugfix

heppstar, 2020-07-07:
Hi,

I have the same problem. You also get it with "#define TEMP_SENSOR_0 999" that I use for my CNC-machine. Can not start it now.

Regards,
heppstar

ellensp, 2020-07-07:
@heppstar Please don't hijack this issue. Issues relating to  thermistor type 999 and 998 have already been reported and fixed in bugfix see https://github.com/MarlinFirmware/Marlin/issues/18525

setar, 2020-07-07:
> #define REVERSE_TEMP_SENSOR_RANGE_1047 1

the fix work fine: 
Normal Marlin startup complete,
and I checked the temperature with an external thermocouple on the multimeter at 25, 50, 100, 150, 250 C - everything is fine, the difference is <1C

ellensp, 2020-07-08:
This is now fixed and merged into latest bugfix

closed 2020-07-08 by ellensp