MarlinFirmware/Marlin #18560: X SHIFTING WHEN ENABLING # linear advance IN 2.0.5.3  ( i am using closed loop servo )

KARTHICKROBO, 2020-07-06:
i am using  modified closed loop  3d  printer with 2 nm servo stepper  mks gen l board 

no way i loss steps  

### Bug Description

but when i enable  ( linear advance )  only x axis shifts 

[image]
[image]

but print quality is great !

only x shits about 30 deg angle  y axis is good 

k = 0.6

if i reduced k value to k=0.4

it x shifts opposite direction 

### Steps to Reproduce

1.    enable linear advance and upload to control board ( my case closed loop servo )  but i think its not a issue  i am using ldca2575   servo driver 
2.    k = 0.6   or 0.4 (low shifting compared to 0.6)  and start xyz calibration print 

**Expected behavior:**    smooth print with out any artifacts   

**Actual behavior:**   only x axis shifts

ellensp, 2020-07-06:
Please attach your configuration files. might be something else in there that is a contributor

ellensp, 2020-07-06:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

KARTHICKROBO, 2020-07-06:
I ATTACHED CONFIG FILE

KARTHICKROBO, 2020-07-06:
error message   

(old and other version compiling good )  

but  this one not ??

avr-gcc: error: CreateProcess: No such file or directory

Multiple libraries were found for "U8glib.h"
 Used: C:\Users\robo\Documents\Arduino\libraries\U8glib
 Not used: C:\Program
Using library U8glib at version 1.18 in folder: C:\Users\robo\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

KARTHICKROBO, 2020-07-06:
full error report 

Arduino: 1.8.11 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\robo\AppData\Local\Arduino15\packages -hardware C:\Users\robo\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\robo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\robo\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10811 -build-path C:\Users\robo\AppData\Local\Temp\arduino_build_705505 -warnings=none -build-cache C:\Users\robo\AppData\Local\Temp\arduino_cache_908244 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\robo\Downloads\Marlin-bugfix-2.0.x (2)\Marlin-bugfix-2.0.x\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\robo\AppData\Local\Arduino15\packages -hardware C:\Users\robo\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\robo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\robo\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10811 -build-path C:\Users\robo\AppData\Local\Temp\arduino_build_705505 -warnings=none -build-cache C:\Users\robo\AppData\Local\Temp\arduino_cache_908244 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\robo\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\robo\Downloads\Marlin-bugfix-2.0.x (2)\Marlin-bugfix-2.0.x\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Users\robo\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\robo\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\robo\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10811 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\robo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\robo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\robo\\AppData\\Local\\Temp\\arduino_build_705505\\sketch\\Marlin.ino.cpp" -o nul
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\HAL.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\HAL_SPI.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\MarlinSerial.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\Servo.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\eeprom.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\fast_pwm.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\fastio.cpp
Using cached library dependencies for file: C:\Users\robo\AppData\Local\Temp\arduino_build_705505\sketch\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp
Alternatives for U8glib.h: [U8glib_Arduino-master@1.19.1 U8glib@1.18]
ResolveLibrary(U8glib.h)
  -> candidates: [U8glib_Arduino-master@1.19.1 U8glib@1.18]
[... cut ...] ketch\\src\\HAL\\LPC1768\\timers.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\LCD_I2C_routines.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\u8g\\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "sketch\\src\\HAL\\LPC1768\\usb_serial.cpp.o" "ske

KARTHICKROBO, 2020-07-06:
solved  working fine now  in bugfix 2.0.x 

thanks it solved now but too slow it  machine capable of more speed 

 can we do all calculation in   (  slicer )  instead of on printer controller  it self we can save time and money ?

ellensp, 2020-07-06:
Your running an obsolete 8 bit controller. You can get Get a nice new 32bit skr 1.4 for around $20usd + drivers.

closed 2020-07-06 by ellensp

ellensp, 2020-07-06:
For non marlin firmware issues please use one of the following resources.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

KARTHICKROBO, 2020-07-06:
OK THANKS