MarlinFirmware/Marlin #18569: [BUG] FILAMENT_CHANGE_UNLOAD_LENGTH setting is ignored by Marlin

KL-XL, 2020-07-07:
### Bug Description

Using BTT FRS 1.2. Changing values for FILAMENT_CHANGE_UNLOAD_LENGTH and FILAMENT_CHANGE_FAST_LOAD_LENGTH has no effect. When filament runout event is triggered, it still uses the default lengths.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4881923/Config.zip)

### Steps to Reproduce

1. Change FILAMENT_CHANGE_UNLOAD_LENGTH from 400 to 0
2. Change FILAMENT_CHANGE_FAST_LOAD_LENGTH from 350 to 50
3. Build
4. Flash firmware
5. Start print
6. Hold filament for the sensor to trigger the runout event and start to unload filament

**Expected behavior:** Filament should not unload and afterwards should purge 50 mm

**Actual behavior:** Filament is unloaded and then loaded for the full length of the Bowden tube (probably 400mm)

#### Additional Information

* This seems to be related to the bug of the firmware flashing not updating the FILAMENT_RUNOUT_DISTANCE_MM setting, but that can at least be set by M412 D?

ManuelMcLure, 2020-07-07:
These settings are stored in EEPROM, so you must execute ``M502`` followed by ``M500`` to get the firmware values to override the EEPROM values.

KL-XL, 2020-07-07:
OK. Thanks. Looking at the result of M503 there is this line: M603 U350 L400 so I presume I can set the values from G-Code, right?

KL-XL, 2020-07-07:
Yes, setting M603 U0 L50 and then M500 does the trick. Thanks

closed 2020-07-07 by KL-XL