MarlinFirmware/Marlin #18572: [BUG] Compiler error on the latest bugfix.

bill-orange, 2020-07-07:
Here's the error from VSCODE:

Compiling .pio\build\LPC1768\src\src\sd\Sd2Card.cpp.o
Marlin\src\module\temperature.cpp: In static member function 'static bool Temperature::wait_for_hotend(uint8_t, bool, bool)':
Marlin\src\module\temperature.cpp:3159:10: error: expected 'while' before 'Temperature'
 3159 |     bool Temperature::wait_for_bed(const bool no_wait_for_cooling/*=true*/
      |          ^~~~~~~~~~~
Marlin\src\module\temperature.cpp:3159:10: error: expected '(' before 'Temperature'
 3159 |     bool Temperature::wait_for_bed(const bool no_wait_for_cooling/*=true*/
      |          ^~~~~~~~~~~
      |          (
Marlin\src\module\temperature.cpp:3159:36: error: expected primary-expression before 'const'
 3159 |     bool Temperature::wait_for_bed(const bool no_wait_for_cooling/*=true*/
      |                                    ^~~~~
Marlin\src\module\temperature.cpp:3161:11: error: expected primary-expression before 'const'
 3161 |         , const bool click_to_cancel/*=false*/
      |           ^~~~~
Marlin\src\module\temperature.cpp:3163:6: error: expected ')' before '{' token
 3163 |     ) {
      |      ^~
      |      )
Marlin\src\module\temperature.cpp:3159:10: note: to match this '('
 3159 |     bool Temperature::wait_for_bed(const bool no_wait_for_cooling/*=true*/
      |          ^~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\src\src\module\temperature.cpp.o] Error 1
====================================== [FAILED] Took 237.94 seconds ======================================

Environment               Status    Duration
------------------------  --------  ------------
LPC1768                   FAILED    00:03:57.943

Here are  my config files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4882068/Marlin.zip)

Yesterday's build was fine.  I tried recompiling after a clean and deleting .pio, no difference.

T-SquaredProductions, 2020-07-07:
I was just about to post that. I had a compiler error too.

Musicdad12, 2020-07-07:
Same Problem too

thisiskeithb, 2020-07-07:
See PR https://github.com/MarlinFirmware/Marlin/pull/18570 for the fix.

bill-orange, 2020-07-07:
Yup, PR #18570 fixed it.

closed 2020-07-07 by bill-orange