MarlinFirmware/Marlin #18573: [BUG] FIL_RUNOUT_STATE HIGH/LOW always detects no filament

Testmangh, 2020-07-07:
### Bug Description
With both values, HIGH and LOW the print starts with a "no filament detection"
```
#define FIL_RUNOUT_STATE     HIGH // Pin state indicating that filament is NOT present.
#define FIL_RUNOUT_STATE     LOW // Pin state indicating that filament is NOT present.
```
The switch that I'm using sends a HIGH signal when filament is not present.

### My Configurations
configuration.h
configuration_adv.h
[https://github.com/Testmangh/Marlin/tree/Ender-3-Pro/Marlin](url)

### Steps to Reproduce
Start a print and it will instantly detect no filament present with both configs HIGH and LOW.

**Expected behavior:** [What you expect to happen]
If there is filament the print should not stop (run M600 in my case).

thisiskeithb, 2020-07-07:
@robbycandra: Do the merged changes from your Filament Runout State PR still work for you?

thisiskeithb, 2020-07-07:
I just tested the latest `bugfix-2.0.x` (ea94716) on my MK3S/BTT002 build and filament runout works as expected.

Testmangh, 2020-07-07:
I tested treee times yesterday and it didn't work but now it is working I may have messed up something, SORRY!
Thanks!

closed 2020-07-07 by Testmangh

robbycandra, 2020-07-11:
@Testmangh 

I found sometimes the limit switch lever arm not aligned properly with the filament itself.  Therefore the detector report false output.