MarlinFirmware/Marlin #18575: Pin In/Out -definition problem with mixed Arduino IDE and PlatformIO IDE

giPeteR, 2020-07-07:
### Pin In/Out -definition problem with mixed Arduino IDE and PlatformIO IDE

### Background: 
Marlin is a mix of Arduino pin definitions (_pins_arduino.h_) and newer PlatformIO pin definitions (_fastio.h, fastio_1280.h_). This causes problems when some functions expect Arduino pins but are fed PIO pins.

### More background:
By pure luck I stumbled into this post (after 2w of struggle) ...
“[Distinguish between analog/digital auto fans](https://github.com/MarlinFirmware/Marlin/pull/13298#issue-257739613)”
And ATmega2560 pin map [here](https://www.arduino.cc/en/Hacking/PinMapping2560).

### Board:
I have modified a **MIGHTYBOARD_REVE** and replaced ATmega1280 with an ATmega2560 to get more memory (for AutoBedLevel, etc). I had then to ditch Sailfish and use Marlin, which I liked a lot.

### Firmware:
**Compiling in Arduino IDE works fine.** Everything works as expected in the printer.
I’m not 100% sure what version compiles in Arduino IDE though. It says:

> Marlin bugfix-2.0.x, 
> Last Updated: 2018-01-20, 
> Compiled: Nov 19 2019
> EEPROM=V76 Marlin=V64
> 

The editor sucks and the compiler is dead slow, so I moved on to **VS Code** and **PlatformIO**.
I have tried:
- Marlin 2.0.x
- Marlin Bugfix-2.0.x (downloaded 2020-07-05)

The result is the same, regardless.

I cannot compile newer Marlins in Arduino IDE.
I cannot compile my older Marlin bugfix-2.0.x in PlatformIO.

### The problem:
The **main problem** is that the use of _digipot_mcp4018.cpp_ **redefines** my other pins. So, my **HEATER_0_PIN** (Arduino pin 6 / H3) becomes an **Input** rather than **Output**.

**Expected behavior:** 
> Send: M43 I P6
> Recv: PIN:   6   Port: H3  HEATER_0_PIN  **Output** = 0 ...

Works with the Arduino IDE.

**Actual behavior:**
> Send: M43 I P6
> Recv: PIN:   6   Port: H3  HEATER_0_PIN  **Input** = 0 ...

Does not work with PlatformIO.

I can’t get a newer Marlin compiled with PlatformIO to work, and I don’t want to use the old Arduino FW (which is probably lightyears behind).

### Root cause:
- _MIGHTYBOARD_REVE_ uses “_Marlin\src\feature\digipot\digipot_mcp4018.cpp_” to set motor currents.

- _digipot_mcp4018.cpp_ uses: 
`#include <SlowSoftI2CMaster.h>  // https://github.com/stawel/SlowSoftI2CMaster`
It is not the Master Branch, because that one lacks Input Pullup resistor settings.
But, PlatformIO loads the Master Branch from GitHub, and not the one suggested in _digipot_mcp4018.cpp_.

The following pins are used by _digipot_mcp4018.cpp_:
```
#define DIGIPOTS_I2C_SCL     76  // J5 (p76 is Arduino pin. Physical is 68.)
#define DIGIPOTS_I2C_SDA_X   57  // F3
#define DIGIPOTS_I2C_SDA_Y   61  // F7
#define DIGIPOTS_I2C_SDA_Z   65  // K3
#define DIGIPOTS_I2C_SDA_E0  27  // A5
#define DIGIPOTS_I2C_SDA_E1  77  // J6
```

If you change pin 76 to anything else, let’s say 77 (same as E1):
`#define DIGIPOTS_I2C_SCL  77  // J6`
Then HEATER_0_PIN (Arduino pin 6 / H3) works as an Output.
I have no idea why pin 76 and pin 6 would interfere.

### My concern here is:
- Does PlatformIO compile the correct source? (Obviously not.)
- How can I make sure the correct pins are used?
- Do I need to probe each and everyone of the 80 pins in order to see they are defined correctly...!? (_No, no, please noooo..._)
- And then pins may still be reassigned later, or maybe used by the wrong subroutines.

### Reproducing:
In order to test it is enough to use the standard configs. I have provided mine as well.

Change in Configuration.h:
`#define MOTHERBOARD BOARD_MIGHTYBOARD_REVE`

Change in Configuration_adv.h:
```
#define DIGIPOT_MCP4018
  #define DIGIPOT_I2C_NUM_CHANNELS 5
  #define DIGIPOT_I2C_MOTOR_CURRENTS { 0.5, 0.5, 0.5, 0.5, 0.5 }

```
### Analog vs Digital pin:
How can Marlin know if a pin should be _analog_ or _digital_?
Example:
```
#define EX1_HEAT_PIN   6     // H3 - 15         (Could as well be Analog 6 - F6)
#define TEMP_BED_PIN  15     // K7 - 69 / ADC15 (Could as well be Digital 15 - J0)

```
### Marlin is great:
I really appreciate the great work done by Marlin. Nothing else would have let me enable Auto Bed Level and other great features. But it is for sure not for beginners 😉

Thanks
Best Regards
Peter R

[Marlin_bugfix20_Arduino_PIO_pin-problems.zip](https://github.com/MarlinFirmware/Marlin/files/4883201/Marlin_bugfix20_Arduino_PIO_pin-problems.zip)

grauerfuchs, 2020-07-09:
Which env are you using in the platformio.ini? There are known issues in some upstream (older) libraries, but it's down to the Arduino-shipped default pins config not supporting a board with more than 70 pins and the problem libraries not detecting an out-of-bounds read on the pins array. The Mightyboard envs (MightyBoard1280, MightyBoard2560) in platformio.ini use the MegaCore board/pin definitions to get around this, since MegaCore was designed to address up to 100 pins. If you're using the standard mega2560 env, it uses the Arduino definitions and so won't properly run on the MightyBoard without first manually expanding the pins definition.

giPeteR, 2020-07-11:
> Which env are you using in the platformio.ini? There are known issues in some upstream (older) libraries, but it's down 

Thanks @grauerfuchs
I think you nailed it. 
I've been using _VS Code_ for a while but was new to _PlatformIO_.
I got there by installing [Auto Build Marlin](https://github.com/MarlinFirmware/AutoBuildMarlin) which by default suggests "**standard mega2560 env**", nothing else.
But when trying the **MegaCore** I got errors saying that _firmware.elf_ wasn't found... I think. Did a compile with **-v** which took 3h15 and when trying to save the log it disappeared.
I then thought of making a fresh install of Marlin, Python, PlatformIO, etc.
Curious of where [get-platformio.py](https://docs.platformio.org/en/latest/core/installation.html#id3) put the files, I ran a Debug on it. Unfortunately **in my source folder**. Maybe needless to say, but an installer wipes not only its own temporary files but in this case just about all of my source files.
I do have backups, but they are weeks old, and you manage to do a lot in a few weeks.
Anyway, I just wanted to tell why it took me two days to respond. And your help was really good. But now I will focus on my garden for a while :/

grauerfuchs, 2020-07-11:
I'm guessing that the autobuild scripts aren't yet updated for full support of the new env. Use of a different env will result in the compiled code being placed in a different folder, so that might explain why it can't find the file. If it hasn't been updated for the env, it may also not be updated for the new configuration options that are Mightyboard specific, either. This might be a time to compile from a manual config and try the built-in Build and Upload commands as part of PlatformIO. 

@thinkyhead , would you know if the "Auto Build Marlin" extension has been updated to support the new MightyBoard envs?

giPeteR, 2020-07-13:
Hi @grauerfuchs
So now I spent about another 20h trying to understand why I get this error:
```
avr-g++: error: CreateProcess: No such file or directory
*** [.pio\build\MightyBoard2560\firmware.elf] Error 1

```
I can compile the code with the _standard_ PIO **env:mega2560**
But when I use PIO **env:MightyBoard2560** the linker (?) seems to fail.

I have _**not**_ installed or used the _Auto Build Marlin_ since I wanted to avoid even more layers to the compiler. It didn't make any difference earlier, except for not using the MegaCore.

I can see that all _.a-files_ seems to be there. 

I also tried to add some debug prints to all the _.py-scripts_ used.
I tried to do a normal build and then a verbose build (since doing verbose of all took well over 3h). It doesn't show anything.
I have an extremely hard time trying to get any debug output of the stage saying:
```
avr-g++: error: CreateProcess: No such file or directory
*** [.pio\build\MightyBoard2560\firmware.elf] Error 1

```
I added _--trace_ and _-v_ everywhere, to no avail.

Attached are logs from **env:MightyBoard2560**, **env:mega2560** and **platformio.ini**
You can use VS Code to compare them side by side.
First is a clean build, then a verbose, in the same file, with a bunch of my debug prints starting with "***".
[Bugfix_2.0.x_PIO.zip](https://github.com/MarlinFirmware/Marlin/files/4913191/Bugfix_2.0.x_PIO.zip)

The command line looks like this:
`C:\Users\Peter\.platformio\packages\toolchain-atmelavr\bin\avr-g++ -v --help  -o .pio\build\MightyBoard2560\firmware.elf -fmerge-all-constants -Os -mmcu=atmega2560 -Wl,--gc-sections -flto -fuse-linker-plugin @"C:\Peter\SW\Marlins\Bugfix\.pio\build\MightyBoard2560\longcmd-9223ba72fc9729c757141d18cb66ca39" -L.pio\build\MightyBoard2560 -Wl,--start-group --trace .pio\build\MightyBoard2560\libbf3\libWire.a .pio\build\MightyBoard2560\lib19c\libLiquidCrystal.a .pio\build\MightyBoard2560\libe7e\libSoftwareSerial.a .pio\build\MightyBoard2560\lib643\libSPI.a .pio\build\MightyBoard2560\libcd8\libTMCStepper.a ".pio\build\MightyBoard2560\libbc4\libAdafruit MAX31865 library.a" ".pio\build\MightyBoard2560\lib9ca\libAdafruit NeoPixel.a" .pio\build\MightyBoard2560\lib3ee\libU8glib-HAL_ID1932.a .pio\build\MightyBoard2560\libb4c\libArduino-L6470.a .pio\build\MightyBoard2560\lib410\libSlowSoftI2CMaster.a .pio\build\MightyBoard2560\lib50b\libLiquidTWI2.a .pio\build\MightyBoard2560\lib0bf\libSailfishLCD.a .pio\build\MightyBoard2560\lib94a\libEEPROM.a .pio\build\MightyBoard2560\libb8a\libServo.a .pio\build\MightyBoard2560\libFrameworkArduinoVariant.a .pio\build\MightyBoard2560\libFrameworkArduino.a -lm -Wl,--end-group`
with _-v (--help) --trace_ added by me.

I have no idea what do do next :(
Any help is appreciated.

I might as well ditch MegaCore and rewrite those routines not understanding +70 pins. At least the _old_ platform compiles... and works 90%'ish.

grauerfuchs, 2020-07-13:
All right then, let's try this. Rather than changing all of the libs (which might be overwritten if and when they're updated upstream), try the following replacement pins file instead:
[pins_arduino.h.txt](https://github.com/MarlinFirmware/Marlin/files/4913363/pins_arduino.h.txt)
The file is properly named `pins_arduino.h` and is typically installed in `[user home folder]/.platformio/packages/framework-arduino-avr/variants/mega`. This version has been modified with the additional pins needed to handle the MightyBoard. I used this file on my compiles prior to the inclusion of the new env, so I know it worked with the mega2560 env on the 2.0-bugfix branch at least as of a few months ago. I still have multiple printers in a build farm running on the firmware compiled with it.

I can't guarantee this file won't be overwritten on a PIO update, but a single file replacement is still easier than modifying all of the libs.

As soon as I get the time to do so, I'll take a look at the posted results and if necessary, test a new build environment to see what's going on with the compiler.

giPeteR, 2020-07-15:
Thanks for `pins_arduino.h`, but I already modified that one... or at least I know I've done so many times. The compiler complains if you don't add the extra pins.

Now the _funny_(?)  part: All of a sudden my `HEATER_0_PIN` is not redefined, it actually works! I have no idea why :(
I will probably do a fresh install of everything on a rainy week.

Few notes though:

- MegaCore won't compile. I had to use mega2560.

- The pins for I2C motor currents lacks pullups (as mentioned in the beginning). I have to issue...
`M42 I M2 P76` 
`M42 I M2 P57` 
`M42 I M2 P61` 
`M42 I M2 P65` 
`M42 I M2 P27` 
`M42 I M2 P77`
...in my g-code. Have not checked yet that currents can be changed.

grauerfuchs, 2020-07-15:
Unfortunately, I haven't had a chance to look at it or test the code yet. My build box decided it wanted to become long-term fertilizer and it corrupted all its backups on the way out, so I'm starting from the ground up anyway. What I can say is that the MegaCore env will compile properly under Linux, with last night's bugfix-2.0.x, and with the configs I managed to salvage, so there may be something wonky in your build environment or it could be a Windows issue. I'm not sure yet, and it's going to be a while before I can test thanks to all of the rebuilding I have yet to complete.

giPeteR, 2020-07-16:
I feel for you. It's not funny to start over. The fact that you replied kept me going. I'm not sure yet if everything works as expected. Let's see about that.

Yes my environment may very well be fishy. I could also try to compile for a complete different platform with default example files. And as you say, maybe Windows setup isn't fully functional.

Maybe this should be closed for the moment? 
Some of the problems still exist but I should probably go over them and clarify.

closed 2020-07-20 by giPeteR

thinkyhead, 2020-07-20:
In Windows world, it can be helpful to move your Marlin source as close to the root of C:\ as possible to keep the paths to files as short as possible. I'm not sure what the `CreateProcess: No such file or directory` error specifically means, but it's a problem at the build environment level and not an issue with any of Marlin's files.

giPeteR, 2020-07-20:
I can try that. I made sure to leave out any spaces in the folder names. I understand (now) it's not due to Marlin but rather PlatformIO (or a combo).

One thing though, the _digipots_ doen't work unless I download 
` https://github.com/stawel/SlowSoftI2CMaster`
separately and replace that library.

I put it in _Marlin/lib_ and added to _platformio.ini_
`lib_dir = Marlin/lib`