MarlinFirmware/Marlin #18583: Compilation error for btt skr 1.1 board.

nchik, 2020-07-07:
Hello, an error occurs when compiling Marlin 2.0.5.3 for the btt skr 1.1 board. 
This code:
`In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/HAL.h:36,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/./STM32F1/fastio.h:30:10: fatal error: libmaple/gpio.h: No such file or
directory
   30 | #include <libmaple/gpio.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_btt\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
============================================ [FAILED] Took 109.39 seconds ============================================`

GMagician, 2020-07-07:
try bugfix

thisiskeithb, 2020-07-07:
Please follow the Issue Template when filing a bug report:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

florianschroen, 2020-07-07:
A fix for this is noted on [reddit](https://www.reddit.com/r/ender3/comments/hfz0dt/help_with_marlin_compilation/fw5jb6r)

> In platformio.ini, scroll down to [env:STM32F103RC_btt_512K] (around line 323) and add underneath it:
> `board_build.core  = maple`
> This tells platformio to use the maple core instead of stm32duino, which became the default in 7.0 (this was the breaking change).

boelle, 2020-07-07:
@nchik 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

also RTFT

sjasonsmith, 2020-07-07:
I’m going to close this since a workaround was posted and this is already fixed in the bugfix branch.

One more thought though, is that none of this is relevant to an SKR 1.1, only to the various SKR Mini and SKR E3 variants. If you truly have an SKR 1.1 you should be using the `LPC1768` environment.

closed 2020-07-07 by sjasonsmith

zouden, 2020-07-08:
@sjasonsmith I don't see the fix in the bugfix branch: the latest revision of platformio.ini still does not specify the maple core. Can you link to the commit or the PR?

sjasonsmith, 2020-07-08:
@zouden it is fixed by requiring an older ststm32 version, rather than explicitly stating maple.

`platform      = ststm32@<6.2.0`

Please try it out and reply again if that is not working.

zouden, 2020-07-08:
That is not the suitable fix IMHO. 

The issue was caused by the new release (7.0) of ststm32 switching to `stm32duino` as the default core (instead of `maple`). [The release notes state that maple should still be used on maple-based boards](https://github.com/platformio/platform-ststm32/releases/tag/v7.0.0). Marlin uses maple, but because maple was the default we never needed to specify it in platformio. 

Thus the correct fix is to specify  `board_build.core = maple` in platformio.ini rather than forcing an old version of ststm32.

sjasonsmith, 2020-07-08:
Feel free to post a pull request to update ststm32 for STM32F1 (and only STM32F1) boards. I am already working on it for other STM32 boards, where quite a lot of additional work is needed. 

Recently dependencies in most of Marlin have been made more strict to avoid uncontrolled upgrades such as this causing problems in released builds. That dependency has simply not being updated yet.

As it stands, compilation is fixed. There is no bug with the current configuration. Choosing to rely on an older dependency is not a bug, simply a decision.

For what it’s worth, I’m not sure whether Maple is going to continue existing in PlatformIO, or be removed entirely in the near future. Moving those boards away from Maple is probably more important than the cosmetic change of saying it is using ststm32 7.0.

zouden, 2020-07-08:
Fair enough, I see the logic now.

thinkyhead, 2020-07-15:
> For what it’s worth, I’m not sure whether Maple is going to continue existing in PlatformIO

As a "deprecated" framework, Maple will be around for a while, but it will definitely go away at some point.