MarlinFirmware/Marlin #18584: [FR] Unit reference for BLTouch Offset (Commented in-code, of course)

T-SquaredProductions, 2020-07-07:
### Description

It's not quite clear what the units for a BLTouch's offset are, to the casual 3d printer enthusiast. Are they steps? Are they millimeters?

### Feature Workflow

1. All that needs to be done is a comment that says  "* Specify a Probe position as { X, Y, Z } **[in mm]/[in steps]**" in the BLTouch probe offset section

sjasonsmith, 2020-07-18:
Configuration files are currently a bit inconsistent with whether they identify units. Some constants have `(mm)` in comments to indicate units, in many other places (such as `NOZZLE_TO_PROBE_OFFSET` as indicated here leave it unspecified.

It has never seemed like a major concern for me, since mm is the universal unit used throughout unless otherwise specified. If it is worth updating this, then the entire files should be scrubbed and clarified.

qwewer0, 2021-01-02:
@T-SquaredProductions Would https://github.com/MarlinFirmware/Marlin/pull/20633 work for you?

thinkyhead, 2021-01-03:
All Marlin configuration options that refer to a length, size, or distance are specified in millimeters. There should be no ambiguity since there are no exceptions.

closed 2021-01-03 by thinkyhead