MarlinFirmware/Marlin #18585: [BUG] Bugfix-2.0 Build Error

KL-XL, 2020-07-07:
### Bug Description

Firmware fails to build despite the "passing" sticker.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4886558/Config.zip)

### Steps to Reproduce

1. Open in VS Code
2. Press Build

**Expected behavior:** Successful build
**Actual behavior:** Failed build

#### Additional Information

* [image]

p3p, 2020-07-07:
Make sure you are using the most recent commit, should be fixed.

KL-XL, 2020-07-07:
Thank you. 45 min ago I merged the most recent commit and it failed to build. Just a min ago I did the pull again and there were 4 more. Now it builds successfully.

closed 2020-07-07 by KL-XL