MarlinFirmware/Marlin #18587: [BUG] (short description)

Cartms1, 2020-07-07:
### Bug Description
Board menu does not accept changes to fans 2 or 3. 
When the pin assignments are changed, flipping fan 1 to another port, that port will then exclusively function.

### My Configurations

https://github.com/Cartms1/FrankenZFirmware
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4887141/Marlin.zip)
### Steps to Reproduce

1. Power on printer
2. Enter main menu
3. Enter Temperature menu
4. Select fan 2 or 3
5. Change speed to any value
6. Submit change

**Expected behavior:** [What you expect to happen]
The fan in question to power on and the menu screen to update with the selected value

**Actual behavior:** [What actually happens]
absolutely nothing, no changes made

###Additional Info

I honestly have no Idea what is going on, looking through the various chunks of code attached to the fans, i have found nothing amiss with what I have.

ellensp, 2020-07-08:
"CONFIGURATION_H_VERSION 020000" this is a old version of marlin.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-07-08:
and also edit your title

[BUG] (short description)

does not say much

Cartms1, 2020-07-08:
Tried that, I was using a modified version of 020000,
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4893559/Marlin-2.0.x.zip)
 because of the fact that marlin as is can not function with the SKR pro 1.1. (without major modification)
Now the "Bugfix" has obfuscated the hell out of the code so now I can't override the sanity check that was erroneously telling me that I only have 1 hotend, and the pin assignments are also wrong for the screen, so that's fun. Before the sanity checks were easy to access, and I have no idea why that was changed.
[bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4893561/bugfix-2.0.x.zip)

I'm including both the updated version, and the one I have that can actually compile correctly, but doesn't have functioning fans.

NOTE: my screen firmware is up to date, and updating the firmware to the screen changed nothing.

My setup uses a main hotend for support and a dual, non-mixing hotend for model. Marlin really needs to differentiate between extruders and hotends because it often calls them both the same thing when there is really no reason to assume they are interchangable.

thinkyhead, 2020-07-08:
> marlin as is can not function with the SKR pro

That's alarming! Please report your bugs related to Marlin failing to compile and run on the SKR Pro 1.1 in a separate issue. We should take care of that first.

Cartms1, 2020-07-09:
Posted the other one bug report. 

The non bugfix version actually compiles and works aside from the fans. I originally had wire management problems when i was first doing test prints but even then the fan for the hotends wasn't coming on. I also included the full firmware due to how many files I had to modify to get marlin to compile previously, namely sanity checks associated with the thermistors which kept screaming that i only have one hotend when i had stated i had two prior. Even when I overrode everything I still had that error, so I went into the sanity check and disabled it, because it was the only link in the chain that had any issues with compilation at that point.

Cartms1, 2020-07-10:
Apparently the ability to have more than one hot end is a new feature for marlin that needs to be added.

sjasonsmith, 2020-07-11:
@Cartms1 the original issue you described regarding the fan menu was fixed in the bugfix-2.0.x branch two weeks ago.
I have had trouble following the comments above to understand whether you have tested with the bugfix branch.

Here are links to the other Bug and Fix:

https://github.com/MarlinFirmware/Marlin/issues/17570
https://github.com/MarlinFirmware/Marlin/pull/18400

closed 2020-07-14 by thinkyhead