MarlinFirmware/Marlin #18588: [BUG] Changing #define PROBING_MARGIN value, has no affect

keleticsaba, 2020-07-07:
### Bug Description

Changing #define PROBING_MARGIN value, has no affect

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[https://github.com/keleticsaba/Marlin-2.0.5.3-for-Geeetech-A10-under-construction-/blob/2.0.x/Marlin/Configuration.h](https://github.com/keleticsaba/Marlin-2.0.5.3-for-Geeetech-A10-under-construction-/blob/2.0.x/Marlin/Configuration.h)
[https://github.com/keleticsaba/Marlin-2.0.5.3-for-Geeetech-A10-under-construction-/blob/2.0.x/Marlin/Configuration_adv.h](https://github.com/keleticsaba/Marlin-2.0.5.3-for-Geeetech-A10-under-construction-/blob/2.0.x/Marlin/Configuration_adv.h)

### Steps to Reproduce

1. Upload firmware with changed #define PROBING_MARGIN value
2. Auto home
3. Level bed

**Expected behavior:** Make the probing not so close to the edge

**Actual behavior:** The probing executes exatly the same place, too close to the edge

#### Additional Information

-

thisiskeithb, 2020-07-07:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

sjasonsmith, 2020-07-08:
That setting was still `MIN_PROBE_EDGE` in 2.0.5.3. If bugfix configs were used by 2.5.3 code it would not behave correctly.
I can tell from the `platformio.ini` in that fork that it doesn't have all the recent changes that have gone into `bugfix-2.0.x`.

keleticsaba, 2020-07-08:
Then why is it not updated in https://github.com/MarlinFirmware/Configurations ?

ellensp, 2020-07-08:
Perhaps you didn't select the bugfix branch of configurations?

keleticsaba, 2020-07-08:
> Perhaps you didn't select the correct branch of configurations?

No,
[https://raw.githubusercontent.com/MarlinFirmware/Configurations/bugfix-2.0.x/config/examples/Geeetech/A10/Configuration.h](https://raw.githubusercontent.com/MarlinFirmware/Configurations/bugfix-2.0.x/config/examples/Geeetech/A10/Configuration.h)
it is using "PROBING_MARGIN"

keleticsaba, 2020-07-08:
> Perhaps you didn't select the correct branch of configurations?

óóó b@ssza@ meg a kurv@ élet

closed 2020-07-08 by keleticsaba