MarlinFirmware/Marlin #18591: [BUG] Automatic Bed Leveling not working correctly

johniak, 2020-07-08:
### Bug Description

Hi,

I'm trying to use autoleveling feature on Marlin (bugfix 2./2.0.5.3 tested on both). I tried to use bilinear mode and also UBL.
The result is always the same. When I'm trying to print test object (with command `G26`) the right side of the table is too close to the nozzle, and the left side is too far.

I tried different positioning of my table (changed manually) which you can see on pictures below
[image]

[image]

[image]

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4890096/config.zip)

### Steps to Reproduce

1. Set temperatures to PLA printing temperatures
2. Wait until temperature is stabilized
3. Run `G28`
4. Run `G29`
5. Run `G26`

**Expected behavior:** Print perfect test bed leveling object, with z axis compensation

**Actual behavior:** Always left side is too far to the nozzle and right side too close to the nozzle

AnHardt, 2020-07-08:
A really nice issue description - but boring.
Please compare with #17399, #15753, #15206, #11234. #12349 and several others. [Visualisation of x-sled-rotation](https://github.com/AnHardt/Marlin/issues/84)

johniak, 2020-07-08:
Hi,
Thank you very much for your fast response. As I understand, my x linears are probable twisted
[image]

ugh, so this will be propably really hard to fix :(

AnHardt, 2020-07-08:
Could also be a too short cable or filamet pulling on the carriage when left or right or anything able to make probing systematic unreliable - so that nozzle-tip to probes trigger-point distance differs when moving around the bed.

When using UBL you are able to edit the probed mesh to match the reality - without hardware fixing the system.

lrpirlet, 2020-07-08:
Well, when epatel introduced mesh levelling, the real condition was that ONLY z was NOT at the tip of the hot end... It was expected that Delta-X and Delta-Y was ZERO or at least small enough for the error to be negligible...
Looking at the photo, this is not the case...

In fact, looking at your setup, you have the X axis acting as a spring. The two bars of the X should define a XZ plane. That plane should be able to take up the torque formed by the head... Physics tels me that without counter weight, the force tend to infinitum for that that plane to stay a plane... In fact, your graphics shows very well that the correction looks the same all along the Y axis... 

AND, because the probe is a the end of that lever, the error is bigger. if the head goes down .2 mm, the probe goes down even more...

Solution would be to insure that the torque is nul, that would not be possible without a major redesign... 
Or you may get back in time and use a stupid switch, on a arm, moved by a servo, in between the head and the bed. That insure that delta-X and delta-Y are ZERO...
In fact the epatel solution was a 3D printed cap that contained a switch coupled with the z endstop. He would then map the mesh, store it and use it until the need to retune it again.

I went through the same trouble as you, for the same exact reason: the XZ plane twisted under the action of the couple due to the weight of the head. 

ref: #1619

johniak, 2020-07-08:
@AnHardt I resolved it exactly in this way, thank you very much

closed 2020-07-08 by johniak