MarlinFirmware/Marlin #18592: [Minor BUG] Filament Runout State is incorrectly reported by M119

minosg, 2020-07-08:
### Bug Description

This is a minor bug with little to no effect in the functional behavior of the printer, but which will confuse new users and people following up online tutorials.

There has recently been a change to `FIL_RUNOUT_INVERTING` --> `FIL_RUNOUT_STATE` in pull request [18537](https://github.com/MarlinFirmware/Marlin/pull/18537/)

I believe that logic needs to be inverted in the [endstops.cpp](https://github.com/MarlinFirmware/Marlin/pull/18537/commits/e20bc8c1ffdaf4c3e9967821d5522ec888140379)

```
// From
print_es_state(READ(FIL_RUNOUT_PIN) != FIL_RUNOUT_STATE, PSTR(STR_FILAMENT_RUNOUT_SENSOR));

// To 
print_es_state(READ(FIL_RUNOUT_PIN) == FIL_RUNOUT_STATE, PSTR(STR_FILAMENT_RUNOUT_SENSOR));
```

Since according to the comment the state which we configure on the config is the one to trigger the "No filament State"

When that is set to LOW -> M119 should display Triggered when the pin is LOW and open when filament is present
When that is set to HIGH -> M119 should display Triggered when the pin is HIGH and open when filament is present.

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4890152/configs.zip)

### Steps to Reproduce

1. Enable filament run-out
2. Use m119 to test sensor and it will display triggered when it has filament and open when it does not
3. Start a print and verify that the sensor logic is correct, and will print with filament in and pause with filament removed.

**Expected behavior:** 
I would expect M119 to show triggered when the filament is not present on the sensor

**Actual behavior:**
The exact oppposite

#### Additional Information

The issue can be fixed by inverting the logic of the endpoints.cpp detection logic as seen here:

https://github.com/MarlinFirmware/Marlin/pull/18537/commits/e20bc8c1ffdaf4c3e9967821d5522ec888140379

```
```

GMagician, 2020-07-08:
Ops I just posted the same...confirm it's reverted also to me....I'll post a PR

GMagician, 2020-07-08:
I think there are more that one point where the meaning is reversed.
In my opinion when STATE is HIGH, when input is 1 it means that filament is missing. But it seems that @robbycandra made the change considered it working at reverse

minosg, 2020-07-08:
Lol perfect timing mate. Sequential bug numbers...

GMagician, 2020-07-08:
@minosg in my opinion this is not a bug. M119 report "filament" not "filamen runout" so it's correct a triggered when filament is present

closed 2020-07-08 by thisiskeithb

minosg, 2020-07-08:
@GMagician true it is good to be consistent. The same thing meant exactly the opposite a week ago. Even the official documentation reflects the old context. 

M119 is a standard command new users will use to debug their setup and try to follow guides.

Anyhow the future will tell how users perceive this

GMagician, 2020-07-08:
@minosg in such case we may change "filament" to "filament run out" but I don't know if it may bread something somewhere else

thinkyhead, 2020-07-09:
> M119 report "filament" not "filament runout" so it's correct a triggered when filament is present

As I understand it, it should be reporting the same thing as before the latest change. The meaning of `FIL_RUNOUT_STATE LOW` should be the same as `FIL_RUNOUT_INVERTING false`. If I reversed the meaning of the "triggered" state in `M119` then that would be an error. But, these really should be functionally equivalent:
```diff
-        print_es_state(extDigitalRead(pin) != FIL_RUNOUT_INVERTING);
+        print_es_state(extDigitalRead(pin) != FIL_RUNOUT_STATE);
```

minosg, 2020-07-09:
I honestly did not expect this issue to get that much attention so apologies for raising it.

It is a mind twister going around the logic and either way it's a not issue other than being confusing.

thisiskeithb, 2020-07-09:
After reading this again, I'm confused.

On my [MK3S/BTT002](https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209) build with a filament runout sensor, it used `FIL_RUNOUT_INVERTING true` before and now uses `FIL_RUNOUT_STATE HIGH`.

This is the `M119` output when filament is present:
```log
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: filament: TRIGGERED
```

Is that correct?

GMagician, 2020-07-09:
@thisiskeithb nothing has changed from previous in code. Only thing is that previously Marlin was considering the HIGH level equal to filament present and you had to invert logic (true) when the sensor was LOW when filament was present. 
Now LOW or HIGH are tied to the sensor function. Sensor is for run out then what you write is the state when sensor sense a filament missing.

What may be considered wrong is what M119 reports. But if you considered that it states "filament" then triggered may signal that filament is triggered, not the run out is triggered. In my opinion it will be better to change message to "filament run out" so triggered may be easily understood

robbycandra, 2020-07-11:
@thisiskeithb , the filament : TRIGGERED is indeed confusing status.

thinkyhead, 2020-07-14:
Endstop switches are triggered when an object like an X carriage or a Bed is pressed against them (causing them to become OPEN in most cases). Marlin has always labeled these as "TRIGGERED" which is a word that —confusingly— can refer to a switch or a condition.

As a special member of the "endstops" family of switches, a Filament Sensor is considered "triggered" when filament is present. A better name for the feature might be "`FILAMENT_SENSOR`". But since they are conventionally referred to as "filament runout sensors" and not "filament presence sensors" the name is what it is.