MarlinFirmware/Marlin #18598: [BUG] (Multiple BLTouch Issues)

Dracrius, 2020-07-09:
### Bug Description

Since updating to Marlin 2.0.5.3 from 2.0.4.1 I noticed my BLTouch v3 was reporting random major dips in my bed. I decided to try and tune my BlTouch settings to improve reliability BUT this lead me down a hole that only resulted in even more issues. I'll admit I went a bit crazy trying to figure this out and lost track of some things I tried but what I have verified should be enough for someone to help me track it down.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

I setup my BLtouch identically in both firmware versions. I just verified by reinstalling my 2.0.4.1 build that with it and Octoprints Bed Level Visualizer report a VERY consistent bed level mesh!

Flashing Marlin 2.0.5.3 results in a fairly repeatable 1 in 3 bed level test result in random massive dips in the mesh

Enabling  #define MULTIPLE_PROBING 2 changes the result from a bad mesh to an entire printer fail with an M112 Restart Printer Message basically the same 2 out of 3 times. Strangely it fails mid air on a random position. some times its in the first 3 some times its one of the last 3 but it will prob once in said position then drop the pin to prob again but immediately fail and begin to blink before lowering to the bed.

**Expected behavior:** Consistent probing like 2.0.4.1 without the need for Multi Probing and or successful multi probing 

**Actual behavior:** Dips in mesh or complete failures
#### Additional Information

[Marlin 2.0.5.3 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4893943/Marlin.2.0.5.3.Configs.zip)

thisiskeithb, 2020-07-09:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Dracrius, 2020-07-09:
@thisiskeithb Alright I did as you asked but I also decided to add my settings in stages. By choosing exactly what I used in 2.0.4.1 I got consistent meshes 3 times in a row so I slowly added in the handful of new settings I had been advised to use and was trying to add. 

I expected it to be #define FAN_SOFT_PWM but that added without issue.

Next I readded 

Configuration.h
#define HOMING_FEEDRATE_XY (50*60)

Configuration_adv.h
#define SLOWDOWN_DIVISOR 8
#define ADAPTIVE_STEP_SMOOTHING

At this point I had my first two meshes both show a dip and a different place. I missed the first bit of the log the first time so I only have the M111 S247 detailed terminal from the second but the second one did the same thing so it worked out. I will attach the Terminal Log, Config Zip and a Bed Visualizer Graph of said failure.

[image]
[M111 S247 Terminal Log of a Bad bed Level.txt](https://github.com/MarlinFirmware/Marlin/files/4894692/M111.S247.Terminal.Log.of.a.Bad.bed.Level.txt)
[Marlin Bugfix 2.0.x Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4894693/Marlin.Bugfix.2.0.x.Configs.zip)

Dracrius, 2020-07-09:
I'll do more test to see if I can narrow down an exact setting

Dracrius, 2020-07-09:
I just ran about 10 levels in a row with #define HOMING_FEEDRATE_XY (50*60) still changed but 

#define SLOWDOWN_DIVISOR 2
//#define ADAPTIVE_STEP_SMOOTHING

Reset to these defaults. It has resulted in a very repeatable mesh.
[image]

If you need me to dial it down to one of those two setting it will have to wait till tomorrow as I need to catch some sleep but I hope this was enough to get an idea of what is happening.

GMagician, 2020-07-09:
Whats' happens with such value in slowdown (2) is that logic works, here is code
`if (WITHIN(moves_queued, 2, (BLOCK_BUFFER_SIZE) / (SLOWDOWN_DIVISOR) - 1))`
Usually `BLOCK_BUFFER_SIZE` is 16 and with SLOWDOWN_DIVISOR=8 you get 16/8-1 is 1and move_queued will probably never within 2 and 1

qwewer0, 2020-07-09:
`ENDSTOP_INTERRUPTS_FEATURE` 32 and `SLOWDOWN_DIVISOR` 8 is better of worse? Not sure how it works?

GMagician, 2020-07-09:
@qwewer0 in my opinion endstop is better because it detect faster the trigger. About slowdown divisor what I can see is that with 8 and block_buffer_size to 16 the slowdown code doesn't work but AFAIK such code is used to other purposes, not related to home or probe

Dracrius, 2020-07-09:
> @qwewer0 in my opinion endstop is better because it detect faster the trigger. About slowdown divisor what I can see is that with 8 and block_buffer_size to 16 the slowdown code doesn't work but AFAIK such code is used to other purposes, not related to home or probe

While it wouldn't surprise me that it is the slowdown devisor since that seemed like a really random setting to tweak i had never changed before but I stumbled onto this guide (https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/) which has a tone of suggested changes id never seen before and adaptive sync and slow divisor where the last ones Id never used before but had tried in 2.0.5.3.

I honestly dont know why youd compair it to ENDSTOP_INTERRUPTS_FEATURE @qwewer0 they have nothing to do with one another. From the little they say in the configs the Slowdown Devisor says "If defined the movements slow down when the look ahead buffer is only half full" so if i understand that correctly it relates to slowing the printer down if your buffer becomes to full as to not overload it.

Now the suggested reason for changing this is in the merge request #17171 stating that 32bit boards were not even coming close to half fulling there buffers with a setting of 2 though they say they are using 4 not a wopping 8. I and other users are i'm guessing using 8 simply due to the guides on reddit as that same user made a e3 2.0 tutorial as well which is the mainboard The3DE is using in the other issue.

thisiskeithb, 2020-07-09:
I'm closing this issue since there's already one open for ABL/BLTouch compensation issues: https://github.com/MarlinFirmware/Marlin/issues/17262

Dracrius, 2020-07-09:
Id argue that one has become a mess and doesnt actually identify the issue but I guess this is still linked at minimum. This issue also related the multi probing causing entire failures which is not under discussion at all on the other side so we must keep that in mind as well.

thisiskeithb, 2020-07-09:
> This issue also related the multi probing causing entire failures which is not under discussion at all on the other side

Do you have a link? It sounds like multiple issues still need to be consolidated.

Dracrius, 2020-07-09:
I will test more in a day or two when I am back home on the multi probing but this is very much a diffrent beast then #17262 but very much related to or the solution for #18526 with us finding that either the adaptive sync settings has confilcts with other parts of the code or a problem with setting the slowdown devisor to high, but the result is random massive dips in an otherwise perfect mesh vs the issue you posted shows poor probing results at best and they dont mention outright failures with multiple probing enabled which is being used as stated in their first post of that issue.

Dracrius, 2020-07-09:
Wow the issue you linked got salty not sure how much headway youll make over there but honestly after reading through the posts it is unrelated @thisiskeithb

txt4nk, 2020-07-09:
Dracrius, I am one of the people in 17262 that is having issues. I am building firmware right now that removes multiple probes and your homing feedrate changes to see if by any chance that those two modifications may help the issue we are having.

Dracrius, 2020-07-09:
@txt4nk I will post my Marlin changes text file so you can compare our other settings as well

txt4nk, 2020-07-09:
That would be great. @thisiskeithb are you okay with Dracrius and myself continuing troubleshooting between ourselves in this closed thread? Any findings worth sharing will be posted in #17262.

Dracrius, 2020-07-09:
[Marlin Firmware Changes.txt](https://github.com/MarlinFirmware/Marlin/files/4897999/Marlin.Firmware.Changes.txt)
This lists all changes I make to the example configs. Most of it is personal like temps, Machine name and z Max height and I also left in the settings I had changed but noted the defaults they should be returned to.

txt4nk, 2020-07-09:
alright so I put glass on my ender 3 this morning, and just now changed my firmware. I ran a bed visualizer run in octoprint and it looks pretty decent.. Starting a 160mm X 160mm single layer square right now to test my first layer. I'll post results as soon as its done.

Dracrius, 2020-07-09:
@txt4nk @thisiskeithb I had a second between work to run another test. I enabled ADAPTIVE_STEP_SMOOTHING and then reflashed. After the third bed level I got the dip again 
[image]

I am now going to sanity check that is is just ADAPTIVE_STEP_SMOOTHING by disabling it and resetting SLOWDOWN_DIVISOR to 8.

Dracrius, 2020-07-09:
Also @GMagician the SKR Mini e3 1.2 example config has the block_buffer_size set to 32 not 16. I'll admit I do not know how these settings actually interact with anything as I could sware making the SLOWDOWN_DIVISOR bigger would tell the printer to slow down more often making it trigger at an 8th rather then a half of the buffer size but the original pull request mention increasing it for 32bit boards and at least one user with a very popular tutorial for SKR Mini boards suggest setting it to 8.

txt4nk, 2020-07-09:
My test print failed, left half of the bed was not sticking, whereas the right side looked semi okay. Just this morning I got a great first layer using the z end stop switch and removing ABL completely. However with that being said.. It did not fail as bad as it has in the past. Its either something to do with the settings you provided, or the glass. I'm running another visualizer right now on the glass, and will put my spring steel sheet on right after to see what the difference is between the two via bed visualizer.

txt4nk, 2020-07-09:
Whel.... apparently my glass is warped lol
[image]

Dracrius, 2020-07-09:
haha well glass isnt perfect and bed visualizer/ bed probing is very accurate that looks WAY worse then it actually is if you look at the scale we are talking 100ths of a mm differences. I'm on test 6 with ADAPTIVE_STEP_SMOOTHING disabled and SLOWDOWN_DIVISOR set to 8 no issues yet.

sjasonsmith, 2020-07-09:
This issue may be a duplicate of others, but I’ll reopen it based on the quality of the discussion here. The several open issues related to this may need re-consolidated at some point, but it is hard to tell which are truly duplicates and which are now.

txt4nk, 2020-07-09:
I agree with the fact that the delta is super minor, however the corrections being made during the print don't seem to be doing the trick. At this point I do believe that my issue and your issue are two entirely different animals.

Dracrius, 2020-07-09:
[image]
This was replicated about 10 times in a row with SLOWDOWN_DIVISOR set to 8 so the only outlier is ADAPTIVE_STEP_SMOOTHING. If someone could confirm by enabling ADAPTIVE_STEP_SMOOTHING and running half a dozen probes you should see a failure in the first 3 but I have seen times you luck out and it will work many times without issue.

Dracrius, 2020-07-09:
This can also be seen in the Terminal like so
```

Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.017 +0.060 +0.088
Recv:  1 -0.022 -0.012 +0.023
Recv:  2 -0.025 -0.005 +0.005
```
Just take note of a few that match and if one G28 results in a z of -1.### that would be this bug.

Dracrius, 2020-07-09:
@txt4nk what Z-Probe Offset are you using? did you try dialing it in just a hair? Sounds like your just a bit to high unless you seeing scaring from being to low and lifting in other places of your first layer.

I don't always get perfect adhesion on glass. I have to clean it between every print and really dialed in my z-offset as you can easilly just go a bit to low and guarantee itll stick without scraing the bottom layer to much but you get a decent Elephants foot so I dial mine back to almost perfect and even with cleaning get a failure maybe 1 in 10 times almost definitely just do to the house ac kicking in or really even a door opening. Glass is amazing but finicky!

txt4nk, 2020-07-09:
I got it figured out actually. This morning when I went to glass, I put my OEM bed back on (pretty sure its warped). I just put my Spool3D cast aluminum bed back on and so far, my first layer is looking EPIC.

txt4nk, 2020-07-09:
[image]
Aside from the creases made when removing a single layer from the sheet. This thing looks great.

AnHardt, 2020-07-09:
Neither 'SLOWDOWN_DIVISOR' nor 'BLOCK_BUFFER_SIZE' should have much influence. During probing about every second move calls `synchronize()`, emptying the block-buffer.
While `ADAPTIVE_STEP_SMOOTHING` purposely maximizes the amount of high priority stepper-interrupts. Those can very well delay the detection of endstops/probes. Likely the limit for that maximization is currently a bit high. In that calculation the maximum duration of the stepper-interrupt is estimated - but likely guessed a bit to low.
Anyone having an idea for measuring and averaging that time?

sjasonsmith, 2020-07-09:
I’m seeing some issues with step smoothing and linear advance in my open STM32 PR. For an SKR Pro I used an output line to verify the the estimates are off by around 10%.

On ARM It would probably be possible to use the CPU counters to time the duration and store it to a variable and print to serial.

AnHardt, 2020-07-09:
> 10%.

I'd have guessed its more - but doubling the amount of interrupts in every iteration adds up the error fast - especially when probing with low speeds.

> and print to serial.

My idea was more to adjust on the fly - more like what i did with display throttling.

sjasonsmith, 2020-07-09:
> My idea was more to adjust on the fly - more like what i did with display throttling.

I like that. Printing was only recommended as a first debug step to confirm the presence of a problem.

sjasonsmith, 2020-07-09:
The duration of the the stepper interrupt is dependent on lots of fragile times values. It was 10% on that config, but the error likely varies depending on the set of enabled features and hardware configuration.

AnHardt, 2020-07-10:
(Like always) the simple ideas are not that simple when you look on them closer.

We can only measure the time inside the interrupt. But for knowing how much time a single interrupt takes away from the other processes we have to add the time it takes to call and return from the interrupt (saving/restoring registers). That not only differs by the used processor type but for example also for using the F4s FPU or not. For me this takes out the 'beauty' of the idea.

As a workaround we could experiment with adding a ISR_SAFETY_CYCLES [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper.h#L221)
`#define ISR_EXECUTION_CYCLES(R) (((ISR_BASE_CYCLES + ISR_S_CURVE_CYCLES + (ISR_LOOP_CYCLES) * (R) + ISR_LA_BASE_CYCLES + ISR_LA_LOOP_CYCLES)) / (R))`
 ISR_SAFETY_CYCLES could have values from 0 to a few hundred.

AnHardt, 2020-07-10:
Could the calculation of `oversampling_factor` really randomly result in 100% processor load alone for the stepper interrupt?
Can't find a margin for 'normal' execution and the other interrupts.

thierryzoller, 2020-07-10:
I can confirm that I have issues with ABL and that I have ADAPTIVE_STEP_SMOOTHING enabled. Let's see if we can duplicate theresults.

thierryzoller, 2020-07-10:
Referencing a Closed but not solved issue that is the same
https://github.com/MarlinFirmware/Marlin/issues/17872

qwewer0, 2020-07-10:
Can someone check out https://github.com/MarlinFirmware/Marlin/issues/18616 and evaluate it?

thisiskeithb, 2020-07-10:
> Can someone check out #18616 and evaluate it?

Use one of the resources I linked in my comment on that post.

sjasonsmith, 2020-07-11:
I just tested this on my SKR E3 Mini 1.0, and confirmed that my prove accuracy decreased significantly with `ADAPTIVE_STEP_SMOOTHING` enabled. I didn't see a 1mm change, but a range of `0.007` increased to `0.210`.

qwewer0, 2020-07-11:
I'm not too sure about the `ADAPTIVE_STEP_SMOOTHING` thing.
By disabling it my standard deviation decreased from 0.0017 to 0.0023, so I'm a bit confused. 
And without it, the combined XY movements are more loud than with it, any idea?

sjasonsmith, 2020-07-11:
@qwewer0, `ADAPTIVE_STEP_SMOOTHING` works by doubling the interrupt rate until it exceeds the predicated maximum rate. It seems that in some case it is increasing the rate too far, which prevents the homing code from detecting the probe's trigger.

Whether it reaches that point depends on the speed you are moving, steps/mm, and other aspects of your configuration. This is why it might cause problems during homing on one machine and not another.

qwewer0, 2020-07-11:
Ok, so I tested it again and my results are different than what I had, so it's a bit inconclusive.

M48 P50
```
ADAPTIVE_STEP_SMOOTHING Enabled

READ: Mean: 0.001200 Min: -0.003 Max: 0.005 Range: 0.008
READ: Standard Deviation: 0.001819
READ: Mean: -0.000600 Min: -0.005 Max: 0.005 Range: 0.010
READ: Standard Deviation: 0.001772
READ: Mean: 0.000450 Min: -0.003 Max: 0.002 Range: 0.005
READ: Standard Deviation: 0.001710
```

```
ADAPTIVE_STEP_SMOOTHING Disabled

READ: Mean: 0.000350 Min: -0.003 Max: 0.005 Range: 0.008
READ: Standard Deviation: 0.001871
READ: Mean: -0.000600 Min: -0.005 Max: 0.002 Range: 0.007
READ: Standard Deviation: 0.001908
READ: Mean: -0.000100 Min: -0.003 Max: 0.002 Range: 0.005
READ: Standard Deviation: 0.001729
```

Dracrius, 2020-07-11:
@qwewer0 that is probably just do to its nature of not being a 100% consistent problem. Youd be best to run sevral test with both configurations. From my testing with my bltouch speeds and grid it would prop up in 1in 3 tests most commonly but a range of 6-10 test would be more conclusive as this is what i did and could then reproduce it very consistently.

qwewer0, 2020-07-11:
M48 P50

<details>
<summary>ADAPTIVE_STEP_SMOOTHING Disabled</summary>

<details>
<summary>ENDSTOP_INTERRUPTS_FEATURE Disabled</summary>

```
Mean: -0.002950 Min: -0.005 Max: 0.002 Range: 0.007
Standard Deviation: 0.001981
Mean: -0.001450 Min: -0.008 Max: 0.005 Range: 0.013
Standard Deviation: 0.002602
Mean: -0.001800 Min: -0.008 Max: 0.002 Range: 0.010
Standard Deviation: 0.002238
Mean: -0.002350 Min: -0.008 Max: 0.002 Range: 0.010
Standard Deviation: 0.002259
Mean: -0.002600 Min: -0.008 Max: 0.002 Range: 0.010
Standard Deviation: 0.002234
Mean: -0.003400 Min: -0.008 Max: 0.000 Range: 0.008
Standard Deviation: 0.002107
Mean: -0.003050 Min: -0.008 Max: 0.002 Range: 0.010
Standard Deviation: 0.001955
Mean: -0.004600 Min: -0.010 Max: 0.002 Range: 0.012
Standard Deviation: 0.002311
Mean: -0.004300 Min: -0.010 Max: 0.000 Range: 0.010
Standard Deviation: 0.002293
Mean: -0.004950 Min: -0.008 Max: 0.000 Range: 0.008
Standard Deviation: 0.002091
```

</details>

<details>
<summary>ENDSTOP_INTERRUPTS_FEATURE Enabled</summary>

```
Mean: 0.001750 Min: -0.003 Max: 0.007 Range: 0.010
Standard Deviation: 0.002250
Mean: 0.002650 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.001831
Mean: 0.004100 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.001855
Mean: 0.004350 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.002050
Mean: 0.003500 Min: 0.000 Max: 0.012 Range: 0.012
Standard Deviation: 0.002345
Mean: 0.004600 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.001960
Mean: 0.003350 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.002157
Mean: 0.003750 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.001887
Mean: 0.003100 Min: 0.000 Max: 0.007 Range: 0.007
Standard Deviation: 0.001908
Mean: 0.002650 Min: 0.000 Max: 0.005 Range: 0.005
Standard Deviation: 0.001613
```

</details>

</details>

<details>
<summary>ADAPTIVE_STEP_SMOOTHING Enabled</summary>

<details>
<summary>ENDSTOP_INTERRUPTS_FEATURE Disabled</summary>

```
Error:Probing Failed
Error:Probing Failed
Error:Probing Failed
Error:Probing Failed
Error:Probing Failed
Mean: 0.001950 Min: -0.003 Max: 0.007 Range: 0.010
Standard Deviation: 0.002079
Mean: 0.000750 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001887
Mean: 0.000350 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001733
Mean: -0.000550 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001890
Error:Probing Failed
```

</details>

<details>
<summary>ENDSTOP_INTERRUPTS_FEATURE Enabled</summary>

```
Mean: 0.002600 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001800
Error:Probing Failed
Mean: 0.006650 Min: 0.005 Max: 0.010 Range: 0.005
Standard Deviation: 0.001776
Mean: 0.003600 Min: 0.000 Max: 0.005 Range: 0.005
Standard Deviation: 0.001338
Mean: 0.006450 Min: 0.002 Max: 0.010 Range: 0.008
Standard Deviation: 0.001422
Mean: 0.004750 Min: 0.002 Max: 0.010 Range: 0.008
Standard Deviation: 0.002016
Mean: 0.004700 Min: 0.002 Max: 0.010 Range: 0.008
Standard Deviation: 0.001706
Mean: 0.002000 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001871
Mean: 0.001650 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001704
Mean: 0.001100 Min: -0.003 Max: 0.005 Range: 0.008
Standard Deviation: 0.001744
```

</details>

</details>

Dracrius, 2020-07-11:
That makes sense, Endstop Iterupt feature cercumventing the issue with Adaptive Smoothing and probing but im not certain thatd work for all printers as the feature description says.

Enable this feature if all enabled endstop pins are interrupt-capable. This will remove the need to poll the interrupt pins, saving many CPU cycles.

But the extra data is great should make linking similar issues back here easier. I also think there was talk of adaptive smoothing causing other issues anyways.

sjasonsmith, 2020-07-12:
I experimented today with reducing the maximum ISR rate to see if accuracy returned. I had to go all the way to 50% to get most of my accuracy back.

I ran out of time today, but I’m experimenting with ways to either check endstops more often or disable step smoothing while homing/probing.

jbbandos, 2020-07-12:
For what is worth, with ADAPTIVE_STEP_SMOOTHING disabled I have done several bed probings (G29) on a row, without a single error.  All of them for 16 points, on an Ender 3 pro with a SKR Mini E3 V2 board. With it enabled, I'd have 3 bad bed probings out of 4 due to probing errors - non-existing holes detected, as @Dracrius documented. Values between below -0.8.

sjasonsmith, 2020-07-13:
I believe the problem is happening because the Servo interrupt is lower priority than the STEP and TEMP ISRs. When using `ADAPTIVE_STEP_SMOOTHING` it is possible to have two STEP interrupts with a TEMP interrupt between them, which blocks the Servo interrupt from running long enough for the output PWM signal to be out of the acceptable BLTouch range.

In the image below, the yellow vertical line is where the Servo PWM needed to start in order to have the correct length. I have added outputs so I could see when interrupts are active, and you can see that the servo code is being blocked by the other interrupts.

[image]

AnHardt, 2020-07-13:
@sjasonsmith 
You are absolutely right. That is the (or one of the) reasons for random **actions** of the **servo** of the BL-Touch. However "late detection" of the "signal change" at the BL-Touches output pin is not related to the servo-interrupt-priority. Here the priority of the Pin-Change_Interrupts matters if `ENDSTOP_INTERRUPTS_FEATURE` is in use. If not the evaluation is done mostly in the temp-interrupt. I am still sad that the limit switch detection is no longer in the stepper-interrupt. That was expensive but exact.

sjasonsmith, 2020-07-13:
@AnHardt that is what I suspected as well, but it turned out to be wrong. When inspecting the input, I could not find any patterns to correlate bad probes with slow detection of the line change. I used output lines and code instrumentation to let me measure the delay between the BLTouch output and the code detecting the change.

This is why I looked at the output PWM, and found that every bad probe had a bad output pulse 2-3 pulses before the probe detected the bed. It seems that after a bad PWM pulse, the BLTouch needs to receive 2-3 good pulses before it signals. This is a 40-60ms delay!

I was going to post a PR tonight, but GitHub went down. I’ll put it up tomorrow.

On my system, I was testing using the BLTouch in SW mode without endstop interrupts. Endstop interrupts would probably make it a tiny bit better, in particular if probing at higher speeds.

sjasonsmith, 2020-07-13:
Late detection could still be an issue for anybody using neither endstop interrupts nor SW Mode. In that case you could probably miss the pulse entirely.

AnHardt, 2020-07-13:
However. We could get rid of the servo problems on some boards by using a hardware-PWM for it. The servo-libraries all are written to handle multiple servos on random pins with one timer. In case of Marlin we mostly have only one servo. We could look up if the servo-pin can be driven by a timer, If that's the case we can, if there is only one servo, setup a hardware-PWM, else we could switch to the normal servo-library. That would not use more timers than before - just maybe an other. If the second servos pin is on the same timer even that could get the hardware-PWM. (In case of SKR PRO both of the BL-Touch pins are on the same timer. The probe-pin could be somewhere else. If having the right pins and timer up to 4 servos would be possible on the STM32F4s)

A similar attempt could work for TONE - but i could not find a board matching the requirements.

--------

Best would be to drop BL-Touch support because of being closed source - and my traditional sermon about patents and so on.

AnHardt, 2020-07-13:
Started to investigate `ADAPTIVE_STEP_SMOOTHING` and 'MULTY_STEPPING' in depth.
But will not finish that until Wednesday - other real live things to do.

Discovered some stupid things.
`ADAPTIVE_STEP_SMOOTHING` currently can occupy 100% of processor power for the stepper-interrupt alone by stepping at the wrong speed.
`SLOWDOWN` is completely ineffective in some cases because the saved time there is happily consumed from `ADAPTIVE_STEP_SMOOTHING`.
'MULTY_STEPPING' can step more than microsteps/full-step at once - unavoidably loosing steps.

---

For testing i tend currently to suggest:
```diff
// The minimum allowable frequency for step smoothing will be 1/10 of the maximum nominal frequency (in Hz)
- #define MIN_STEP_ISR_FREQUENCY MAX_STEP_ISR_FREQUENCY_1X
+ #define MIN_STEP_ISR_FREQUENCY (MAX_STEP_ISR_FREQUENCY_1X / 2)
```
```diff
      #if ENABLED(ADAPTIVE_STEP_SMOOTHING)
        // Decide if axis smoothing is possible
        uint32_t max_rate = current_block->nominal_rate;  // Get the maximum rate (maximum event speed)
        while (max_rate < MIN_STEP_ISR_FREQUENCY) {         // As long as more ISRs are possible...
          max_rate <<= 1;                                   // Try to double the rate
-          if (max_rate >= MAX_STEP_ISR_FREQUENCY_1X) break; // Don't exceed the estimated ISR limit
+          if (max_rate >= MIN_STEP_ISR_FREQUENCY_1X) break; // Don't exceed the estimated ISR limit
          ++oversampling;                                   // Increase the oversampling (used for left-shift)
        }
        oversampling_factor = oversampling;                 // For all timer interval calculations
      #endif

```
and than playing with the factor 2. `(MAX_STEP_ISR_FREQUENCY_1X / 4 * 3)` may also be worth a try.

sjasonsmith, 2020-07-13:
All good points @AnHardt. I saw the same issues you did with adaptive step smoothing. My first attempts were slowing down step smoothing, but that only lessened the problem, without eliminating it.

As for MULTI-STEPPING...I think it is a terrible hack that makes providing clean pulse output impossible. That may have been what caused me to stop trying to fix Linear Advance with TMC drivers so many months ago...

I'll quickly post a PR to increase the  priority of the servo timer, so that people can try it out. A hardware timer solution would be better, but for now this would eliminate the leveling issues caused by unreliable PWM output. If additional time will be invested into it, it's probably best to do it in HAL/STM32 as an extra incentive for moving boards to that HAL.

sjasonsmith, 2020-07-13:
I have posted my PR, so that people can pull it down and see whether it fixes their issues. I look forward to hearing feedback on it.
https://github.com/MarlinFirmware/Marlin/pull/18637

Dracrius, 2020-07-13:
> Best would be to drop BL-Touch support because of being closed source - and my traditional sermon about patents and so on.
@AnHardt I really hope this was mostly a joke because otherwise you'd make ALOT of user very very upset. The bltouch is THE abl unless I'm crazy. It is well made well documented and as far as I can tell the number 1 suggested ABL solution simply based on the amount of material about it in comparison to other methods! It is also more accurate then many other abl options I'd pursonally never install one of those EZABL's. 

You've got me quite concerned as firstly I'm pretty sure this would effect any ABL that uses Servo pins not just BLtouch and secondly a solution is obviously being made without the need to see their source, so the statment kind of seems uncalled for!

Dracrius, 2020-07-13:
@sjasonsmith I'll happily give it a try but based on these discussions is Adaptive Step Smoothing actual a setting worth using. Unless i misunderstand your conversation it sounds like it "breaks" other things such as 

>`ADAPTIVE_STEP_SMOOTHING` currently can occupy 100% of processor power for the stepper-interrupt alone by stepping at the wrong speed.

So I'd happily test if this fixed probing but from what i understand Adaptive Smoothing just causes problems in its current state as I can only imagine that kind of behavior is not intended!

[17 more replies]