MarlinFirmware/Marlin #18603: [BUG] Jnnction Deviation

ShadowOfTheDamn, 2020-07-09:
### Bug Description
 
in planner.cpp line 2335 Junction deviation junction_cos_theta variable seems to be calculated with extruder speed.
1- why?
2- when you add a 4th element to a 3D vector then dot product is not equal to any 3D trigonometry function we know, and thus I think this is why enabling Junction Deviation and Linear Advance will result in unknown behavior of the extruder path. even if you are gonna use the extruder speed corrections from the linear advance in junction deviation function I think you need to add that speed to the entrance and exit velocity and then do the calculations for a 3D vector.

please tell me why I'm wrong.
regards

ellensp, 2020-07-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-09 by ellensp

qwewer0, 2020-07-09:
@ShadowOfTheDamn You might want to comment on https://github.com/MarlinFirmware/Marlin/issues/17342, @xda-bam might can answer your question.