MarlinFirmware/Marlin #18604: [BUG] Z_SAFE_HOMING dont work if Y_HOME_DIR = 1 (CoreXY BLV mgn cube)

wolfgangmauer, 2020-07-09:
### Bug Description
On home Z the x-carriage do not move to the middle of the bed, stops  @ X_MIN, Y_MAX

#define Y_HOME_DIR 1
#define Z_SAFE_HOMING
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4897193/Marlin.zip)

### Steps to Reproduce

#define Y_HOME_DIR 1
#define Z_SAFE_HOMING
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

thisiskeithb, 2020-07-09:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

wolfgangmauer, 2020-07-09:
The same behavior with the bugfix-2.0.x (TMC 0.7.1)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4897695/Marlin.zip)

wolfgangmauer, 2020-07-09:
G34 also fails if Y_HOME_DIR = 1 !!!

ellensp, 2020-07-09:
could you attach the top of your build log?  I would like to see version of frameworks, libraries etc, thanks.

ellensp, 2020-07-09:
you have USE_YMIN_PLUG and USE_YMAX_PLUG defined, Is there a reason i've missed for this?

wolfgangmauer, 2020-07-09:
I tryed YMIN and YMAX, maybe i forgot ;-)

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.4
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.7.1
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode
upload disk:  E:
Checking size .pio\build\LPC1768\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  30.9% (used 10116 bytes from 32736 bytes)
Flash: [====      ]  39.1% (used 185776 bytes from 475136 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed
CURRENT: upload_protocol = mbed
Looking for upload disk...
Use manually specified: E:
Uploading .pio\build\LPC1768\firmware.bin
Firmware has been successfully uploaded.
(Some boards may require manual hard reset)
=============================================================================== [SUCCESS] Took 13.47 seconds ===============================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   SUCCESS   00:00:13.470
LPC1769                   IGNORED
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
================================================================================ 1 succeeded in 00:00:13.470 ================================================================================

Terminal will be reused by tasks, press any key to close it.

ellensp, 2020-07-10:
You don't actually describe the issue. "dont work" is a might vague.
What exactly does it not do when Y_HOME_DIR = 1?

If you home X, does X axis move towards home?  Does it stop?
if you home Y does Y axis move towards home? Does it stop?
If you home Z, 
does X move to X bed size/2?
does Y move to Y bed size/2?        
does Z move towards home? Does it stop?

From your config It looks like you have switch based endstops on X and Y, can you confirm?

Did you remove the jumpers that connect the diag stepper driver pins to the endstop pins?

wolfgangmauer, 2020-07-10:
I have switch based endstops on X and Y and removed the diag jumper.
When i home X movement is correct, stop is correct.
When i home Y movement is correct, stop is correct.

When i home Z movement is incorrect, X AND Y stay in place an do not move to the middle of the bed.
BLTouch is release the pin and z is moving. X/Y still @X_MIN/Y_MAX

If i put the endswitch to Y_MIN and set Y_HOME_DIR = -1 everything works fine.

ellensp, 2020-07-10:
Thanks for the clarification.

wolfgangmauer, 2020-07-10:
If there is something to test, just let me know

ellensp, 2020-07-10:
just attempting to build up a test machine currently to replicate your machine enough...

ellensp, 2020-07-10:
Test machine built. 
Copied as much of your machine as I can..
Ie SKR 1.3, 5 * tmc2209, RRD GLCD, core XY machine.  
I used 12v power supply, but only have and only one Z stepper connected, and no bltouch,  

Confirmed that Y_Home_dir -1 works as expected,
But I cannot get  Y_Home_dir 1 to fail.

Now there are a few things I had to change from your configs
disable bltouch. (I don't have one, I just used a switch) 
disabled AUTO_BED_LEVELING_BILINEAR as no bltouch.
Microstepping 16 -> 32 
bed size 200x200
inverted a few endstops so they worked with my normally closed endstops.
X_MIN_POS 0
Y_MIN_POS 0
Z_MIN_POS 0
X_MAX_POS 200
Y_MAX_POS 200
Z_SAFE_HOMING_X_POINT 50
Z_SAFE_HOMING_Y_POINT 50
disable Z_STEPPER_AUTO_ALIGN (due to my Z only having one stepper)
 
Testing procedure
Used current bugfix and have a endstop on y-min and y-max
Test 1:
Set Y_HOME_DIR -1 recompiled and uploaded.
home X, it homed to X-min
home Y, it homed to Y-min
home Z, it moved to X50,Y50 and homed to z-min

Test2:
Set Y_HOME_DIR 1 recompiled and uploaded.
home X, it homed to X-min
home Y, it homed to Y-max
home Z, it moved to X50,Y50 and homed to z-min

So it could be one of the setting I had to change like the missing bltouch,,,, 

Perhaps you could change a few setting to match mine tests and see if the issue goes away.
[diffs.zip](https://github.com/MarlinFirmware/Marlin/files/4903124/diffs.zip)
Here are all the config changes I made in detail.

wolfgangmauer, 2020-07-10:
I was able to isolate the error.
It seems that it only happens when Y_MIN is negative (calculation error).
If i move the endstop to 235(Bed size) and set Y_MIN_POS 0 everithing works fine.
But for my nozzle cleaning feature i need more space in Y direction....

wolfgangmauer, 2020-07-10:
I'm a little confused with all the tests ;-)
The "error" only occourse when:
USE_YMIN_PLUG is defined AND
USE_YMAX_PLUG is defined BUT only YMAX is used(connected)

#define USE_YMIN_PLUG -> nothing connected
#define USE_YMAX_PLUG -> switch connected

closed 2020-07-10 by wolfgangmauer

brunobasto, 2020-07-29:
I'm running into the same problem. @wolfgangmauer why did you close the issue? Did you find a solution?

wolfgangmauer, 2020-07-29:
My problem was that I defined “USE_YMIN_PLUG” AND “USE_YMAX_PLUG” AND also “Y_MIN_ENDSTOP_INVERTING” was wrong.

Check with M119 if really all your states are right…

brunobasto, 2020-07-29:
> My problem was that I defined “USE_YMIN_PLUG” AND “USE_YMAX_PLUG” AND also “Y_MIN_ENDSTOP_INVERTING” was wrong.
> 
> Check with M119 if really all your states are right…

Ah. Got it. My plugs where fine. The reason my homing wasn't going where I wanted was because I had 

```c++
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_BED_SIZE / 2  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT Y_BED_SIZE / 2  // Y point for Z homing
#endif
```

instead of 

```c++
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (X_BED_SIZE / 2)  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT (Y_BED_SIZE / 2)  // Y point for Z homing
#endif
```

notice the missing parenthesis on the first one. After fixing that it started working

thinkyhead, 2020-07-29:
> notice the missing parenthesis on the first one. After fixing that it started working

I can see no reason why (in the latest code) those parentheses would make any difference.
Are you using an older Marlin? What are your `X_BED_SIZE` and `Y_BED_SIZE` settings?

brunobasto, 2020-07-29:
> I can see no reason why (in the latest code) those parentheses would make any difference.

Me neither. I might be wrong. It's possible that I changed something else and that did the trick, BUT, I see that in the original code whenever they have an expression instead of a literal value, they usually put it between parenthesis.

> Are you using an older Marlin?

No. bugfix-2.0.x

> What are your X_BED_SIZE and Y_BED_SIZE settings?
```c++
#define X_BED_SIZE 140
#define Y_BED_SIZE 190
```

ManuelMcLure, 2020-07-29:
Although you might be using ``bugfix-2.0.x`` you haven't updated your configuration file to match. The current ``bugfix-2.0.x`` ``Configuration.h`` has the following:
```
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing
#endif
```
You need to merge your configuration changes into a newer configuration file format using WinMerge or something similar. This change was made back in April.

thinkyhead, 2020-07-29:
> they usually put it between parenthesis

When Marlin needs to use a `#define` value, since we know it could be `123 > 100 ? X_BED_SIZE - 1 : 123 + 7` with no parentheses, the define is wrapped in some kind of container, usually parentheses, to prevent it from causing trouble. So, if adding parentheses to a define fixes something, it means that the defined value (or something else that uses the defined value) is not properly wrapped up in the code someplace.

> possible that I changed something else and that did the trick

That does seem to probably be the case here.