MarlinFirmware/Marlin #18607: [BUG] MARLIN COMPILE FAILURE FOR BTT SKR PRO 1V1

Cartms1, 2020-07-09:
### Bug Description
So, i have been having a nightmare making marlin compile for this board. I had to do a bunch of modifications to Marlin2.x to get it to compile at all, namely because it refused to accept that I had more than one hotend. I was able to labotomize the sanity checks so I could get around that issue, but overall marlin's support for more than one hotend and the distinction between hotends and extruders is severely lacking. 

[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4899570/Marlin-2.0.x.zip)

Here is the version of marlin 2.x i modified so it could work. This version can't seem to see any fans but fan0, and it will see fan0 regardless of the pin assignment, so it isn't a mechanical problem with the board itself. 

That being said, in an effort to fix this issue, I was told to get the bugfix version of marlin 2x however this version is even harder to make functional on this board, to the point where now i can't even find the sanity checks involved with the thermistors  that are erroneously telling me i don't have two hotends. So this thing doesn't compile at all, and as it actually wipes all pin associations with the screen, i'm pretty sure it wouldn't even function anyways. I am really concerned that Marlin's compatibility has gotten markedly worse, especially when it didn't work to begin with.

[bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4899565/bugfix-2.0.x.zip)

ellensp, 2020-07-10:
You have 3 extruders but only 2 hotends. this is a non standard configuration. 
This should be not listed as a bug, but as a feature request to support this configuration.

Cartms1, 2020-07-10:
I'm sorry, but what does that have to do with the standard firmware not seeing more than one hot end? If I had two hotends and two extruded, I'd be having the same issue.

Cartms1, 2020-07-10:
This isn't a feature request, unless basic functionality is seen as a new feature now.

MoellerDi, 2020-07-10:
I'm not seeing any compile failure for BTT SKR Pro 1.1 on my end with bugfix-2.0.x. It's working just fine for my printer.
I see your are doing some things in your configuration.h I have never seen before, like:
```cpp
#undef HOTENDS
#define HOTENDS 2
#undef HOTEND
#define HOTEND 2
```

Anyway, your hardware configuration is not clear to me because you have the following in your configuration.h:
```cpp
#define EXTRUDERS 3
```
with
```cpp
#define SINGLENOZZLE
```
Could you please help me to understand what you kind of hardware you are using? Are you using a single nozzle with 2 hot-ends and 3 extruders?

I get the following error when I try to compile your configuration as is:
```cpp
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
```
as Marlin understood from your ```#define SINGLENOZZLE```that there is only 1 single hot-end for that single nozzle.

If I comment it out like so ``` //#define SINGLENOZZLE ```, Marlin just compiles fine (with some warnings).
```cpp
Linking .pio/build/BIGTREE_SKR_PRO/firmware.elf
Checking size .pio/build/BIGTREE_SKR_PRO/firmware.elf
Building .pio/build/BIGTREE_SKR_PRO/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.6% (used 16848 bytes from 196608 bytes)
Flash: [==        ]  22.1% (used 231908 bytes from 1048576 bytes)
```

Cartms1, 2020-07-10:
Yeah, as a short term solution that worked, and the bug fix version didn't mess up the screen so that's good. 

I have one nozzle set up for soluble support, the other for model, the description used the phrase "any mixing nozzle" which I have one, so I had that.

I 100% eat the blame, i should have better understood the text in the firmware.

The heater code was to deal with prior errors i had where the system was arguing with me. But in hindsight i think when yall came up with the configurations you assumed someone would buy a whole new printer for a variation in nozzle rather than have more than one type of nozzle on a single printer, then further assumed a mixing nozzle would only ever be alone, and never with another hotend at the same time.

With this known, I'm probably gonna have a whole wall of bugs when I get all of that setup.

Working in a 3d printing research lab for a few years, this wasn't a strange occurrence and I've had to do a lot of firmware mods in the past.

closed 2020-07-10 by Cartms1

MoellerDi, 2020-07-11:
Well as pointed out by @ellensp
> This should be not listed as a bug, but as a feature request to support this configuration.

From my point of view, there is nothing wrong with raising it as a feature request in order to enable Marlin to support a configuration where one would set up a nozzle mixing nozzle with 2 or more extruders assigned and another mixing/none-mixing nozzle (dedicated extruders and hotend).

The added value would be to enable one to have more that one nozzle sizes available for printing (considering 0.4 nozzle (mixing or not) to print the model and e.g. another 0.6 nozzle (mixing or not) for printing support or infill). Why not.

Cartms1, 2020-07-11:
I don't want to cause any trouble. I'm sure i'd get yelled at with the responses I have had so far, and i'm sure I can figure out a work around on my own. Its a bit disappointing, but then again 3D-printing has always been its own biggest enemy.

A major issue i have had in the past is that trying to run support and model through the same nozzle causes a lot of latency issues related to temperature mismatches. But, if I can lobotomize the code enough I should be able to get the gradient systems working. The #ifdef checks in marlin are pretty obnoxious, and the way the functions are so tightly coupled makes it a total mess. Each G-code should be 100% insular and I should be able to pull or push functions at will. 

Additive as a technology is evolving too fast, and with the possibilities of freezing infills that can eliminate the inner surfaces on the horizon, yall will really need to be able to keep up with the changing tech.