MarlinFirmware/Marlin #18609: Cannot compile when using MKS_MINI_12864 display

axjurado, 2020-07-09:
### Bug Description

Arduino cannot compile Marlin 20.5 bug fix when using MKS_MINI_12468 display. The error found is:

Arduino: 1.8.12 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/src/lcd/dogm/ultralcd_DOGM.cpp:42:0:
sketch/src/lcd/dogm/ultralcd_DOGM.h:195:23: error: 'DOGLCD_CS' was not declared in this scope
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                       ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
sketch/src/lcd/dogm/ultralcd_DOGM.h:195:23: note: suggested alternative: 'DOGLCD_SCK'
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                       ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
sketch/src/lcd/dogm/ultralcd_DOGM.h:195:34: error: 'DOGLCD_A0' was not declared in this scope
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
sketch/src/lcd/dogm/ultralcd_DOGM.h:195:34: note: suggested alternative: 'DOGLCD_SCK'
     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
                                  ^
sketch/src/lcd/dogm/ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Click compile in Arduino.

**Expected behavior:** compile successful

**Actual behavior:** error compiling

#### Additional Information

The compile works fine using the riprap graphical display.
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/4900056/config.files.zip)

ellensp, 2020-07-10:
This simply means that no one has yet put a MKS_MINI_12468 on a BOARD_RUMBA. So it doesn't know what to do with the pins.

ellensp, 2020-07-10:
Try this replacement pins_RUMBA.h (in the attached zip file)
[pins_RUMBA.zip](https://github.com/MarlinFirmware/Marlin/files/4900715/pins_RUMBA.zip)

NB according to https://reprap.org/wiki/MKS_MINI_12864 and https://reprap.org/wiki/RUMBA
the two plugs EXP1 and EXP2 need swapped.  Ie you need to plug in EXP1 on the back of the lcd into EXP2 on the rumba and the EXP2 on the back of the lcd into EXP1 on the rumba

Also the plugs may need to be turned around 180 degrees..     one image show the rumba has keys up top, another at the bottom. You need to verify that the two left most pins on exp 2 which are 5v and gnd, go to 5v and gnd on your lcd.

RUMBA EXP1 and EXP2 connectors.

```
     /*              _____                              _____
     *    (KILL) 46 |10 9 | GND                     5V |10 9 | GND
     *        RESET | 8 7 | 49 (SD DET)    (LCD D7) 40 | 8 7 | 41 (LCD_D6)
     *    (MOSI) 51   6 5 | 12 (BTN_EN1)   (LCD_D5) 38   6 5 | 18 (LCD D4)
     * (SD CSEL) 53 | 4 3 | 11 (BTN_EN2)   (LCD_RS) 19 | 4 3 | 42 (LCD_ENA)
     *     (SCK) 52 | 2 1 | 50 (MISO)     (BTN_ENC) 43 | 2 1 | 44 (BEEPER)
     *               -----                              -----
     *                EXP1                               EXP2
     */

```

ellensp, 2020-07-12:
Can you try this please and get back to us.

axjurado, 2020-07-12:
Thanks. I was able to compile the firmware using
#define DOGLCD_CS 38
#define DOGLCD_A0 41

However, once compiled I am seeing issues with the display as it becomes unstable. The screen goes blank and the pixels are in the wrong place too. From what I can tell it is a general issue and it is different depending on the board. Doesn't seem to be fixed as the posts were form a year ago and supposedly are already incorporated into Marlin, but again it still having issues.

Is there any fix fo this issue?

ellensp, 2020-07-12:
look into adding.

#define BOARD_ST7920_DELAY_1 DELAY_NS(0)
#define BOARD_ST7920_DELAY_2 DELAY_NS(63)
#define BOARD_ST7920_DELAY_3 DELAY_NS(125)

axjurado, 2020-07-12:
thanks. without the delays worked better. I have added all sort of combinations, without any luck (including this one)

boelle, 2020-07-20:
@axjurado are you still using arduino ide? try vscode+platofrom.io

axjurado, 2020-07-25:
Thanks but I gave up on this display and return it. Might be an issue with my setup.

I tried this display with small cables connecting directly to the board and worked reasonably well - although never had it running for more than a few minutes at the time. However, mounting on the machine with longer cables the display works for a little bit and then is all blank. Not sure if it is the cable per se or the cable+board combo.

In any case thanks for your help!

closed 2020-07-25 by axjurado