MarlinFirmware/Marlin #18610: [BUG] Ender 3 v2 (CREALITY4) EEPROM not working

CRCinAU, 2020-07-10:
### Bug Description
In bugfix-2.0.x branch, `pins_CREALITY4.h` currently has:

```
/* I2C */
#define IIC_BL24CXX_EEPROM                        // EEPROM on I2C-0
//#define E2END 0x3FFF                            // 16Kb (24c16)
#define IIC_EEPROM_SDA                      PA11
#define IIC_EEPROM_SCL                      PA12

// SD EEPROM was in your original build, so...
#define SDCARD_EEPROM_EMULATION
```
Given these diagrams, the I2C EEPROM config looks correct:
[image]
[image]

However, if I disable `SDCARD_EEPROM_EMULATION`, then an M500 will return 'No EEPROM'.

Is there something else that needs to be changed? From what I can tell, the EEPROM is configured correctly, with the correct pins...

thisiskeithb, 2020-07-10:
Have you verified that there’s an EEPROM chip on the board?

It seems odd that they’d include one and then default to SD card-based EEPROM (unless onboard didn’t work after boards were manufactured??)

CRCinAU, 2020-07-11:
The two diagrams were supplied to me by Creality - so I doubt that the information is incorrect.

I'm trying to get independent verification that the chip is present however...

thisiskeithb, 2020-07-11:
> The two diagrams were supplied to me by Creality - so I doubt that the information is incorrect.

It’s not unheard of for diagrams or documentation provided by a manufacturer to be incorrect, so that’s why I asked if it had been physically verified to have an EEPROM installed🙂

CRCinAU, 2020-07-13:
[image]

U11 on the board.

CRCinAU, 2020-07-13:
Datasheet: https://datasheet.lcsc.com/szlcsc/1810010618_BL-Shanghai-Belling-BL24C16A-RRRC_C191945.pdf

It's the 5 lead TSOT23-5 package.

thinkyhead, 2020-07-13:
Note that the EEPROM is in the display unit, not on the main board.

thinkyhead, 2020-07-13:
Creality3D never got it working either. So perhaps there is a flaw in the board…?

thisiskeithb, 2020-07-13:
> Creality3D never got it working either. So perhaps there is a flaw in the board…?

That was my (second) thought. Bummer since emulated EEPROM eats up precious space on those 256k chips (when flashed to spec 🙂)

thinkyhead, 2020-07-13:
The source and original `firmware.bin` that Creality provided me was using SD-based EEPROM, and I never got the BL24CXX working. It might be worth debugging. I was formerly under the impression the EEPROM was on the display, because they put the BL24CXX code in the `lcd/` folder.

thinkyhead, 2020-07-13:
I suppose this `IIC_BL24CXX_EEPROM` could be a common and typical unit that would work with `I2C_EEPROM`…. I'll have to compare its code to the I2C EEPROM code we're using elsewhere.

CRCinAU, 2020-07-14:
@thinkyhead - I'm trying to get a high-res picture of the mainboard that confirms U11 on the mainboard is actually the EEPROM... Sadly I don't have a physical unit to look at myself... My current belief is that the U11 pictured above is the EEPROM - and that it doesn't live on the screen...

In the meantime, is it worthwhile shifting it to a flash based emulation instead of using the SD Card?

CRCinAU, 2020-07-14:
Hmmm - as a side note, after the cleanup of `pins_CREALITYV4.h`, I get the following warnings:

```
.pio/build/STM32F103RET6_creality/src/src/gcode/lcd/M145.cpp.o: In function `GcodeSuite::M145()':                                                                                                                                             
/home/marlin/git/Marlin/Marlin/src/gcode/lcd/M145.cpp:54: warning: undefined reference to `MarlinUI::material_preset'                                                                                                                         
.pio/build/STM32F103RET6_creality/src/src/gcode/temp/M104_M109.cpp.o: In function `GcodeSuite::M104()':                                                                                                                                       
/home/marlin/git/Marlin/Marlin/src/gcode/temp/M104_M109.cpp:113: warning: undefined reference to `MarlinUI::material_preset'                                                                                                                  
.pio/build/STM32F103RET6_creality/src/src/gcode/temp/M104_M109.cpp.o: In function `GcodeSuite::M109()':
/home/marlin/git/Marlin/Marlin/src/gcode/temp/M104_M109.cpp:198: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/gcode/temp/M106_M107.cpp.o: In function `GcodeSuite::M106()':
/home/marlin/git/Marlin/Marlin/src/gcode/temp/../parser.h:277: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/gcode/temp/M140_M190.cpp.o: In function `GcodeSuite::M140()':
/home/marlin/git/Marlin/Marlin/src/gcode/temp/M140_M190.cpp:87: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/gcode/temp/M140_M190.cpp.o:/home/marlin/git/Marlin/Marlin/src/gcode/temp/M140_M190.cpp:136: warning: more undefined references to `MarlinUI::material_preset' follow
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `lcd_select_language()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:151: warning: undefined reference to `BL24CXX::read(unsigned short, unsigned char*, unsigned short)'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `set_english_to_eeprom()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:161: warning: undefined reference to `BL24CXX::write(unsigned short, unsigned char*, unsigned short)'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `set_chinese_to_eeprom()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:166: warning: undefined reference to `BL24CXX::write(unsigned short, unsigned char*, unsigned short)'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `HMI_ETemp()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:1122: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `HMI_BedTemp()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:1169: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `HMI_FanSpeed()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:1216: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `HMI_Prepare()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:2163: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o: In function `HMI_Temperature()':
/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:2465: warning: undefined reference to `MarlinUI::material_preset'
.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/dwin.cpp.o:/home/marlin/git/Marlin/Marlin/src/lcd/dwin/dwin.cpp:2535: warning: more undefined references to `MarlinUI::material_preset' follow
```
I'll admit, I'm ignorant to how the DWIN parts work - but it looks like there are calls to the BL24CXX functions... I'm not sure if this is just to write the current settings to the EEPROM (an M500 type thing?) or something different...

I'm also starting to wonder if @thinkyhead was correct in the assumption that there is an EEPROM on the screen as well or instead of the mainboard?

I've asked Creality for further information - and we'll see what comes out of that - in the meantime, I'm trying to get someone with physical access to the Ender 3 V2 mainboard to take some high-res pictures of the chip marked U11 on the mainboard - which is supposed to be the EEPROM...

The plot thickens!

CRCinAU, 2020-07-14:
Had a lovely redditor send me this - of the chip in question on the mainboard... 
[image]

CRCinAU, 2020-07-14:
Working on more colab investigation.... with the stock Creality firmware (`Ender-3 V2-Marlin-2.0.1 - V1.0.1.bin`) and no SD card inserted:
```
Send: M500
Recv: echo:No EEPROM.
Recv: ok
```

This would seem to indicate that they're ONLY using the EEPROM to store settings from the screen - and none of Marlin's settings.

thinkyhead, 2020-07-15:
> Had a lovely redditor send me this - of the chip in question on the mainboard...

Aw, I could've done that. I've had the Ender 3 V2 on my work desk all week….

thinkyhead, 2020-07-15:
> This would seem to indicate that they're ONLY using the EEPROM to store settings from the screen - and none of Marlin's settings.

That is correct. Now that you mention it, I was planning to make a proper `BL24CXX` variation of the EEPROM under the `HAL/STM32F1` but just never got around to it. Then I'd need to also set aside a section of it for the display.

CRCinAU, 2020-07-15:
You're doing one better than me - I've had to figure it all out by trial and error on other peoples hardware :laughing:

If it's only a single byte that Creality use to set the screen language, would be make more sense to bring the EEPROM into a standard i2c type EEPROM, and store the screen settings in with the Marlin settings?

Or as we know its a 16Kb chip, use the last 4Kb for Marlin, and the beginning for the screen?

I'm not sure how feasible that would be - but I assume it'd be something like the flash eeprom emulation? I tried that with the Ender 3 V2 - but couldn't get the firmware to boot afterwards - I'm guessing something with the screen or EEPROM conflicting...

thinkyhead, 2020-07-20:
> would be make more sense to bring the EEPROM into a standard i2c type EEPROM

The EEPROM chip on the board is IIC and requires an additional EEPROM style to be added to the HAL. I'm still working on it, and I should have it together real soon now.

closed 2020-07-20 by thinkyhead