MarlinFirmware/Marlin #18611: Marlin Compiling Issue with Printrboard

penpenLoveUwU, 2020-07-10:
### Bug Description
`Compiling .pio\build\at90usb1286_dfu\lib901\ADC\ADC.cpp.o
In file included from C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:39:0,
                 from C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:59,
                 from C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\adc.cpp:30:
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC/settings_defines.h:60:2: error: #error "Board not supported!"
 #error "Board not supported!"
  ^
Archiving .pio\build\at90usb1286_dfu\libf90\libEEPROM.a
In file included from C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:59:0,
                 from C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\adc.cpp:30:
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:112:23: error: 'ADC_REFERENCE' 
has not been declared
     void setReference(ADC_REFERENCE ref_type);
                       ^
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:430:20: error: 'ADC_INTERNAL_SOURCE' has not been declared
     int analogRead(ADC_INTERNAL_SOURCE pin) __attribute__((always_inline))
                    ^
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:757:11: error: 'IRQ_NUMBER_t' does not name a type
     const IRQ_NUMBER_t IRQ_ADC; // IRQ number
           ^
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h: In member function 'void ADC_Module::continuousMode()':
C:\Users\11358\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:250:9: error: 'setBitFlag' is not a member of 'atomic'
         atomic::setBitFlag(adc_regs.SC3, ADC_SC3_ADCO);
         ^
compilation terminated due to -fmax-errors=5.
Archiving .pio\build\at90usb1286_dfu\lib0c3\libSlowSoftI2CMaster.a
*** [.pio\build\at90usb1286_dfu\lib901\ADC\ADC.cpp.o] Error 1`
### My Configurations
[image]
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4900317/Configurations.zip)
(I didn't change anything besides the MOTHERBOARD name)
### Steps to Reproduce

1. [First Step] set MOTHERBOARD and environment
2. [Second Step] Build
3. [and so on...]

**Expected behavior:** [What you expect to happen] compile successfully

**Actual behavior:** [What actually happens] compile error

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-07-10:
I can confirm that I can build my own Printrbot Play configuration from March 15, but it breaks if I rebase it onto bugfix-2.0.x.

I thought it might have been due to a platform upgrade, but my old branch was also using the updated platform.

sjasonsmith, 2020-07-10:
This appears to have been caused by the following commit, which adds the `Marlin` folder to the include search path. I'm not sure yet exactly what that is causing to go wrong.

https://github.com/MarlinFirmware/Marlin/commit/67b2b78717ed1dfd1950872832e94a11e77357a8

sjasonsmith, 2020-07-10:
It seems that adding `Marlin` as the default include folder is causing it to detect library usage from other HALs, so it is compiling in libraries which are incompatible with the board being used.

I found that if I delete the `Marlin/HAL/LPC1768` folder the build will succeed, so there is something very strange going on when that include folder is updated.

@penpenLoveUwU, for now just delete the following line at the top of the `platformio.ini` file. If you get a warning about an `include` folder not existing, just ignore it.

`include_dir  = Marlin`

penpenLoveUwU, 2020-07-10:
Hi, thank you so much @sjasonsmith ! However, after deleting that line, I get a new error:
`Archiving .pio\build\at90usb1286_dfu\libaf1\libU8glib-HAL_ID1932.a
Linking .pio\build\at90usb1286_dfu\firmware.elf
avr-g++: error: CreateProcess: No such file or directory
*** [.pio\build\at90usb1286_dfu\firmware.elf] Error 1`

I also tried Re-ARM RAMPS and it turns out to be fine. Not sure why this happens to Printrboard:) Again, thank you so much!

sjasonsmith, 2020-07-10:
Try renaming the environment in platformio.ini to something shorter. For example, change `at90usb1286_dfu` to `at90`.

A whole bunch of additional source files were added recently to to support a new type of LCD screen. This causes path length limits to be exceeded when linking them together at the end.

There are probably some other ways to avoid this, but I think renaming the environment is the easiest.

penpenLoveUwU, 2020-07-10:
Thanks so much, it works!

Fixx, 2020-07-14:
I also cannot compile newer Marlin bugfix-2.0.x via Arduino after configuration(can compile on stock). I only deleted Marlin/HAL/LPC1768 folder and it compiles.

thinkyhead, 2020-07-15:
> It seems that adding Marlin as the default include folder is causing…

Interesting. The thing is that the "`Marlin`" folder is automagically added to `.vscode/c_cpp_properties.json` when the VSCode "PlatformIO IDE" extension interprets the `platformio.ini` file. Without the added line `include_dir  = Marlin`, VSCode adds both "`Marlin`" and "`Marlin/include`" to `.vscode/c_cpp_properties.json`, which leads to the (harmless) warning about the missing `include` folder.

I would like to eliminate that warning because it generates _constant_ posts to the support groups.

thinkyhead, 2020-07-15:
> avr-g++: error: CreateProcess: No such file or directory

That's a spurious error that I see pretty frequently on a first build attempt. The next "Build" always succeeds.

closed 2020-07-15 by thinkyhead