MarlinFirmware/Marlin #18613: Visual Studio Code build with Anet A8 failed

bryanshaw46, 2020-07-10:
### Bug Description

The terminal window says: Compiling .pio\build\sanguino1284p\src\src\module\tool_change.cpp.o
Marlin\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope  
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
Marlin\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope  
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^  
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^
*** [.pio\build\sanguino1284p\src\src\module\motion.cpp.o] Error 1
=============================== [FAILED] Took 111.89 seconds ===============================

### My Configurations

Anet A8
[Marlin_Anet_A8_Config.zip](https://github.com/MarlinFirmware/Marlin/files/4902529/Marlin_Anet_A8_Config.zip)
[image]

### Steps to Reproduce

1. Downloaded configuration zip file from https://github.com/MarlinFirmware/Configurations
2. Unzipped, went to configuration example folder, Anet, A8 and copied attached files to C:\Marlin2_0\Marlin-2.0.x\Marlin, replacing the two existing folders. 
3. Download and install latest Visual Studio Code (VSC)
4. Install PlatformIO extension (PIO) and Auto Build Marlin extension (ABM)
5. On left panel of VSC, click on ABM and then click on the Hammer Button

**Expected behavior:** [What you expect to happen]
Build to succeed

**Actual behavior:** [What actually happens]
The above error

#### Additional Information
I am Bryan Shaw, I am a donor and my email address is bryanshaw46@yahoo.com

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-07-10:
Not a bug. Your using an incorrect version of the config files for the version of Marlin your using.
Your config file is newer than your code expects.
In Particular X_HOME_BUMP_MM, Y_HOME_BUMP_MM and Z_HOME_BUMP_MM where replaced with HOMING_BUMP_MM in newer config files. 

See Issue 3 in https://reprap.org/forum/read.php?415,874307

closed 2020-07-10 by ellensp

bryanshaw46, 2020-07-10:
Thank you!

 
 

Closed #18613.