MarlinFirmware/Marlin #18614: ADC_BUTTON_DEBOUNCE_DELAY has no effect

bcdiaconu, 2020-07-10:
### Bug Description

Whatever value I set for `ADC_BUTTON_DEBOUNCE_DELAY` it will not make any change in buttons debounce. I have tried `256`, `512` and `1024`. The buttons repreat at same (human distinguishable) rate.

### My Configurations

Configs for bugfix_2.0.x branch:
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h
https://github.com/bcdiaconu/Marlin/blob/bugfix-2.0.x/Marlin/ZonestarP802Qs.h

### Steps to Reproduce
1. set `ADC_BUTTON_DEBOUNCE_DELAY ` to any value
2. build
3. write firmware to board
4. press and hold center button and whatch enter/exit frequency

**Expected behavior:** [What you expect to happen]

Setting `ADC_BUTTON_DEBOUNCE_DELAY` to a higher value to have a lower enter/exit menu frequency when keeping center button pressed.
Setting `ADC_BUTTON_DEBOUNCE_DELAY` to a lower value to have a higher enter/exit menu frequency when keeping center button pressed.

**Actual behavior:** [What actually happens]

Changing `ADC_BUTTON_DEBOUNCE_DELAY` has no effect on enter/exit menu frequency when keeping center button pressed.

ellensp, 2020-07-11:
Please read https://stackoverflow.com/questions/25991367/difference-between-throttling-and-debouncing-a-function#:~:text=Throttling%20will%20delay%20executing%20a,event%20that%20fires%20multiple%20times.

What you want is not debounce

ellensp, 2020-07-11:
try changing #define ADC_MIN_KEY_DELAY 100  to a larger number in ultralcd.cpp

bcdiaconu, 2020-07-15:
In this case, I don't know if it si implemented, but it would nice to have for ADC keys contextual behavior. It should be able to interpret a long stroke as a single command (no matter how long the stoke or have different ADC_MIN_KEY_DELAY value) while in some menu items like setting Z probe offset or moving the axes to be a repetitive action while button is being pressed with a lower ADC_MIN_KEY_DELAY. 

And another thing would be to add ADC_MIN_KEY_DELAY to Config_adv.h file wehre the other ADC keys related defines are.

closed 2020-08-20 by github-actions[bot]