MarlinFirmware/Marlin #18616: [Q] UBL Bed Topography

qwewer0, 2020-07-10:
### Bug Description

Without loading the stored UBL mesh, G29 J<`GRID_MAX_POINTS_X`> (J7) gives way different results than with UBL mesh loaded.

**G29 J7, G29 T, without UBL mesh loaded:**

   ( 10,220)                                      (222,220)
       0       1       2       3       4       5       6      
6 | +0.147  +0.147  +0.146  +0.145  +0.144  +0.144 [+0.143]
  |
5 | +0.143  +0.142  +0.142  +0.141  +0.140  +0.139  +0.139 
  |
4 | +0.139  +0.138  +0.137  +0.136  +0.136  +0.135  +0.134 
  |
3 | +0.134  +0.133  +0.133  +0.132  +0.131  +0.131  +0.130 
  |
2 | +0.130  +0.129  +0.128  +0.128  +0.127  +0.126  +0.125 
  |
1 | +0.125  +0.125  +0.124  +0.123  +0.123  +0.122  +0.121 
  |
0 | +0.121  +0.120  +0.120  +0.119  +0.118  +0.117  +0.117 
       0       1       2       3       4       5       6      
   ( 10, 10)                                      (222, 10)
[image]

**G29 J7, G29 T, with UBL mesh loaded:**

   ( 10,220)                                      (222,220)
       0       1       2       3       4       5       6   
6 | +0.289  +0.231  +0.204  +0.146  +0.175  +0.124 [+0.250]
  |
5 | +0.197  +0.156  +0.140  +0.096  +0.133  +0.092  +0.245 
  |
4 | +0.158  +0.124  +0.100  +0.047  +0.083  +0.030  +0.166 
  |
3 | +0.113  +0.080  +0.073  +0.032  +0.079  +0.038  +0.166 
  |
2 | +0.141  +0.118  +0.106  +0.068  +0.099  +0.053  +0.187 
  |
1 | +0.137  +0.113  +0.094  +0.051  +0.100  +0.069  +0.207 
  |
0 | +0.245  +0.189  +0.177  +0.121  +0.160  +0.137  +0.275 
       0       1       2       3       4       5       6   
   ( 10, 10)                                      (222, 10)
[image]

Is this how it suppose to work?

### My Configurations

Ender 3, SKR Mini E3 v1.2, BLTouch v3.1
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4904900/Configuration.zip)

### Steps to Reproduce

1. Start printer
2. M420
3. G29 T
4. G29 J<`GRID_MAX_POINTS_X`>
5. G29 T (Compare)
6. M420 L0 S0
7. G29 J<`GRID_MAX_POINTS_X`>
8. G29 T (Compare)

<details>
  <summary>Steps in detail</summary>

SENT: M420
READ: echo:Bed Leveling OFF
Bed Leveling OFF
READ: echo:Fade Height 10.00
Fade Height 10.00
READ: ok
SENT: G29 T
READ: //action:notification Doing G29
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,220)                                      (222,220)
READ:         0       1       2       3       4       5       6      
READ:  6 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  5 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  4 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  3 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  2 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  1 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:    |
READ:  0 |  0.000   0.000   0.000   0.000   0.000   0.000   0.000 
READ:         0       1       2       3       4       5       6      
READ:     ( 10, 10)                                      (222, 10)
READ: 
SENT: G29 J7
READ: //action:notification Doing G29
READ: Tilting mesh point 1/49
READ: 
READ: Tilting mesh point 2/49
...
READ: Tilting mesh point 49/49
READ: 
READ: X:264.8600 Y:226.0700 Z:7.5000 E:0.0000 Count X:21189 Y:18086 Z:3000
SENT: G29 T
READ: //action:notification Doing G29
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,220)                                      (222,220)
READ:         0       1       2       3       4       5       6      
READ:  6 | +0.147  +0.147  +0.146  +0.145  +0.144  +0.144 [+0.143]
READ:    |
READ:  5 | +0.143  +0.142  +0.142  +0.141  +0.140  +0.139  +0.139 
READ:    |
READ:  4 | +0.139  +0.138  +0.137  +0.136  +0.136  +0.135  +0.134 
READ:    |
READ:  3 | +0.134  +0.133  +0.133  +0.132  +0.131  +0.131  +0.130 
READ:    |
READ:  2 | +0.130  +0.129  +0.128  +0.128  +0.127  +0.126  +0.125 
READ:    |
READ:  1 | +0.125  +0.125  +0.124  +0.123  +0.123  +0.122  +0.121 
READ:    |
READ:  0 | +0.121  +0.120  +0.120  +0.119  +0.118  +0.117  +0.117 
READ:         0       1       2       3       4       5       6      
READ:     ( 10, 10)                                      (222, 10)
READ: 	
SENT: M420 L0 S0
READ: Mesh loaded from slot 0
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,220)                                      (222,220)
READ:         0       1       2       3       4       5       6      
READ:  6 | +0.223  +0.167  +0.143  +0.087  +0.119  +0.070  +0.199 
READ:    |
READ:  5 | +0.139  +0.100  +0.087  +0.046  +0.084  +0.046  +0.202 
READ:    |
READ:  4 | +0.107  +0.076  +0.055  +0.004  +0.043  -0.008  +0.130 
READ:    |
READ:  3 | +0.070  +0.039  +0.036  -0.003  +0.046  +0.007  +0.139 
READ:    |
READ:  2 | +0.106  +0.085  +0.076  +0.040  +0.074  +0.030  +0.167 
READ:    |
READ:  1 | +0.109  +0.088  +0.072  +0.031  +0.082  +0.054  +0.195 
READ:    |
READ:  0 | +0.225  +0.171  +0.163  +0.109  +0.150  +0.129  +0.271 
READ:         0       1       2       3       4       5       6      
READ:     ( 10, 10)                                      (222, 10)
READ: 
READ: Mesh is valid
READ: Storage slot: 0
READ: echo:Bed Leveling OFF
Bed Leveling OFF
READ: echo:Fade Height 10.00
Fade Height 10.00
READ: X:264.8600 Y:226.0700 Z:7.5010 E:0.0000 Count X:21189 Y:18086 Z:3000
READ: ok
SENT: G29 J7
READ: //action:notification Doing G29
READ: Tilting mesh point 1/49
READ: 
READ: Tilting mesh point 2/49
...
READ: Tilting mesh point 49/49
READ: 
READ: X:264.8600 Y:226.0700 Z:7.4360 E:0.0000 Count X:21189 Y:18086 Z:3001
SENT: G29 T
READ: //action:notification Doing G29
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,220)                                      (222,220)
READ:         0       1       2       3       4       5       6      
READ:  6 | +0.289  +0.231  +0.204  +0.146  +0.175  +0.124 [+0.250]
READ:    |
READ:  5 | +0.197  +0.156  +0.140  +0.096  +0.133  +0.092  +0.245 
READ:    |
READ:  4 | +0.158  +0.124  +0.100  +0.047  +0.083  +0.030  +0.166 
READ:    |
READ:  3 | +0.113  +0.080  +0.073  +0.032  +0.079  +0.038  +0.166 
READ:    |
READ:  2 | +0.141  +0.118  +0.106  +0.068  +0.099  +0.053  +0.187 
READ:    |
READ:  1 | +0.137  +0.113  +0.094  +0.051  +0.100  +0.069  +0.207 
READ:    |
READ:  0 | +0.245  +0.189  +0.177  +0.121  +0.160  +0.137  +0.275 
READ:         0       1       2       3       4       5       6      
READ:     ( 10, 10)                                      (222, 10)
READ: 

</details>

thisiskeithb, 2020-07-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-10 by thisiskeithb

qwewer0, 2020-07-10:
If this is how it suppose to work, then ok, could have said that, but if it is not then the question title changes to bug...

thierryzoller, 2020-07-10:
I don't think this was a support question, I think qwewer0 was pointing out what he believes is not the way it is supposed to work (i.e bug) by asking the rethorical question whether it is supposed to work this way.

qwewer0, 2020-07-10:
Yes, I could have made the title [BUG] UBL Bed Topography, but I'm not sure that it is a bug or not.
I admit my mistake, next time I want to open an issue I will do that as [BUG].

Dracrius, 2020-07-10:
@qwewer0 I really would suggest reading more into ubl.
The links thisiskeithb should help but you can start at [Marlins UBL wiki page](https://marlinfw.org/docs/features/unified_bed_leveling.html) as I have honestly never used it (dont really see the need with the bltouches repeatability and my glass bed) but after looking it up that page list a VERY specific setup procedure that meantions at least a few steps that will cause issues if skipped.

qwewer0, 2020-07-10:
@Dracrius I started with that wiki and set up everything perfectly, and I can use UBL for prints just right.

Dracrius, 2020-07-10:
@qwewer0 and @thierryzoller no offence but you both miss the point of github issues on such a massive open source project. This has become obvious in this issue and #17262. Firstly calling it [Bug] would not change the fact that it is a question not an actual issue. Github is not for questions or real issues would get lost in the sea of questions (and some times still do) especially at current with the amount of ABL issues adding what may be normal behaviour will just cause confusion for everyone. They have had a discord for a little while now, yes it is new but Discord is the place for questions or first step trouble shoting if you are not certain if it is even an ISSUE!

Secondly issues can not be left indefinitly if unsolved and lacking response for three very simple reasons, some are just user error (most likely to never come back to say anything when they realize their mistake) it is imposible to solve an issue if no one with the problem is around to test the fixes. Others get solved by unrelated patches for instance if noone realized Adaptive Smoothing was causing ABL issues apon its update in lets guess about a month because it was already needed MANY ABL issues would have likely been abandoned by the reporter by then (as this has already happend in two abl issues i added replys to in the last few days). Do you honestly expect them to keep those reports indefinitly despite them likely being solved since new people are not adding to it.

You will notice as one person joked the Boelle "bot" (pretty sure its an actual person haha) who pokes all abandoned issues before they personally try and close them. They can be annoying but they do it so as long as your paying attention to your own issue you can confirm it is still an issue before it gets closed without needing to ping it every few days manually or with a bot.

This reply in and of itself does not belong in an issue but i felt the need to clarify the use case of github issues to avoid future hostilities due to misunderstanding.

Dracrius, 2020-07-10:
@qwewer0 i am also quite confused why you made an issue then if it prints properly. You will most likely find Discord faaaar more helpful and responsive to this kind of question/disscussion!

qwewer0, 2020-07-10:
@Dracrius I didn't opened this issue because I have problem with UBL, but because I think that something isn't right with some internal UBL or other code that causes the G29 J bed measurements to use the the stored mesh data without the "printer knowing it".
In my understanding if there isn't a loaded mesh, so all the mesh points are zero, then G29 J should use the actual measured values without comparing it to the stored mesh. If it would be the case, then my first G29 J7, G29 T would show close if not the same mesh as the stored mesh data, but it shows an almost completely flat bed, and my bed isn't flat at all.

MoellerDi, 2020-07-10:
Well best to my very limited knowledge (I just started to discover the code base) and according to [Marlins UBL wiki page](https://marlinfw.org/docs/features/unified_bed_leveling.html), ```G29 J7``` will just tilt the mesh (even an all zero mesh, let's call it a plane) according to what it finds. IMHO ```G29 J7``` will (IMHO) not generate a new mesh for you. That's maybe why you see an almost completely flat (but tilted) bed if you execute it w/o loaded and valid mesh data.

qwewer0, 2020-07-10:
@MoellerDi Ok, thanks, so it doesn't twist the mesh just tilt it? Then a J1 is equal to J7?

Dracrius, 2020-07-11:
@qwewer0 from the (G29 Bed Level Unified)[https://marlinfw.org/docs/gcode/G029-ubl.html] page.

[J<int>] | Grid (or 3-Point) leveling:  These options calculate a plane and adjust the existing mesh to the bed tilt. If a value is provided, probe a grid with the given number of points, squared. With no value, probe 3 points to find the plane of the bed.

Based on everything I read their MoellerDi is right and all of this is exactly what youd expect to happen. A J 1 would just equal a 1x1 title check vs a 7x7 title check basically if that makes it any clearer.

thierryzoller, 2020-07-11:
Thanks everyone, except @Dracrius who took an estimate of 10 minutes to type a reply that is basically explaining OP and myself for how stupid we are for pointing out a potential bug.

thisiskeithb, 2020-07-11:
Other OSS projects may use their issue queue as a means to answer users questions, but that's not done here because it's a huge time sink.

As I said in my comment before closing, _"This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here."_ and _"For best results getting help with configuration and troubleshooting, please use the following resources: (*snipped list*)."_

It's only then, _"After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose)."_ as to not eat up our time chasing down user questions which should have been asked elsewhere first.

Dracrius, 2020-07-11:
@thierryzoller your commitment to take offence to anything said in your direction is impressive. I spent good time writing that to make it adondently clear that not only do i understand the misunderstanding on your part but that their are reasons not to ask questions or expect bugs to stay open indeffinitly. 

I also went ahead and reasearched @qwewer0 problem myself and gave multiple replies pointing them in the right direction of the exact information they needed in a closed issue. So it is quite clear either other things are going on for you or you just like to take spite and assume when someone tells you a rule its a personel attack. 

At this point I can safely say I never thought either of you were stupid or even wrong to make said mistakes. I only took issue with you take it personal when others tried to tell you the repos guidlines for posting code issues. I dont think a single person so far meant either of you mallis or was even mad yet you continue to lash out.

I will site again the discord where disscussions is kind of the point of said platform. It is also a great place to say hey is this normal? Which was the point of this discussion as this is EXACTLY as intended/ expected behavior.

Marlin has sooo many features its id expect assumed you dont know them all i deffinitley don't so basically the maintainers do ask that if we experimenting with a new one we do our do dilagince on other platforms before attepting to use the issues section of a github as a discussion platform. That is not its intended use case and polutes the humungus issue pool that is already I am sure a nightmare to sort through!

Dracrius, 2020-07-11:
To be clear i have and am just trying to mend bridges because the maintainers are doing there best to follow their own rules and not turn issues into disscussions but you seemed so offended I thought a little polite discussion was needed!