MarlinFirmware/Marlin #18617: error "X_HOME_BUMP_MM" was not declared in this scope

sacy80, 2020-07-10:
got error complie marlin 2.0.x
board = SKR_MINI_V1_1
Printer = Delta

error:
Marlin\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:71: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                       ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:80: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
Marlin\src\module\motion.cpp:77:89: note: suggested alternative: 'HOMING_BUMP_MM'
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }
                                                                                         ^
Marlin\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'
 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);
 ^~~~~~~~~~
*** [.pio\build\STM32F103RC_btt_512K\src\src\module\motion.cpp.o] Error 1
In file included from Marlin\src\module\../inc/../core/boards.h:24:0,
                 from Marlin\src\module\../inc/MarlinConfigPre.h:35,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\planner.h:33,
                 from Marlin\src\module\planner.cpp:65:
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const abce_long_t&, feedRate_t, uint8_t, const float&)':
Marlin\src\module\planner.cpp:2003:62: warning: large integer implicitly truncated to unsigned type [-Woverflow]
             g_uc_extruder_last_move[N] = (BLOCK_BUFFER_SIZE) * 2; \
                                          ~~~~~~~~~~~~~~~~~~~~^~~
Marlin\src\module\../inc/../core/macros.h:433:24: note: in definition of macro 'EVAL1'
 #define EVAL1(V...)    V
                        ^
Marlin\src\module\../inc/../core/macros.h:432:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:431:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))
                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:432:24: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                        ^~~~~
Marlin\src\module\../inc/../core/macros.h:432:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:431:30: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))
                              ^~~~~
[... 65 lines cut ...]
                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:430:30: note: in expansion of macro 'EVAL4'
 #define EVAL8(V...)    EVAL4(EVAL4(V))
                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:429:30: note: in expansion of macro 'EVAL8'
 #define EVAL16(V...)   EVAL8(EVAL8(V))
                              ^~~~~
Marlin\src\module\../inc/../core/macros.h:422:24: note: in expansion of macro 'EVAL16'
 #define EVAL(V...)     EVAL16(V)
                        ^~~~~~
Marlin\src\module\../inc/../core/macros.h:474:33: note: in expansion of macro 'EVAL'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                 ^~~~
Marlin\src\module\../inc/../core/macros.h:459:3: note: in expansion of macro 'ENABLE_ONE_E'
   _RPT_OP(_RPT_I)                                             \
   ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:474:38: note: in expansion of macro '_REPEAT'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                      ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:475:33: note: in expansion of macro 'REPEAT_S'
 #define REPEAT(N,OP)            REPEAT_S(0,N,OP)
                                 ^~~~~~~~
Marlin\src\module\planner.cpp:2015:7: note: in expansion of macro 'REPEAT'
       REPEAT(EXTRUDERS, ENABLE_ONE_E); // (ENABLE_ONE_E must end with semicolon)
       ^~~~~~

GMagician, 2020-07-10:
Maybe old and not updated configuration?

thisiskeithb, 2020-07-10:
> Maybe old and not updated configuration?

Yep.

___

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-10 by thisiskeithb