MarlinFirmware/Marlin #18622: [BUG] Extruder issue (screeching, wrong steps) with TMC2209 in UART on SKR board

Chris0812, 2020-07-11:
### Bug Description

When using Watterott TMC2009 on SKR 1.3 or SKR1.4, the extruder doesn't work: steps/mm setting are completely different from actual extrusion. The extruder is screeching and skipping steps like it does in lin.advance or with way too low current. But even with 1000mA current setting.
Tried different steppers, drivers (SD6128, A4988), cables and Z,E0,E1 driver positions: 
Always only the Extruder motor not working.

Side note: X,Y,Z work perfectly silent and with the correct microsteps.

I also made a clean firmware install without any UART traces, works again fine. Will try tomorrow to maybe find the one line of code in config that was the reason.

### My Configurations

Writing on the phone, will be added later

### Steps to Reproduce

1. Load Firmware with my config (with adjusted E steps)
2. Place any driver on E0 
3. Use extruder motor 
4. Try retraction and extrusion in printing scenario

**Expected behavior**
Audible movement but nothing out of the ordinary

**Actual behavior**
Wrong extrusion values contradictory to the actual E step/mm set
Retracts not possible, only screeching

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AnHardt, 2020-07-11:
> Load Firmware with my config (with adjusted E steps)

Without configuring max-speeds, accelerations, jerk-speed for your extruder it's completely random if your extruder can work or not.

thisiskeithb, 2020-07-12:
I don’t think this is a Marlin bug since you have to modify your driver [as described here](https://github.com/MarlinFirmware/Marlin/issues/14683#issuecomment-513269551). There’s also some additional information in Watterott’s [TMC2209 Notes section](https://learn.watterott.com/silentstepstick/pinconfig/tmc2209/).

I’ve done the above mods on Watterott TMC2209s (and some other hackery to get the DIAG signal working since Watterott moved the pin location) and can confirm they work with Marlin.

Chris0812, 2020-07-12:
I think, I can actually exclude the drivers themselves as an issue but should remain to keep all details here...

I just realized that the factor ~2,4 in the wrong steps/mm appeared (extrusion was 2,4 times the set value) and also that this is the ratio of cubic mm/s Vs linear mm/s ...
Although having volumetric E enabled in firmware and slicer, could here be the problem? Compiling right now.

thisiskeithb, 2020-07-13:
What is the status of this? I can't tell what's being asked anymore or if your issue has been resolved.

shitcreek, 2020-07-13:
Closing because this appears to be a case of user error.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-13 by shitcreek