MarlinFirmware/Marlin #18626: [BUG]TMC2208 with SKR 1.3 LOUD

danis101, 2020-07-12:
### Bug Description

My steppers  are loud, same as on a4988 when using stealthchop, when I turn off stealthchop they are making high pitched sound

### My Configurations

SKR 1.3 + tmc 2208 on x, y, z axis and tmc2209 on E
Frame and motors from creality cr10-s5
Everything on 12V

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4908441/config.zip)

### Steps to Reproduce

1. install TMC2208 from BTT v3
2. setup UART MODE
3. Try to turn on and off stealthchop 
4. Hear the sound

**Expected behavior:** I expected that after upgrading drivers to TMC2208 motors will be less loud but nothing special happened expect that I can change voltages directly from marlin

**Actual behavior:** Motors are actually loud, with stealthchop set to "OFF" in marlin they squeeks and making high  pitched sound that makes my ears bleeding

#### Additional Information

If you need exactly what it souds like I can record it but I hope you trust me without recording

What I actually tried?
- cut off ms3 pin
- Put a jumper like here (this in green circle and blue  https://user-images.githubusercontent.com/3089594/61545986-eeaa0d80-aa48-11e9-8438-4da4395db08a.png)
- after cutting ms3 pin Ive got TMC connection error
When I connected stepsticks as standalone I remember that they were much quieter than with UART
About currents in config - Im lowering it even more in marlin menu just to that point they don't skip steps and stays cool

boelle, 2020-07-12:
you need to run at 24V to get the least noise, Trinamic will say the same thing if you ask them
and stealthchop is the low noise mode and spread cycle is the "normal" mode

danis101, 2020-07-12:
so I cant get less noise on actual  setup with all heaters 12V? Is that normal?
This is with "stealthchop in marlin on" https://www.dropbox.com/s/bfqy1yryx0cqc87/20200712_144128.mp4?dl=0
and this movie shows turning off stealthchop and on and movement with turned off stealthchop from menu - https://www.dropbox.com/s/l2xv08vdpe5v9vf/20200712_144211.mp4?dl=0

danis101, 2020-07-12:
OK my new discover is:
When I set TMC driver current from 650 to 1100 (Motor is rated to 1,5A so it should be safe) and turn off stealthchop the motor isn't squeaking but I dont hear any difference when moving y axis with stealthchop on or off - What is going on?

shitcreek, 2020-07-13:
I'm running the current at 700 for X and Y and 800 for the extruder without any noise of issue. If there is a VREF pot on your stepper driver, you may need to adjust it.

shitcreek, 2020-07-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-13 by shitcreek