MarlinFirmware/Marlin #18627: [QUESTION] What's the purpose of "Bump distribution date" commits?

AlexanderAmelkin, 2020-07-12:
I may be missing something, but to the best of my knowledge:

1. The state of the codebase in git is determined by the hash
2. If there are no changes, the hash remains the same and that allows for telling if two versions of binaries are using the same codebase if the (abbreviated) hash can be visible to the user
4. It is much more valuable to know what codebase was used than to know when a robot last touched the repository
5. When a commit that functionally changes nothing, but only updates the date string is pushed, it creates a new hash and defeats the item 2 above
6. Everyday "Bump distribution date" commits clutter the `git log` output and make it harder to tell what has actually changed between two arbitrary versions
7. This bumping is unique to this project and not seen anywhere else, including such major projects as Linux kernel

I feel like I don't understand the idea behind this bumping. Could anyone please clarify?

P.S. Sorry for hijacking the [BUG] type, but there is no [QUESTION] type issue, and the forum doesn't look like the place where housekeeping questions like this are discussed.

ellensp, 2020-07-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-12 by ellensp

thinkyhead, 2020-07-13:
> I feel like I don't understand the idea behind this bumping. Could anyone please clarify?

- We are sent configurations and Marlin folders all the time with no `.git` folder included and configurations modified by the user.
- Vendors and users fork Marlin all the time. Vendors in particular like to change the version number and other information, make lots of customizations, remove and add files, and so on. Often by the time I get a source sent me to fix up, it has gone through two subcontractors working between Islamabad and Shenzhen.

The most common problem I have when receiving one of these piles of code is, where did it originate? What commit or version or date is the root of this blob? Git (unlike CVS and SVN) provides no $TOKENS$ for this kind of data. And often the person sending me the code is not the person who did the mods.

So, the date bump was implemented to make it easier for me and others to find the starting source and see all customizations in isolation when we locate, or are provided, mysterious Marlin source code.

HTH

AlexanderAmelkin, 2020-07-15:
@thinkyhead, I totally feel your pain!

Git is marvelous but it lacks some features that veteran users of version control systems may be missing. It takes time to adapt, find new ways or sometimes change the mindset and approach. It took me years to switch my way of thinking from svn to git, and I'm still not sure I have switched completely. Switching from cvs to svn was much faster.

May I suggest you do bumps only upon a new commit/merge? You're anyway doing it using github workflows, it should be easy to move the job from the periodic workflow to the on-merge/on-push workflow. It could even be done with `git commit --amend --no-edit` to get rid of an explicit version bump commit. I would also suggest to use abbreviated hash instead of distribution date, because there may have been multiple commits on one day, and the hash gives a much better answer to your main question. 

When I have time, I'll try to implement what I suggested, and post a pull request.

Sorry if I misunderstand the cause and my suggestions are inapplicable.

thisiskeithb, 2020-07-15:
`bugfix-2.0.x` is updated nearly every day (only to slow down before releases or if Scott needs a much needed break), so I'm not sure this is a real issue.

thinkyhead, 2020-07-15:
It's unfriendly to `--amend` a public repository. And, I find that seeing two Date Bump commits close to one another inspires me to be more productive.