MarlinFirmware/Marlin #18628: [BUG] encoder stoped working after marlin 2.0 upgrade

bohbotjames, 2020-07-12:
### Bug Description

The rotary encoder of my reprap discount smart controller has a bug and only works when I turn it very fast. I can give more details tommorow about my configuration.h and I will also take a video of it to demo it. It worked fine on marlin 1.0, and this bug appeared after I changed to 2.0.

### My Configurations

I am running the latest release (2.0.5.4) on an arduino mega and ramps 1.5 board. 

### Steps to Reproduce

1. Install the latest release of marlin 2.0
2. click on the rotary encoder button to get the the first menu window
3. turn rotary encoder

**Expected behavior:** 

The selected item in the menu should move up or down depending on what side you turn the encoder

**Actual behavior:** 

The selected item doesn't move up or down and just stays selected on the first item. If I were to move the encoder very fast, it goes to the last item, but it skips all the items in between, and if I turn it very fast again, it goes back to the first item.

#### Additional Information

I know that it is not a hardware problem because it worked properly on version 1.0, and because when I turn the encoder on the info screen, the printer speed changes properly and doesn't glitch. Maybe I will try the nightly build tommorow and see if that fixes it.

shitcreek, 2020-07-13:
Please follow the template and include your config files. Did you do a line by line comparison with your old config files?

bohbotjames, 2020-07-13:
I turned on the printer today and the rotary encoder started working normally, but last Friday it wasn't working. I don't know what changed, but since it is working now, you can close this issue. I am having problems with the heated bed though, but I will open another issue for that.

swilkens, 2020-07-13:
You can close the issue yourself, a button exists when adding a new reply to the issue :)

shitcreek, 2020-07-13:
PLEASE DO NOT JUST OPEN ANOTHER THREAD!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-07-13 by shitcreek