MarlinFirmware/Marlin #18633: [BUG] (short description)

bill-orange, 2020-07-12:
Bug Description

Description of the bug :

 Un-commenting `#define BABYSTEP_MILLIMETER_UNITS ` results in Z-offset adjustments made on the LCD to move in large increments with each encoder click.  On my machine one click equaled 8 mm.  Massive in Z-offset terms.

My Configurations:

Attached, but please note that to create the bug, line 1568 in configuration_adv.h need to be un-commented.  It is commented out at present because I commented it out to see if the problem disappeared.  It did.

Steps to Reproduce:

1. Enable BABYSTEP_MILLIMETER_UNITS.
2. Load and run.  always clear the old EEPROM.
3. Adjust Z-offset in configuration.
4. Observe big steps.

Expected behavior:  small steps

Actual behavior: enormous steps

Additional Information:

At first I thought this was a display issue but changing Z-offset during printing resulted in a massive Z move that appeared consistent with the displayed value.

Can someone confirm this behavior?  I doubt this problem is an artifact of my setup, but you never know.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4909496/Marlin.zip)

sjasonsmith, 2020-07-12:
I am sure this is because your `BABYSTEP_MULTIPLICATOR_Z` is set to 8. I've never actually used this feature, so I'm not sure whether this will take a floating point value such as 0.025.

bill-orange, 2020-07-12:
I will test in the next compile. You are correct that a multiplier of around 0.02 should be about right if that fixes it.

bill-orange, 2020-07-12:
Yes!  that did indeed fix the problem.  

Marlin accepted a `BABYSTEP_MULTIPLICATOR_Z` of 0.02.  I think the default was 1.  A value of 1 would be be unusable for Z-offset adjustment.

I am not sure what the right thing to do is.  A comment line saying to reduce `BABYSTEP_MULTIPLICATOR_Z` if `BABYSTEP_MILLIMETER_UNITS` is set, is one possibility.  An #if statement dividing `BABYSTEP_MULTIPLICATOR_Z` by 50 if `BABYSTEP_MILLIMETER_UNITS`  is set is another possibility.  

I will leave this open my now so greater minds can address it if desired.

sjasonsmith, 2020-07-12:
Maybe `BABYSTEP_MULTIPLICATOR_Z` should be replaced with something like `BABYSTEP_DISTANCE` or `BABYSTEP_INCREMENT`, which would be either steps or mm, depending on `BABYSTEP_MILLIMETER_UNITS`.

bill-orange, 2020-07-12:
Another possibility is make babystepping in mm fixed with no option and set the the increment to 0.02 or some other small number. Millimeter movement seems so intuitive I don’t know if we even need the old step option.

thisiskeithb, 2020-07-13:
> I don’t know if we even need the old step option.

I disagree, but that's why we have options 🙂

thinkyhead, 2020-07-15:
A sanity-check should be added that ensures the size of babysteps is not too large. They really should be under 0.1mm when using mm units.

bill-orange, 2020-07-15:
@thinkyhead  That seems like a good solution.

bill-orange, 2020-07-16:
A sanity-check was added to ensure the size of babysteps is not too large. They should be under 0.1mm when using mm units.  Since this was done, this issues is closed.  If the sanity check has some unexpected consequence, that would be a new issue.

closed 2020-07-16 by bill-orange