MarlinFirmware/Marlin #18634: [BUG] [Marlin 2.0.x] LCD messages 3 Point bedleveling

roel8032, 2020-07-13:
LCD messages 3 Point bedleveling.

Anet A6. No need to see the configfiles. It is not in there.

3 point bedleveling
LCD message:
Probing point 1: LCD message: Probing Point 0/3
Probing point 2: LCD message: Probing Point 1/3
Probing point 3: LCD message: Probing Point 2/3

it should be:
Probing Point 1/3
Probing Point 2/3
Probing Point 3/3

ellensp, 2020-07-13:
and what lcd do you have?   we cant tell, because you didn't include your Configuration files. Could be stock, could be anything.
Please attach your configuration files.

roel8032, 2020-07-13:
I updated the post.

 

Met vriendelijke groeten,

 

 

Roel Kisteman

06-22.25.44.80

ellensp, 2020-07-13:
Now can you edit your reply to remove all e-mail address, so the spam bots looking for e-mail address cant harvest them.

ellensp, 2020-07-13:
Your still missing the point, what if we need other settings? language? probe type?  What version of marlin? and a million other questions.

We also want a configuration file that is full and complete so we can drop it in to marlin and replicate this issue.

Attach your config files!

Lord-Quake, 2020-07-13:
@roel8032
I just tested it on my Ender 3 in UBL configuration, latest bug fix and it shows it correctly (1/3, 2/3 and 3/3) on the screen.

So.... post the information as requested by @ellensp. It won't kill you if you want to be helped here.

roel8032, 2020-07-13:
OK.
But UBL (Unified Bed Leveling), is not AUTO_BED_LEVELING_3POINT

Lord-Quake, 2020-07-13:
I took a look at the code:

We might have to take a closer look at the following in G29.ccp and add i+1 to the string output: 

        if (verbose_level) SERIAL_ECHOLNPAIR("Probing point ", int(i), "/3.");
        TERN_(HAS_DISPLAY, ui.status_printf_P(0, PSTR(S_FMT " %i/3"), GET_TEXT(MSG_PROBING_MESH), int(i)));

roel8032, 2020-07-13:
Thanks. Now it is working OK

I changed this in:
\Marlin\src\feature\bedlevel\abl\G29.cpp

line689

//TERN_(HAS_DISPLAY, ui.status_printf_P(0, PSTR(S_FMT " %i/3"), GET_TEXT(MSG_PROBING_MESH), int(i)));
TERN_(HAS_DISPLAY, ui.status_printf_P(0, PSTR(S_FMT " %i/3"), GET_TEXT(MSG_PROBING_MESH), int(i+1)));

The old code above i commented.

Thank you.

roel8032, 2020-07-13:
> and what lcd do you have? we cant tell, because you didn't include your Configuration files. Could be stock, could be anything.
> Please attach your configuration files.

roel8032, 2020-07-13:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(150)
#define ST7920_DELAY_2 DELAY_NS(150)
#define ST7920_DELAY_3 DELAY_NS(150)

But before i used with the same hardware:
//#define ANET_FULL_GRAPHICS_LCD 

But since Marlin 2.0 that gives bad pictures

Lord-Quake, 2020-07-13:
I've put up a pull request: https://github.com/MarlinFirmware/Marlin/pull/18639

Lord-Quake, 2020-07-15:
The commit has been processed into the latest bug fix.
Please test once more and if it is working as expected close this issue.

roel8032, 2020-07-15:
Al working good in the latest bug fix.

closed 2020-07-15 by roel8032