MarlinFirmware/Marlin #18635: \HAL\STM32\timers.cpp - error "Default timer interrupt priority is unspecified or set to a value which may degrade performance."

happy77fr, 2020-07-13:
### Bug Description

Compilation error on \HAL\STM32\timers.cpp for the malrin-=bugfix branch

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4912359/Marlin.zip)

### Steps to Reproduce

Arduino 1.8.13 -> compilation for RemRam V1 board with latest STM32 core

**Expected behavior:** compile

**Actual behavior:** 

"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m7 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Temp/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IX:\\Marlin" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/avr" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Drivers/STM32F7xx_HAL_Driver/Inc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Drivers/STM32F7xx_HAL_Driver/Src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/STM32F7xx" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Middlewares/OpenAMP/virtual_driver" -DSTM32F7xx -DARDUINO=10813 -DARDUINO_REMRAM_V1 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"REMRAM_V1\"" -DSTM32F765xx -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Drivers/CMSIS/Device/ST/STM32F7xx/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\system/Drivers/CMSIS/Device/ST/STM32F7xx/Source/Templates/gcc/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\cores\\arduino" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\variants\\REMRAM_V1" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\libraries\\SrcWrapper\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\libraries\\SPI\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\libraries\\IWatchdog\\src" "-IC:\\Users\\xxxx\\OneDrive - Intel Corporation\\Documents\\Arduino\\libraries\\U8glib\\src" "-IC:\\Users\\xxxx\\OneDrive - Intel Corporation\\Documents\\Arduino\\libraries\\arduino_571695\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\libraries\\SoftwareSerial\\src" "C:\\Temp\\sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp" -o "C:\\Temp\\sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp.o"
C:\Temp\sketch\src\HAL\STM32\timers.cpp:45:4: error: #error "Default timer interrupt priority is unspecified or set to a value which may degrade performance."
   45 |   #error "Default timer interrupt priority is unspecified or set to a value which may degrade performance."
      |    ^~~~~

boelle, 2020-07-13:
try to compile with vscode+platform.io

sjasonsmith, 2020-07-13:
Ahh, you found my new compiler warning!

It is getting increasingly difficult to work within Arduino for these newer boards. Many thing in the frameworks can only be overridden with preprocessor defines. We can do this readily in PlatformIO, but in Arduino I suppose you would have to modify a board definition somewhere.

Feel free to comment out that error to let you compile, you just won't get the benefits of the change that added it. Primarily that gave better tone generation when enabling SPEAKER.

closed 2020-07-14 by thinkyhead