MarlinFirmware/Marlin #18636: Compilation error - STM32 core misaligned with marlin fw on HardwareTimer::attachInterrupt function

happy77fr, 2020-07-13:
### Bug Description

Error at compilation time with the latest STM32 core and marlin-bugfix versions on the function HardwareTimer::attachInterrupt

"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m7 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\\Temp/sketch/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IX:\\Marlin" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F7xx_HAL_Driver/Inc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F7xx_HAL_Driver/Src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32F7xx" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/virtual_driver" -DSTM32F7xx -DARDUINO=10813 -DARDUINO_REMRAM_V1 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"REMRAM_V1\"" -DSTM32F765xx -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F7xx/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F7xx/Source/Templates/gcc/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\variants\\REMRAM_V1" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\SrcWrapper\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\SPI\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\IWatchdog\\src" "-IC:\\Users\\xxxx\\OneDrive - Intel Corporation\\Documents\\Arduino\\libraries\\U8glib\\src" "-IC:\\Users\\xxxx\\OneDrive - Intel Corporation\\Documents\\Arduino\\libraries\\arduino_571695\\src" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\libraries\\SoftwareSerial\\src" "C:\\Temp\\sketch\\src\\HAL\\shared\\eeprom_if_i2c.cpp" -o "C:\\Temp\\sketch\\src\\HAL\\shared\\eeprom_if_i2c.cpp.o"
C:\Temp\sketch\src\HAL\STM32\timers.cpp: In function 'void HAL_timer_enable_interrupt(uint8_t)':
C:\Temp\sketch\src\HAL\STM32\timers.cpp:163:62: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
  163 |       timer_instance[timer_num]->attachInterrupt(Step_Handler);
      |                                                              ^
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/analog.h:45,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h:8,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:41,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from c:\temp\sketch\src\hal\shared\marduino.h:36,
                 from c:\temp\sketch\src\hal\stm32\hal.h:28,
                 from c:\temp\sketch\src\hal\hal.h:26,
                 from c:\temp\sketch\src\inc\marlinconfig.h:30,
                 from C:\Temp\sketch\src\HAL\STM32\timers.cpp:24:
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:10: note: candidate: 'void HardwareTimer::attachInterrupt(callback_function_t)'
  134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
      |          ^~~~~~~~~~~~~~~
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:46: note:   no known conversion for argument 1 from 'void(HardwareTimer*)' to 'callback_function_t' {aka 'std::function<void()>'}
  134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
      |                          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note: candidate: 'void HardwareTimer::attachInterrupt(uint32_t, callback_function_t)'
  138 |     void attachInterrupt(uint32_t channel, callback_function_t callback); // Attach interrupt callback which will be called upon compare match event of specified channel
      |          ^~~~~~~~~~~~~~~
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note:   candidate expects 2 arguments, 1 provided
C:\Temp\sketch\src\HAL\STM32\timers.cpp:166:62: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
  166 |       timer_instance[timer_num]->attachInterrupt(Temp_Handler);
      |                                                              ^
In file included from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/analog.h:45,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h:8,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:41,
                 from C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from c:\temp\sketch\src\hal\shared\marduino.h:36,
                 from c:\temp\sketch\src\hal\stm32\hal.h:28,
                 from c:\temp\sketch\src\hal\hal.h:26,
                 from c:\temp\sketch\src\inc\marlinconfig.h:30,
                 from C:\Temp\sketch\src\HAL\STM32\timers.cpp:24:
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:10: note: candidate: 'void HardwareTimer::attachInterrupt(callback_function_t)'
  134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
      |          ^~~~~~~~~~~~~~~
[... cut ...]
C:\Users\xxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note:   candidate expects 2 arguments, 1 provided

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4912406/Marlin.zip)

### Steps to Reproduce

Standard compilation with arduino 1.8.13

**Expected behavior:** Compilation

**Actual behavior:** no compilation - see above for details

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-07-13:
is this the same as https://github.com/MarlinFirmware/Marlin/issues/18635 ??

sjasonsmith, 2020-07-13:
@boelle no, not the same.

@happy77fr the STM32 support you have installed is too new for Marlin. You are using "STM32 Cores" version 1.9. and you need to use 1.7.

This is another area where things are much easier in PlatformIO, it can ensure the correct version is used, no matter which version you actually have installed.

happy77fr, 2020-07-13:
I tried indeed with core 1.7.0 and the compilation error is gone....... I am using arduino ide for now but I will install vscode+platform IO. Thank you!!!!

closed 2020-07-13 by happy77fr