MarlinFirmware/Marlin #18640: "Preheat ABS bed" doesn't work with Ender-3

mgsantana, 2020-07-13:
Hi there.

I've been wondering why the "Preheat ABS bed" doesn't work with Ender-3 even with branch bugfix-1.1.x. Every time I need to use Control > Temperature > Bed after select "Preheat ABS bed" because only the hot-end is heated that way.

I already did an EEPROM reset but nothing changed.

See my config files attached.

[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/4913965/Configuration_files.zip)

shitcreek, 2020-07-13:
```
// Preheat Constants
#define PREHEAT_1_TEMP_HOTEND 185
#define PREHEAT_1_TEMP_BED     45
#define PREHEAT_1_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_2_TEMP_HOTEND 240
#define PREHEAT_2_TEMP_BED      0
#define PREHEAT_2_FAN_SPEED   255 // Value from 0 to 255
```

It looks like your ABS preheat bed temp is set to 0

thisiskeithb, 2020-07-13:
@shitcreek beat me to it, but the bed was set to 0 in the original Ender-3 firmware, so that's why I used that temperature in the config when I contributed it a couple years ago.

Marlin `1.1.x` has been EOL'd and those temperatures are now the following in Marlin `2.0.x`:

```cpp
// Preheat Constants
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 185
#define PREHEAT_1_TEMP_BED     45
#define PREHEAT_1_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_2_LABEL       "ABS"
#define PREHEAT_2_TEMP_HOTEND 240
#define PREHEAT_2_TEMP_BED     70
#define PREHEAT_2_FAN_SPEED   255 // Value from 0 to 255
```

mgsantana, 2020-07-13:
Oh gosh, I'm sorry guys!

I should be realized that, my bad.

closed 2020-07-13 by mgsantana