MarlinFirmware/Marlin #18645: Geeetech A20T freezing at Stepper change

Bergstern, 2020-07-14:
Hi,
I have flashed the newest Marlin versiin to my A20T (one nozzle, 3steppers).
Everything looks fine and i can print with extruder 0 but when i try to extrude with E1 or E2 i get a freezing. I use a configurated (inverted axis for X,Y,Z TMC2208) version from the examples folder fir the A20T. It seems there is a bug in the new firmware so i cant use the other steppers? 
Perhaps someone can help me.

Legarin, 2020-07-14:
I test it but don't work

Legarin, 2020-07-15:
can I help you with something ?, I make my printer available for testing

thinkyhead, 2020-07-15:
- Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop the ZIP on your next reply.

- Get the latest `bugfix-2.0.x` before proceeding.
- Check the silkscreen on your board to see if it's a different GT2560 board. The expected version is 3.0.
- Add `#define MIXER_NORMALIZER_DEBUG` to your config before re-flashing for testing.
- Try setting `MIXING_STEPPERS` to 2 to see if the machine works when there is only `E0` and `E1`.
- Connect to the printer from a host so that you can issue commands and see the output.
- Enable `DIRECT_MIXING_IN_G1` and experiment with `M165` to see if any mixes work.
- Create virtual tools with `M164` and experiment with `T0`, `T1`, etc., which selects a virtual tool.

AEZ41, 2020-07-17:
Hello,
Same problem on my printer. My board version is 4.0 ( black color ). Is Marlin compatible with v4 ?

Legarin, 2020-07-17:
Please ZIP up your Configuration.h and Configuration_adv.h files and drop the ZIP on your next reply.
***[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4940632/Marlin.zip)

Get the latest bugfix-2.0.x before proceeding.
***It's Ok.

Check the silkscreen on your board to see if it's a different GT2560 board. The expected version is 3.0.
***In theory they are the same except for the screen connector and little else. the version 1.1.8 delivered by the manufacturer is 
   the same for both boards. And in v1.1.8 all works correctly.

Add #define MIXER_NORMALIZER_DEBUG to your config before re-flashing for testing.
***Ok.
Try setting MIXING_STEPPERS to 2 to see if the machine works when there is only E0 and E1.

Connect to the printer from a host so that you can issue commands and see the output.
***Ok

Enable DIRECT_MIXING_IN_G1 and experiment with M165 to see if any mixes work.
***Ok

Create virtual tools with M164 and experiment with T0, T1, etc., which selects a virtual tool.
***just by sending the T1 command it is already blocked and does not show any output later, also it must be turned off and on since the reset button does not respond either.

***Another thing that I forgot to comment on is that if in the printer menu you heat the extruder and place the mixture in any of the 3 extruders and then move the extruder, it works correctly.

I think I found the problem, in version 1.1.8 the tools are created as follows:
T0 = 100% / 0% / 0%
T1 = 0% / 100% / 0%
T2 = 0% / 0% / 100%

in version 2.0.5.4 the tools are not created at startup.

if you create them like this:

M163 S0 P1
M163 S1 P0
M163 S2 P0
M164 S0

M163 S0 P0
M163 S1 P1
M163 S2 P0
M164 S1

M163 S0 P0
M163 S1 P0
M163 S2 P1
M164 S2

everything works well.

So for my part, in the absence of others trying it, the only problem is adding that to the start code before extruding and knowing if it can be set by default in the config.h

Legarin, 2020-07-18:
the only thing that his thing would be for marlin to say that the tool does not exist instead of being blocked

Legarin, 2020-08-17:
The isue it's fixed in 2.0.6.

Thanks

closed 2020-08-17 by ellensp