MarlinFirmware/Marlin #18647: [BUG] NeoPixel in LPC1768 errors

Tillema, 2020-07-14:
### Bug Description

After compiling and recompiling and again and again, I couldn't solve this bug.

Compiling .pio\build\LPC1768\src\src\sd\usb_flashdrive\lib-uhs2\usbhost.cpp.o
Compiling .pio\build\LPC1768\libf8e\Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o
Compiling .pio\build\LPC1768\CMSIS\lib\chanfs\rtc176x.c.o
Compiling .pio\build\LPC1768\CMSIS\lib\usb\cdcuser.cpp.o
Compiling .pio\build\LPC1768\CMSIS\lib\usb\mscuser.cpp.o
Compiling .pio\build\LPC1768\CMSIS\lib\usb\usbcore.cpp.o
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1873:9: error: 'gpio_set' was not declared in this scope
         gpio_set(pin);
         ^~~~~~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1873:9: note: suggested alternative:
In file included from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21:0,
                 from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note:   'LPC176x::gpio_set'
 [[gnu::always_inline]] inline void gpio_set(const pin_t pin, const bool value) {
                                    ^~~~~~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1874:9: error: 'time' has not been declared
         time::delay_ns(550);
         ^~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1876:9: error: 'gpio_clear' was not declared in this scope
         gpio_clear(pin);
         ^~~~~~~~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1876:9: note: suggested alternative:
In file included from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21:0,
                 from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:28:36: note:   'LPC176x::gpio_clear'
 [[gnu::always_inline]] inline void gpio_clear(const pin_t pin) {
                                    ^~~~~~~~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1877:9: error: 'time' has not been declared
         time::delay_ns(450);
         ^~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1881:9: error: 'gpio_set' was not declared in this scope
         gpio_set(pin);
         ^~~~~~~~
.pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1881:9: note: suggested alternative:
In file included from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21:0,
                 from C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1768\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\Users\simon\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note:   'LPC176x::gpio_set'
 [[gnu::always_inline]] inline void gpio_set(const pin_t pin, const bool value) {
                                    ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\libf8e\Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o] Error 1

I don't know how to fix this issue. All is set correct in the Configuration.h. Configuration_adv.h still have to be modified.

### My Configurations

[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/4919138/Configuration_Files.zip)
[Complete_Marlin_with_configs.zip](https://github.com/MarlinFirmware/Marlin/files/4919207/Complete_Marlin_with_configs.zip)

### Steps to Reproduce

1. Download the latest version of Visual Studio Code
   1.1. Install PlatformIO
2. Download the latest version of Marlin 2.0.x
3. Open the Marlin folder in VSC
4. Go to platformio.ini and set default_envs to LPC1768 (line 21)
5. Copy and replace the config files as submitted
6. Hit Ctrl+Alt+B to build
7. Wait a minute to see this error

#### Additional Information

Downloaded Marlin on Jule 13, 2020, around 3:00 PM GMT+01
Rough config:
- Default printer settings: Anet A8
- Board = BTT SKR V1.3
- Display = BTT TFT35 V3.0
- Stepper drivers = TMC2209 UART

ellensp, 2020-07-14:
Need to top of your compile log, shows libraries used, also not if you going to pate it you need to remove < and > around libraries of github hides the name.

Tillema, 2020-07-14:
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)        
--------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.3
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.7.1
|-- <Adafruit NeoPixel> 1.5.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o

p3p, 2020-07-14:
In platofrmio.ini (LPC1768 section) change `Adafruit NeoPixel@1.5.0`  to `Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip` I missed the commit hat changed it to the official branch that doesn't currently support the current lpc176x framework instead of my fork.

ellensp, 2020-07-14:
Adafruit NeoPixel 1.5.0  thats new ill give it a test...

Tillema, 2020-07-14:
> 
> 
> In platofrmio.ini (LPC1768 section) change `Adafruit NeoPixel@1.5.0` to `Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip` I missed the commit hat changed it to the official library that doesn't currently support the current lpc176x framework instead of my fork.

Changed, but gave an error within seconds (see below). Tried with an underscore between Adafruit and Neopixel, but then initial errors occur.

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.3
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo 1.0.0
|-- <LiquidCrystal 1.0.0
|-- <U8glib-HAL 0.4.1
|-- <TMCStepper 0.7.1
|-- <Adafruit NeoPixel 1.5.0
|-- <SailfishLCD
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Assembler messages:
Fatal error: can't create .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o: No such file or directory
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o: No such file or directory
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o: No such file or directory
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\simon\\Downloads\\MARLIN\\Firmware\\.pio\\build\\LPC1768\\.sconsign38.db.tmp':
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1381:
    _exec_main(parser, values)
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1344:
    _main(parser)
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1119:
    nodes = _build_targets(fs, options, targets, target_top)
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1318:
    jobs.run(postfunc = jobs_postfunc)
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Job.py", line 113:
    postfunc()
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\Script\Main.py", line 1315:
    SCons.SConsign.write()
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\SConsign.py", line 117:
    syncmethod()
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 147:
    f = self._open(self._tmp_name, "wb", self._mode)
Exception ignored in: <function dblite.__del__ at 0x000001C2FC70F550>
Traceback (most recent call last):
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 143, in __del__
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 140, in close
  File "C:\Users\simon\.platformio\packages\tool-scons\script\..\engine\SCons\dblite.py", line 147, in sync
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\simon\\Downloads\\MARLIN\\Firmware\\.pio\\build\\LPC1768\\.sconsign38.db.tmp'

p3p, 2020-07-14:
That error is with platformio not Marlin, Remove any cache folders (such as `.pio` in the Marlin directory) to see if that makes it happy.

Tillema, 2020-07-14:
Whoops. Tried. Don't make any difference on the initial issue.

ellensp, 2020-07-14:
please attach current platformio.ini

p3p, 2020-07-14:
Changing to `Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip` fixed your issue. make sure you delete the cache folder as instructed, or click the clean button in vscode before compiling.

ellensp, 2020-07-14:
[env:LPC1768]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip
board             = nxp_lpc1768
build_flags       = -DU8G_HAL_LINKS -IMarlin/src/HAL/LPC1768/include -IMarlin/src/HAL/LPC1768/u8g ${common.build_flags}
# debug options for backtrace
#  -funwind-tables
#  -mpoke-function-name
lib_ldf_mode      = off
lib_compat_mode   = strict
extra_scripts     = Marlin/src/HAL/LPC1768/upload_extra_script.py
src_filter        = ${common.default_src_filter} +<src/HAL/LPC1768>
lib_deps          = Servo
  LiquidCrystal@1.0.0
  U8glib-HAL@0.4.1
  TMCStepper@~0.7.1
  Adafruit NeoPixel@1.5.0   <====================need to edit this
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip

should be Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip

Auravendill, 2020-07-14:
[platformio.ini.txt](https://github.com/MarlinFirmware/Marlin/files/4919418/platformio.ini.txt)
 I still get the error with the changed platformio.ini and after using the clean button

> Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_atmega_st7920_hw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_atmega_st7920_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_atmega_sw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_atxmega_hw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_atxmega_st7920_hw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_esp32_ssd_i2c_fn.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_i2c.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_io.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_linux_ssd_i2c.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_msp430_hw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_null.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_psoc5_ssd_hw_parallel.c.o
.pio\libdeps\LPC1768\U8glib-HAL_ID1932\src\clib\u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
   44 | static uint8_t u8g_i2c_opt;   /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */
      |                ^~~~~~~~~~~
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_raspberrypi_hw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_raspberrypi_ssd_i2c.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_samd51_hw_spi.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_samd51_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_std_sw_spi.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_stm32duino_hw_spi.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_com_stm32duino_ssd_i2c.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_cursor.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_delay.cpp.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_a2_micro_printer.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_flipdisc_2x7.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_gprof.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ht1632.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ili9325d_320x240.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ks0108_128x64.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_lc7981_160x80.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_lc7981_240x128.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_lc7981_240x64.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_lc7981_320x64.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ld7032_60x32.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_null.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_pcd8544_84x48.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_pcf8812_96x65.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_sbn1661_122x32.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1306_128x32.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1306_128x64.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1306_64x48.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1309_128x64.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1322_nhd31oled_bw.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1322_nhd31oled_gr.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1325_nhd27oled_bw.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1325_nhd27oled_bw_new.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1325_nhd27oled_gr.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1325_nhd27oled_gr_new.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1327_96x96_gr.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1351_128x128.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_ssd1353_160x128.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_64128n.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_dogm128.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_dogm132.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_lm6059.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_lm6063.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_nhd_c12832.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7565_nhd_c12864.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7687_c144mvgd.c.o
Compiling .pio\build\LPC1768\lib329\U8glib-HAL_ID1932\clib\u8g_dev_st7920_128x64.c.o
[... cut ...]
rumba32_mks               IGNORED
include_tree              IGNORED
===================================================================================== 1 failed, 0 succeeded in 00:02:03.027 ===================================================================================== 
Der Terminalprozess "C:\Users\Daniel\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

p3p, 2020-07-14:
> In platofrmio.ini (LPC1768 section) change Adafruit NeoPixel@1.5.0 to Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip I missed the commit to the 2.0.x branch that changed it to the official library that doesn't currently support the current lpc176x framework instead of my fork.

You need to change it in the LPC1768 section not at the top.

Auravendill, 2020-07-14:
Thanks, that solved it :)

closed 2020-07-14 by ellensp